platformio/platformio-core #704: Add support for all PIC microcontrollers

radut, 2016-06-22:
I want to be able to use platformio for
PIC 12 Family
PIC 16 Family
PIC 18 Family
PIC 24 Family

Also there is a alternative with pic for arduino named pinguino http://www.pinguino.cc

palfrey, 2016-07-26:
Is this a duplicate of #684?

Palatis, 2016-09-26:
I'm working on `platform-mcs51`, maybe you can benifit from this.
https://github.com/Palatis/platform-mcs51.git

currently `pio` only recognize `gcc` and `clang` for compiler, a bit tedious to work with other compilers, but it works.

brunobasto, 2016-12-07:
+1 support for pic micros would be nice

Synthron, 2020-04-27:
+1 for PIC Series support.
is there any update on when they will be added? I have tons of PIC16F and PIC18F controllers and these are the only controllers I have and can't use `pio` for.

devrim-oguz, 2020-07-16:
Any updates on this?

devrim-oguz, 2020-11-20:
This needs to happen already! +1 For the PlatformIO PIC support

btribley, 2021-02-16:
I guess I will stick with MPLAB on the remaining PIC projects I have. The Arduino is such a nice platform. I used to be a PIC guy but going forward all my new designs will use Arduino. They have a software standard and all the Atmel chips have libraries that make them work the same. Microchip has never understood the importance of firmware standardization. They keep playing with I2C, serial, just about everything making each little variant incompatible with the last. I am done with PIC. Platform IO supports Arduino.

mario2788, 2021-05-10:
I invite you to know JAL to work all 8-bit MCUs of michrochip:
http://justanotherlanguage.org/
https://www.youtube.com/watch?v=J3Vok7s2fXk&t=3s

There is also a core for JAL in Visual Studio:

Requirements & Installation Process: 

    Download,install and run Visual studio code from  https://code.visualstudio.com/download
    Press Ctrl-Shit-X to invoke the Extensions and search for JAL (or trying to open a JAL file will also prompt to install the extension)
    Click Install to add the extension to vscode
    Set the path to your jal compiler exe and library file path by clicking the gear icon in extension manager and selecting Extension Settings (It's also possible by going to editor preferences and searching for JAL)
    Press Ctrl-Shift-P and Type/select "Tasks:Open User Tasks''.
    Select Others (Example to run arbitrary command).
    In the opened tasks.json paste the following code

        {
            // prefilled tasks.json for compiling a JAL file
            "version": "2.0.0",
            "tasks": [
                {
                    "label": "Compile JAL File",
                    "type": "process",
                    "command": "${config:jal.paths.exePath}",
                    "args": [
                        "${file}",
                        "-s",
                        "${config:jal.paths.LibPath}"
                    ],
                    "presentation": {
                        "reveal": "always",
                        "panel": "new"
                    },
                    "problemMatcher": [],
                    "group": {
                        "kind": "build",
                        "isDefault": true
                    }
                }
            ]
        }

    Open any JAL file and press Ctrl-Shift-B to build the JAL file and you are ready to rock !

mario2788, 2021-05-15:
Here you can find the compiler and many examples and libraries made in JAL.

https://github.com/jallib/jallib

mkhuthir, 2023-10-28:
+1
10F200, 10F220, and 10F320 Are cute little ones that deserve to be in platformio for hacking projects.

ivootxt, 2023-12-13:
> Is this a duplicate of #684?

Is not. #684 is about specifically 8bit microcontrollers.

abobija, 2024-06-01:
+1

Any plans so far?

s-celles, 2025-07-06:
Also looking for something similar.

I'm first working on some "build and upload" scripts for PIC16F876A with PICkit3 (yes I know it's pretty old but we have at work several boards with this kind of uC and still a lot of PICkit3 and PICkit2... we have no PICkit4 or 5)

I'm using XC8 compiler and `ipecmd` Microchip command line interface to upload hex files but I'm currently locked with it.

See https://forum.microchip.com/s/topic/a5C3l000000BoiHEAS/t390784

Here is current status of my "project" https://github.com/s-celles/PIC_Test_Project

s-celles, 2025-07-15:
# 🚀 PlatformIO Platform for PIC Microcontrollers - Looking for Testers!

After years of community requests ([#704](https://github.com/platformio/platformio-core/issues/704)), I'm excited to announce a working **PlatformIO platform for 8-bit PIC microcontrollers**! 

This project brings the modern development experience of VS Code + PlatformIO to the PIC ecosystem, offering an alternative to MPLAB X with better integration, IntelliSense, and the powerful PlatformIO build system.

## ✅ What's Working Now

The platform currently supports:

- **🔧 xc8-wrapper**: Python wrapper for Microchip's XC8 C compiler
- **📤 ipecmd-wrapper**: Python wrapper for IPECMD (hex upload tool)  
- **🏗️ platform-pic8bit**: Complete PlatformIO platform definition
- **📦 Automatic XC8 installation** through xc8-wrapper
- **🎯 PIC16F876A board support** (with more coming!)

You can now compile and upload PIC code directly from VS Code or using the `pio` command line!

## 🎯 Currently Supported Hardware

- **PIC16F876A** (quite fully tested)
- More boards coming soon based on community feedback

## 🔄 What's Coming Next

- **📦 Automatic IPECMD installer** through ipecmd-wrapper
- **⚙️ PIC-AS assembly support** for those timing-critical sections
- **🔗 Mixed projects** (C + Assembly code)
- **🚀 Advanced build flags** support (build_flags, upload_flags, etc.)
- **🧪 Comprehensive testing** to improve code coverage
- **📋 More board definitions**
- ✨ **Arduino** like framework for PIC

## 🤝 We Need You!

### Looking for Testers

Do you have experience with PIC microcontrollers? I'd love your help testing:

- **Different PIC models** - What MCUs do you use most?
- **Real-world projects** - How does it perform with your actual code?
- **Platform compatibility** - Testing on Windows, Linux, and macOS in that order because I'm developping under Windows, make some tests in Linux containers through Docker and Dev Containers but I've no access to macOS... except in CI with GitHub Actions)
- **Migration scenarios** - Moving complex projects from MPLAB X

### What MCU Support Do You Need?

Please let us know:
- Which PIC microcontrollers you use most frequently
- Whether you have physical development boards available for testing
- What features are most important to your workflow

### How to Get Involved

1. **🧪 Test the platform** with your projects
2. **🐛 Report issues** you encounter  
3. **💡 Request features** or MCU support
4. **📝 Contribute documentation** or examples
5. **🔧 Submit PRs** for improvements

## 🚀 Quick Start

```bash
# Install the platform (coming soon to PlatformIO registry)
pio platform install https://github.com/s-celles/platform-pic8bit

# Create a new project
pio project init -b pic16f876a

# Build and upload
pio run -t upload
```

## 🎯 Why This Matters

- **Modern IDE experience** for PIC development
- **Cross-platform support** (Windows, macOS, Linux)  
- **Integrated debugging** (not yet tested) and IntelliSense (but xc8-clangd is not yet used)
- **Unified build system** with other embedded platforms
- **Version control friendly** project structure
- **Library management** through PlatformIO

## 📞 Get in Touch

- **Found a bug?** Open an issue
- **Have a suggestion?** Start a discussion
- **Want to contribute?** Check our contribution guidelines
- **Need help?** Feel free to ping me!

## 🙏 Acknowledgments

This project builds upon the foundation laid by the PlatformIO team and the long-standing community request for PIC support. Special thanks to everyone who contributed to the original discussion in [issue #704](https://github.com/platformio/platformio-core/issues/704).

---

**Ready to modernize your PIC development workflow?** Star platform-pic8bit repo and let's make PIC development great again! ⭐

*What PIC microcontroller would you like to see supported next? Drop a comment below!*

devrim-oguz, 2025-07-15:
Oh this can’t be real… You guys are awesome!

danyboy666, 2025-07-15:
Great news! Thanks for this. I will get back to you soon I might be able to test some stuff. I have some diy boards around. I have 18F4550 boards on hand.

s-celles, 2025-07-17:
Pinging @Wiz-IO who might also be interested.

He did
https://github.com/Wiz-IO/XC16

s-celles, 2025-07-20:
🚀 Excited to share my latest open-source project: AtPack-TS-Viewer!

What are AtPack files? 🤔
AtPack files are device description packages from Atmel/Microchip that contain comprehensive XML-based information about their microcontroller families. Think of them as "digital datasheets" that include:
- Device specifications and variants
- Memory layouts and register definitions
- Pin assignments and package diagrams
- Fuse/lockbit configurations
- Programming algorithms

DFP vs CMSIS Packs 📊
Device Family Packs (DFP) like AtPack are Atmel/Microchip's equivalent to CMSIS Packs used in the ARM ecosystem. While CMSIS Packs standardize device support for ARM Cortex processors across toolchains, AtPack files serve the same purpose for Atmel's AVR and SAM microcontroller families. Both provide:
- Standardized device definitions
- Cross-toolchain compatibility
- Automated development environment setup

Why build a viewer? 🛠️
These AtPack files contain incredibly rich technical data, but navigating the complex XML structure manually is painful. Developers need an intuitive way to:

✅ Quickly browse device specifications
✅ Calculate fuse/lockbit settings
✅ Visualize pin layouts and packages (packages are only available in ATMEL AtPack files)
✅ Access memory maps and peripheral details

The Solution 💡
Built with React + TypeScript + Vite, my viewer provides:
🔍 Advanced XML parsing of .atpack files
📱 Responsive interface with modern UI components
🧮 Integrated calculators for fuses and lockbits
📊 Interactive package diagrams with zoom/pan
⚡ Real-time visualization of device data

Current Status & Roadmap 🎯
✅ ATMEL AtPack support - Fully functional for AVR and SAM families
✅ Microchip PIC support - Most feature seems to be implemented now (but have a look)

Try it yourself!
🔗 Live Demo: https://s-celles.github.io/atpack-ts-viewer/
🔗 Source Code: https://github.com/s-celles/atpack-ts-viewer

This tool has already saved me countless hours during embedded development projects. Hope it helps fellow embedded developers navigate device specifications more efficiently!

#EmbeddedSystems #Microcontrollers #OpenSource #ReactJS #TypeScript #Atmel #Microchip #DeveloperTools #EmbeddedDevelopment

See also issue about integrating such a viewer
https://github.com/platformio/platformio-core/issues/5217

btribley, 2025-07-20:
This is very nice. I wonder how consistent Microchip is across devices? If they are consistent, there may be a fairly straightforward way to use these files to create device packages for Platform IO automatically. There are thousands of Microchip 8-bit micro controllers, manually porting even a tiny fraction of them will take a lot of time.

Another idea is a procedure to roll your own device package. This would free up Platform IO developers and potentially get a lot more chips supported quickly.

---- On Sun, 20 Jul 2025 08:17:34 -0500 Sébastien Celles ***@***.***> wrote ---

s-celles left a comment https://github.com/platformio/platformio-core/issues/704#issuecomment-3094514561 

🚀 Excited to share my latest open-source project: AtPack-TS-Viewer!

What are AtPack files? 🤔
 AtPack files are device description packages from Atmel/Microchip that contain comprehensive XML-based information about their microcontroller families. Think of them as "digital datasheets" that include:
Device specifications and variants

Memory layouts and register definitions

Pin assignments and package diagrams

Fuse/lockbit configurations

Programming algorithms

DFP vs CMSIS Packs 📊
 Device Family Packs (DFP) like AtPack are Atmel/Microchip's equivalent to CMSIS Packs used in the ARM ecosystem. While CMSIS Packs standardize device support for ARM Cortex processors across toolchains, AtPack files serve the same purpose for Atmel's AVR and SAM microcontroller families. Both provide:
Standardized device definitions

Cross-toolchain compatibility

Automated development environment setup

Why build a viewer? 🛠️
 These AtPack files contain incredibly rich technical data, but navigating the complex XML structure manually is painful. Developers need an intuitive way to:
✅ Quickly browse device specifications
 ✅ Calculate fuse/lockbit settings
 ✅ Visualize pin layouts and packages (packages are only available in ATMEL AtPack files)
 ✅ Access memory maps and peripheral details
The Solution 💡
 Built with React + TypeScript + Vite, my viewer provides:
 🔍 Advanced XML parsing of .atpack files
 📱 Responsive interface with modern UI components
 🧮 Integrated calculators for fuses and lockbits
 📊 Interactive package diagrams with zoom/pan
 ⚡ Real-time visualization of device data
Current Status & Roadmap 🎯
 ✅ ATMEL AtPack support - Fully functional for AVR and SAM families
 ✅ Microchip PIC support - Most feature seems to be implemented now (but have a look)
Try it yourself!
 🔗 Live Demo: https://s-celles.github.io/atpack-ts-viewer/ 
 🔗 Source Code: https://github.com/s-celles/atpack-ts-viewer 
This tool has already saved me countless hours during embedded development projects. Hope it helps fellow embedded developers navigate device specifications more efficiently!

#EmbeddedSystems #Microcontrollers #OpenSource #ReactJS #TypeScript #Atmel #Microchip #DeveloperTools #EmbeddedDevelopment

See also issue about integrating such a viewer
 https://github.com/platformio/platformio-core/issues/5217

s-celles, 2025-07-20:
I don't know if Microchip AtPack files are very consistant. I currently just did a viewer for one device in an AtPack file (supporting at least partially both ATMEL and Microchip AtPack format).
I could probably write some TypeScript code to output a table comparing all devices in an AtPack file.

What I can say is that AtPack files from ATMEL (legacy) and those from Microchip are really differents (just name them .zip and unzip it to have a look).

What I liked at lot in ATMEL files was presence of packages definition. That's probably not the most important feature for our use case.... but that's a nice one. Unfortunatelly I haven't been able to find something similar in Microchip AtPack files.

What could be next steps?
- Creating a new tab to have a comparison of each device in the AtPack file (so it will be written in TypeScript)
- or creating a Python script to output data for each device to be able to automatically create new PlatformIO boards.

What I need to know is what features are we looking for to create new PlatformIO boards because I don't think we need to extract all data there.

s-celles, 2025-07-20:
This other project could help...  https://s-celles.github.io/atpack-python-parser/ (it's still also WIP)
It's a Python library I did. It's an AtPack file parser written in Python
CLI is nice thanks to Typer (it have hierachical commands)
It can output nice formatted tables or JSON directly from CLI.
We can also use this library to query an AtPack file and get some informations about a specific device.
It's also a project made with AI assistance so some features may be buggy... be aware of this!
What I need to know more exactly is what data do we really need to create a PlatformIO board (supposing we are relying on xc8 for compilation and ipecmd for upload).
If I look at https://github.com/Wiz-IO/XC16/blob/main/boards/WizIO-PIC24F16KA301.json I see
- for build: f_cpu
- for upload: maximum_ram_size, maximum_size, Eeprom (addr I think), EepromSize, Config, ConfigSize ...
I think we need to build on top of atpack-python-parser tools to be able to output this.
Any opinion?