platformio/platformio-core #705: Is it possible to use variables in platformio.ini?

sticilface, 2016-06-24:
I'd like to define some variables that are constant between about 6 [env].  is this possible... 

something like 

```
[platformio]
src_dir = ./MyMelvanimate ; keeps arduino happy :)
data_dir = ./MyMelvanimate/data
env_default = esp12e-debug
test_var='-lstdc++ -Wl,-Tesp8266.flash.4m.ld -lc -I"/Users/amelvin/Arduino/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" -L"/Users/amelvin/Arduino/esp8266/tools/sdk/libc/xtensa-lx106-elf/lib"'
```

then later
```
[env:debug]
platform = espressif
framework = arduino
board = esp12e
;build_flags = -lstdc++ -Wl,-Tesp8266.flash.4m.ld -lc -I"/Users/amelvin/Arduino/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" -L"/Users/amelvin/Arduino/esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" ;   -DDebug_ESPManager -DDEBUG_ESP_PORT=Serial -DDebugUDP ; -DDebugMelvanimate ;-DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI
build_flags = $test_var; -lstdc++ -Wl,-Tesp8266.flash.4m.ld
board_flash_mode = qio
board_f_cpu = 160000000L
board_f_flash = 80000000L
lib_install = 13, 547, 89 ; 64 = arduinojson
upload_port = 192.168.1.70 ; nodemcu
```

What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

sticilface, 2016-06-24:
thats great cheers.  Was a question so moved it to the forum.

ivankravets, 2016-06-24:
Why did you close this issue? That is a great feature for PaltformIO 3.0. I had the same idea 👍 

Now, it should be 100% implemented.

ivankravets, 2016-06-24:
Python support variables by default https://docs.python.org/2/library/configparser.html but only for the vars inside environment.

How about to keep Python syntax and extend this format?

```ini
[someenv]
var1 = hello
var2 = %(var1) world  ; Python understands this

[env:myenv]
test = %(someenv.var2)  ; Python doesn't understand this but we will emulate

```

sticilface, 2016-06-25:
This looks great:)

itsjustvenky, 2016-06-26:
Yes.. this is really a nice idea.a great feature indeed

ivankravets, 2016-09-17:
Please re-test it with the latest development version http://docs.platformio.org/en/latest/installation.html#development-version

Docs: http://docs.platformio.org/en/latest/projectconf.html

closed 2016-09-17 by ivankravets

raymusmunt, 2016-09-17:
Works great @ivankravets

ivankravets, 2016-09-17:
@raymusmunt please give us a 🌟 https://github.com/platformio/platformio/stargazers 😊

commit 5e646959 (2016-09-17): Dynamic variables/templates for "platformio.ini" // Resolve #705