platformio/platformio-core #707: Arduino - include paths missing after update (espressif)

euphi, 2016-06-28:
### Configuration

**Operating system**: kubuntu 16.04

**PlatformIO Version** (`platformio --version`): 
PlatformIO, version 2.11.0

### Description of problem
After updating platforms and libs, Arduino include files are no longer found.

#### Steps to Reproduce

1. `platformio update`
2. `platformio run`

(problem is not solved by deleting .pioenvs)

### Actual Results

```
/ian/.platformio/packages/framework-arduinoespressif/tools/sdk/lwip/include -I.pioenvs/esp01/FrameworkArduino -I.pioenvs/esp01/FrameworkArduinoVariant -I.pioenvs/esp01/ESPAsyncTCP_ID305 -I.pioenvs/esp01/AsyncMqttClient_ID346 -I.pioenvs/esp01/Ticker -I.pioenvs/esp01/Json_ID64 -I.pioenvs/esp01/ESP8266WiFi -I.pioenvs/esp01/ESP8266mDNS -I.pioenvs/esp01/Bounce2_ID1106 -I.pioenvs/esp01/ESP8266WebServer -I.pioenvs/esp01/ESP8266HTTPClient -I.pioenvs/esp01/DNSServer -I.pioenvs/esp01/homie-esp8266 -I.pioenvs/esp01/i2c-sensor-hal .pioenvs/esp01/i2c-sensor-hal/I2CDevice_wiringpi.cpp
.pioenvs/esp01/i2c-sensor-hal/I2CDevice_arduino.cpp:7:22: fatal error: Wire.h: No such file or directory
#include <Wire.h>
^
compilation terminated.

```
### Expected Results

Include files are found

### Additional Information

**The content of `platformio.ini`:**
```ini
[env:esp01]
platform = espressif
framework = arduino
board = esp01
#build_flags = -I/home/ian/.platformio/packages/framework-arduinoespressif/libraries/Wire
```

To solve the problem, I added the build_flags to add path to "Wire" library. This is part of the framework/platform, so I expect platformio to do this automatically as it did in previous version.

```Platforms:
ian@IansLaptopLinux:~/Coding/homie-htu21d-test$ platformio platforms show espressif

espressif   - Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications. [ https://espressif.com/ ]
----------
Package: toolchain-xtensa
Alias: toolchain
Version: 2
Installed: 2016-06-18 15:57:20
----------
Package: tool-esptool
Alias: uploader
Version: 9
Installed: 2016-06-28 22:35:35
----------
Package: framework-arduinoespressif
Alias: framework
Version: 13
Installed: 2016-06-28 22:35:44
----------
Package: ldscripts
Version: 23
Installed: 2016-06-28 22:35:45
```

**Source file to reproduce issue:**

I2C_arduino.cpp (part of lib 578 - sensors). In my case, it is not installed from platformio, but from github, because I plan to add a new sensor. I linked the git-source-directory to the platformio's project lib-dir:
```
ian@IansLaptopLinux:~/Coding/homie-htu21d-test$ ls -lR lib
lib:
total 4
lrwxrwxrwx 1 ian ian  27 Jun 28 17:37 homie-esp8266 -> ../../../git/homie-esp8266/
lrwxrwxrwx 1 ian ian  27 Jun 28 17:27 i2c-sensor-hal -> ../../../git/i2c-sensor-hal
-rw-rw-r-- 1 ian ian 891 Jun 28 17:24 readme.txt

```

ivankravets, 2016-06-28:
Could you try to remove `.pioenvs` directory from the project?

euphi, 2016-06-28:
I already tried without success (see "Steps to reproce")

ivankravets, 2016-06-28:
Please add `#include <Wire.h>` to the one of the file from your project. This is not a bug... It's behavior of PlatformIO 2.0. It doesn't look for nested libs by default. 

The library build system in PlatformIO 3.0 is rewritten from the scratch and you won't have similar problems.

Other option is to enable `lib_ldfcyclic`. See docs for platformio.ini. But I don't recommend to do it.

euphi, 2016-06-29:
Thanks, that solved the problem. I thought, the framework libraries are in always in the include-path.

BTW: Are there some "best practices" how to develop for existing platform-IO libraries?

What I did:

1. I want to extend lib 578 (sensors).
2. I forked it on github and cloned it to a local repository.
3. I created a demo project and sym-linked the library to the lib directory of the demo-project.

This is somehow useful, but has drawbacks in IDE-integration (Eclipise).

What I really need is a library project in platformio, so I can have a real C++ project in eclipse and another C++ project (managed by platformio) referencing this library.

ivankravets, 2016-06-29:
If you add new libs or modify platformio.ini, you need to reinit Eclipse project with `pio init --ide eclipse -b defboard`

ivankravets, 2016-07-04:
> BTW: Are there some "best practices" how to develop for existing platform-IO libraries?

You can clone https://github.com/loopj/i2c-sensor-hal to the `project_dir/lib/Sensors` and then edit here. 

Don't forget to re-init Eclipse project if you add new libraries to any of `lib` folders. We have an issue for that to update project settings "on the fly".

ivankravets, 2016-07-04:
If you have any questions, please use Community Forums http://community.platformio.org/

Thanks.

closed 2016-07-04 by ivankravets