platformio/platformio-core #708: Code compiled with the Arduino IDE executes a lot faster than same coded compiled in PlatformIO

ilovetogetspamed, 2016-07-01:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:  OS X El Capitan Version 10.11.5

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.11.0

### Description of problem

Produced file is executes code slowly and hex created is a bit larger.

#### Steps to Reproduce

1. Compile and upload with Arduino IDE for Mac and run program. Time option 2.  Very quick, about 65  milliseconds to complete.

2. Compile and upload with PlatformIO on Mac and run program.  Time option 2.  Very slow, about 13 seconds to complete.

### Actual Results

~ 13 Seconds to complete 1-wire scan (option 2)

### Expected Results

~ 65 milliseconds to complete 1-wire scan (option 2)

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:megaatmega2560]
platform = atmelavr
framework = arduino
board = megaatmega2560
```

**Source file to reproduce issue:**
```cpp
// *** Configure DS2406 List ***
// this module should be converted into a Class...

/***
 * One Wire Configuration
 */
#include <OneWire.h>       // One Wire Library
#include <DS2406.h>        // DS2406 Extension for One Wire

#define DEBUG_OUT 0

#if DEBUG_OUT
  #define DEBUG_PRINT(a) cmdMessenger.sendCmd(kLog, a)
#else
  #define DEBUG_PRINT(a)
[... 240 lines cut ...]
void loop()
{
  // Process incoming serial data, and perform callbacks
  cmdMessenger.feedinSerialData();

}

void copyDevice(uint8_t *addrOriginal, uint8_t *addrNew) {
    for(int i = 0; i < 8; i++) {
            addrNew[i] = addrOriginal[i];
    }
}

void moveDevice(uint8_t *addr, int position) {

}
```

### Additional info

Unless you happen to have a bunch of Maxim DS2406 Dual Addressable Switches (or other 1-wire devices to test with) and supporting hardware, you won't be able to test this code.

what it boils down to is when executing the Code compiled with the Arduino IDE the Scan (option2) go very, very fast, as I would expect.  No so with the code compiled with PlatformIO

Let me know if you need anything else.

valeros, 2016-07-01:
Hi @ilovetogetspamed thanks for the report!
Is it possible for you to test your project with other board (Uno, Leonardo)?

ilovetogetspamed, 2016-07-01:
I can test it on an Arduino Uno.  I’ll let you know the results ASAP.

ilovetogetspamed, 2016-07-01:
Works identically with an Arduino Uno.  Except that the  Uno is just a bit slower.  More like 14 seconds.

valeros, 2016-07-04:
@ilovetogetspamed Is there any relationship between a number of 1-Wire devices and the scan duration?

ivankravets, 2016-07-05:
@ilovetogetspamed do you use the same `OneWire` and `DS2406` libraries?

ilovetogetspamed, 2016-07-05:
Yes, they are the same.

ivankravets, 2016-07-05:
Could you check it? PlatformIO keeps libraries in `~/.platformio/lib` folder, where Arduino has an own location for the libraries.

ilovetogetspamed, 2016-07-05:
You can close this.  

The problem is that I am an idiot!

I had added a function call which was causing the delay I was noticing.  I hadn’t taken it out in the PlatformIO Version of the code.

I wholeheartedly apologize for wasting your time.

Thank you for putting together such a great product.

cfo

ivankravets, 2016-07-05:
Please give PlatformIO repository a star ⭐ https://github.com/platformio/platformio/stargazers

Thanks a lot!

closed 2016-07-05 by ivankravets