platformio/platformio-core #709: BBC MicroBit configuration

tiferrei, 2016-07-01:
What kind of issue is this?

- [x] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS Sierra (10.12)

**PlatformIO Version** (`platformio --version`): 2.11.0

Hello there, I was trying to use platformIO with the microbit board (bbcmicrobit), however, even the simplest program fails to compile because it misses the main file. I have also tried to add the main source code of the framework to the lib, but as it is compiled by `yotta` the structure is not the same, has anyone used the microbit with PlatformIO before?

Run output:

```
platformio run
[Fri Jul  1 15:40:25 2016] Processing bbcmicrobit (platform: nordicnrf51, board: bbcmicrobit, framework: mbed)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/bbcmicrobit/core/MicroBit.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -DTARGET_MCU_NRF51_16K_S110 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DTARGET_LIKE_MBED -DTARGET_CORTEX_M -DTARGET_NRF_LFCLK_RC -DMBED_BUILD_TIMESTAMP=1464362417.38 -DTARGET_NRF51822 -DARM_MATH_CM0 -DTARGET_NRF51_MICROBIT -DTOOLCHAIN_GCC_ARM -DTARGET_NORDIC -DTARGET_MCU_NRF51 -DTARGET_MCU_NORDIC_16K -DTARGET_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_CORTEX_M0 -DPLATFORMIO=021100 -I.pioenvs/bbcmicrobit/FrameworkMbedInc248832578 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-213564679 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-250686161 -I.pioenvs/bbcmicrobit/FrameworkMbedInc2103166472 -I.pioenvs/bbcmicrobit/FrameworkMbedInc636649457 -I.pioenvs/bbcmicrobit/FrameworkMbedInc24945669 -I.pioenvs/bbcmicrobit/FrameworkMbedInc840111810 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-1439381592 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-715659618 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-1703762731 -I.pioenvs/bbcmicrobit/FrameworkMbedInc615864670 -I.pioenvs/bbcmicrobit/FrameworkMbedInc555830279 -I.pioenvs/bbcmicrobit/FrameworkMbedInc2043334312 -I.pioenvs/bbcmicrobit/FrameworkMbedInc226820145 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-464402851 -I.pioenvs/bbcmicrobit/FrameworkMbedInc1031615723 -I.pioenvs/bbcmicrobit/FrameworkMbedInc857456459 -I.pioenvs/bbcmicrobit/platform -I.pioenvs/bbcmicrobit/core -I.pioenvs/bbcmicrobit/drivers -I.pioenvs/bbcmicrobit/types -I.pioenvs/bbcmicrobit/bluetooth .pioenvs/bbcmicrobit/core/MicroBit.cpp
arm-none-eabi-g++ -o .pioenvs/bbcmicrobit/core/MicroBitDevice.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -DTARGET_MCU_NRF51_16K_S110 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DTARGET_LIKE_MBED -DTARGET_CORTEX_M -DTARGET_NRF_LFCLK_RC -DMBED_BUILD_TIMESTAMP=1464362417.38 -DTARGET_NRF51822 -DARM_MATH_CM0 -DTARGET_NRF51_MICROBIT -DTOOLCHAIN_GCC_ARM -DTARGET_NORDIC -DTARGET_MCU_NRF51 -DTARGET_MCU_NORDIC_16K -DTARGET_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_CORTEX_M0 -DPLATFORMIO=021100 -I.pioenvs/bbcmicrobit/FrameworkMbedInc248832578 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-213564679 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-250686161 -I.pioenvs/bbcmicrobit/FrameworkMbedInc2103166472 -I.pioenvs/bbcmicrobit/FrameworkMbedInc636649457 -I.pioenvs/bbcmicrobit/FrameworkMbedInc24945669 -I.pioenvs/bbcmicrobit/FrameworkMbedInc840111810 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-1439381592 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-715659618 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-1703762731 -I.pioenvs/bbcmicrobit/FrameworkMbedInc615864670 -I.pioenvs/bbcmicrobit/FrameworkMbedInc555830279 -I.pioenvs/bbcmicrobit/FrameworkMbedInc2043334312 -I.pioenvs/bbcmicrobit/FrameworkMbedInc226820145 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-464402851 -I.pioenvs/bbcmicrobit/FrameworkMbedInc1031615723 -I.pioenvs/bbcmicrobit/FrameworkMbedInc857456459 -I.pioenvs/bbcmicrobit/platform -I.pioenvs/bbcmicrobit/core -I.pioenvs/bbcmicrobit/drivers -I.pioenvs/bbcmicrobit/types -I.pioenvs/bbcmicrobit/bluetooth .pioenvs/bbcmicrobit/core/MicroBitDevice.cpp
arm-none-eabi-g++ -o .pioenvs/bbcmicrobit/core/MicroBitFiber.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -DTARGET_MCU_NRF51_16K_S110 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DTARGET_LIKE_MBED -DTARGET_CORTEX_M -DTARGET_NRF_LFCLK_RC -DMBED_BUILD_TIMESTAMP=1464362417.38 -DTARGET_NRF51822 -DARM_MATH_CM0 -DTARGET_NRF51_MICROBIT -DTOOLCHAIN_GCC_ARM -DTARGET_NORDIC -DTARGET_MCU_NRF51 -DTARGET_MCU_NORDIC_16K -DTARGET_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_CORTEX_M0 -DPLATFORMIO=021100 -I.pioenvs/bbcmicrobit/FrameworkMbedInc248832578 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-213564679 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-250686161 -I.pioenvs/bbcmicrobit/FrameworkMbedInc2103166472 -I.pioenvs/bbcmicrobit/FrameworkMbedInc636649457 -I.pioenvs/bbcmicrobit/FrameworkMbedInc24945669 -I.pioenvs/bbcmicrobit/FrameworkMbedInc840111810 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-1439381592 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-715659618 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-1703762731 -I.pioenvs/bbcmicrobit/FrameworkMbedInc615864670 -I.pioenvs/bbcmicrobit/FrameworkMbedInc555830279 -I.pioenvs/bbcmicrobit/FrameworkMbedInc2043334312 -I.pioenvs/bbcmicrobit/FrameworkMbedInc226820145 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-464402851 -I.pioenvs/bbcmicrobit/FrameworkMbedInc1031615723 -I.pioenvs/bbcmicrobit/FrameworkMbedInc857456459 -I.pioenvs/bbcmicrobit/platform -I.pioenvs/bbcmicrobit/core -I.pioenvs/bbcmicrobit/drivers -I.pioenvs/bbcmicrobit/types -I.pioenvs/bbcmicrobit/bluetooth .pioenvs/bbcmicrobit/core/MicroBitFiber.cpp
arm-none-eabi-g++ -o .pioenvs/bbcmicrobit/core/MicroBitFont.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -DTARGET_MCU_NRF51_16K_S110 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DTARGET_LIKE_MBED -DTARGET_CORTEX_M -DTARGET_NRF_LFCLK_RC -DMBED_BUILD_TIMESTAMP=1464362417.38 -DTARGET_NRF51822 -DARM_MATH_CM0 -DTARGET_NRF51_MICROBIT -DTOOLCHAIN_GCC_ARM -DTARGET_NORDIC -DTARGET_MCU_NRF51 -DTARGET_MCU_NORDIC_16K -DTARGET_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_CORTEX_M0 -DPLATFORMIO=021100 -I.pioenvs/bbcmicrobit/FrameworkMbedInc248832578 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-213564679 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-250686161 -I.pioenvs/bbcmicrobit/FrameworkMbedInc2103166472 -I.pioenvs/bbcmicrobit/FrameworkMbedInc636649457 -I.pioenvs/bbcmicrobit/FrameworkMbedInc24945669 -I.pioenvs/bbcmicrobit/FrameworkMbedInc840111810 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-1439381592 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-715659618 -I.pioenvs/bbcmicrobit/FrameworkMbedInc-1703762731 -I.pioenvs/bbcmicrobit/FrameworkMbedInc615864670 -I.pioenvs/bbcmicrobit/FrameworkMbedInc555830279 -I.pioenvs/bbcmicrobit/FrameworkMbedInc2043334312 -I.pioenvs/bbcmicrobit/F [... cut ...]
[... 35 lines cut ...]
  uBit.display.scroll(t);

}

void onButton (MicroBitEvent e) {
   if (e.source == MICROBIT_ID_BUTTON_A && e.value == MICROBIT_BUTTON_EVT_CLICK)
   uBit.radio.datagram.send("6");
}

int main()
{
  uBit.init();

  uBit.radio.setGroup(126);
  uBit.radio.enable();

  uBit.messageBus.listen(MICROBIT_ID_BUTTON_A, MICROBIT_EVT_ANY, onButton);
  uBit.messageBus.listen(MICROBIT_ID_RADIO, MICROBIT_RADIO_EVT_DATAGRAM, onData);

  while (1)
    uBit.sleep(10);
}
```

### Additional info

palfrey, 2016-07-22:
I've just been digging around a bit, and this is a lot harder than it looks. Despite the addition of the Micro:bit to the supported boards list, it won't work with the out-of-the-box mbed. https://github.com/lancaster-university/microbit-dal and https://github.com/lancaster-university/microbit have the main sources to all the bits needed to properly support it, and that's not part of the main mbed github yet, so PlatformIO isn't picking that up.

It almost might sense for this to be an entirely separate framework, but I can't seem to find any nice docs on building a new one...

ivankravets, 2016-07-22:
@palfrey please wait for the first [PlatformIO 3.0](https://github.com/platformio/platformio/milestone/24) beta. It will have support for mbed modules.

palfrey, 2016-07-22:
@ivankravets: Awesome! Is there any particular branch or issue I should be watching to keep up with that or be able to test development versions?

ivankravets, 2016-07-22:
I've jus taken a look at https://github.com/lancaster-university/microbit and I don't have good news. It depends on https://github.com/lancaster-university/microbit-dal that is written for CMake and hardcoded for yota https://github.com/lancaster-university/microbit-dal/blob/master/source/CMakeLists.txt

Nevertheless, we can try together to make it working. PlatformIO 3.0 has a new refactored powerful cross platform build system that could be tuned "on-the-fly". No need to write custom frameworks (PlatformIO 2.0) or etc.

If someone is ready to play with it, I can explain.

palfrey, 2016-07-23:
So, I decided to have a quick bit of hacking around this morning, and have come up with https://github.com/palfrey/microbit-blink. It's meant to be a "hello world" app for Micro:bit using PlatformIO that was basically me trying to establish just how much deviation there was in the codebase from core mbed, and the short answer is "lots". Lancaster have all sorts of their own very specific versions of the various libraries that are needed, and it was kinda horrible. OTOH, the repository does compile a file that doesn't whinge for the Micro:bit, but it doesn't seem to do anything useful yet :( I'm heading out to a social event now, but would be interested in exploring this further later this week...

Yes, I'm aware it does things in a really awful way by recompiling everything as it's own source code, not as libraries. It's intended mostly as a "can this be done at all" experiment before then getting into the depths of doing this with the proper PlatformIO way.

ivankravets, 2016-07-24:
@palfrey do you mean that our version of ARM mbed framework from https://github.com/mbedmicro/mbed is not compatible with https://github.com/lancaster-university/microbit ?

palfrey, 2016-07-24:
Possibly not. I've found https://github.com/lancaster-university/mbed-classic as well, and I'm not sure how that fits in with https://github.com/mbedmicro/mbed. TBH, the entire set of repositories seems like someone's hacked something together that they know how to make work, but haven't shared with the rest of us yet...

Other bits of progress made: Tried using the debugging guide at https://docs.mbed.com/docs/debugging-on-mbed/en/latest/Debugging/debugging_microbit/ and got not very far because it pretty much immediately jumps to 0xfffffffe and I'm not entirely sure why. OTOH, the HEX file I've got at the moment gets the following info out of srec_info

```
(ENV)palfrey@weizen:[~/src/microbit-blink] srec_info .pioenvs/bbcmicrobit/firmware.hex -intel
Format: Intel Hexadecimal (MCS-86)
Execution Start Address: 0002A12D
Data:   018000 - 02EA23
```

Note the single Data line, and the complete lack of anything at the beginning of memory. Running similar commands on known good programs (a "hello world" from elsewhere I've got for example) gets me two Data blocks, one starting at 000000 and that's also the same for the HEX files in the debug guide, which makes me think there's something missing there. Bootloader possibly?

palfrey, 2016-07-24:
Hah! Some success. So having been digging around in https://github.com/lancaster-university/microbit-targets and especially https://github.com/lancaster-university/microbit-targets/blob/master/bbc-microbit-classic-gcc-nosd/ld/NRF51822.ld which is the "no softdevice" version of the code, I edited my local ~/.platformio/packages/framework-mbed/variant/NRF51_MICROBIT/mbed/TARGET_NRF51_MICROBIT/TOOLCHAIN_GCC_ARM/NRF51822.ld and changed the Flash origin to "0x00000000". This, in combination with "-DMICROBIT_DBG=1 -DMICROBIT_HEAP_DBG=1" in my build_flags got me the first indication I actually had something half-functional in the form of some heap debug on the serial port. Still not got working leds yet, but at least it's running my code :)

ivankravets, 2016-07-24:
This is not a big problem. We can specify custom LD script per board which will be used instead ARM mbed.

When you have work "blink" project, I'll explain how to use PlatformIO 3.0 and custom build script for the specific library.

palfrey, 2016-07-24:
After quite some struggling, I have a working "blink". Checkout https://github.com/palfrey/microbit-blink and do the following: 

1. Edit  ~/.platformio/packages/framework-mbed/variant/NRF51_MICROBIT/mbed/TARGET_NRF51_MICROBIT/TOOLCHAIN_GCC_ARM/NRF51822.ld and change the Flash origin to "0x0001C000"
2. Run "platformio run"
3. Make sure you've got [SRecord](http://srecord.sourceforge.net/) installed (`brew install srecord` works on a Mac)
4. `srec_cat lib/microbit-targets/bbc-microbit-classic-gcc-s130/bootloader/BLE_BOOTLOADER_RESERVED.hex -intel lib/microbit-targets/bbc-microbit-classic-gcc-s130/softdevice/s130_nrf51_1.0.0_softdevice.hex -intel .pioenvs/bbcmicrobit/firmware.hex -intel -o microbit-combined.hex -intel --line-length=44`
5. Copy `microbit-combined.hex` to the root of your Microbit

There's possibly ways to do this without SRecord, and also this gives you the all-singing all-dancing firmware with BLE support, but rather low spare RAM, so other options are wanted in the future, but this at least gets us somewhere.

ivankravets, 2016-07-24:
Woooow! Great result! 👍 

I see that you "linked" libraries to the source code and used an extra script. How can we simplify it?

1. Please [install in virtual environment PlatformIO 3.0](https://github.com/platformio/platformio/issues/423) (scroll down to the installation section)
2. PlatformIO 3.0 understands ARM mbed "module.json" manifest. It means that "extraIncludes" should be handled automatically.
3. You can extend PlatformIO Library Build System with custom extra script per each library 😊 In this case, need to create [library.json](http://docs.platformio.org/en/feature-platformio-30/librarymanager/config.html#build) file in the root of library and specify `build` field. This manifest could be useful for you to have custom build flags per library or etc.
4. I'm working on the new library manager. It will allow to installation of the library directly from external resource (VCS, Github, etc). I hope that I'll push initial support tomorrow.

P.S: To see verbose build output in PlatformIO 3.0, please use `-v`.
P.S.S: We can have custom `library.json` for lancaster-university's library and keep it in our library mirror repository. Only the manifest file. See examples, https://github.com/platformio/platformio-libmirror/tree/master/configs/mbed

palfrey, 2016-07-24:
So, I've just done step 1, but I'm running into issues with bits of the library support. Someone else has already made a library.json for the core microbit bits (http://platformio.org/lib/show/361/Microbit) which looks perfectly good to me, but as far as I can tell doesn't pull in any of the headers or source from the Microbit git repository referenced there. Given there's a perfectly good "module.json" in that folder with "extraIncludes" set, this is particularly confusing. Any thoughts as to why this one doesn't work?

ivankravets, 2016-07-24:
@palfrey I've just pushed PR to that repo https://github.com/rossng/microbit/pull/1

You can copy my changes to your local `library.json` while manifest's author has not merged yet.

palfrey, 2016-07-24:
Still no luck. https://github.com/palfrey/microbit-blink/tree/platform-3.0 has my WIP v3 version. I've copied your library.json changes into ~/.platformio/lib/Microbit_ID361 (and also https://raw.githubusercontent.com/rossng/microbit/master/module.json just for good luck) and no luck. The build flags specified are not turning up in the build of main.cpp, and so it can't find MicroBit.h

ivankravets, 2016-07-24:
@palfrey This is a bug... I'll fix it tomorrow and reply here. 

Thanks and good night! 😊

ivankravets, 2016-07-25:
@palfrey I have some progress on it. I'll push modified PlatformIO later. All builds success without any hacks. The only one change is replacement with `"twi_master_init_and_clear()", "twi_master_init_and_clear(NRF_TWI1)"`. 

Nevertheless, I have problem with linker. Do you have any ideas?

```
arm-none-eabi-g++ -o .pioenvs/bbcmicrobit/firmware.elf -Wl,--gc-sections -Wl,--wrap,main -mcpu=cortex-m0 -mthumb --specs=nano.specs -Wl,-T"/Users/ikravets/.platformio/packages/framework-mbed/variant/NRF51_MICROBIT/mbed/TARGET_NRF51_MICROBIT/TOOLCHAIN_GCC_ARM/NRF51822.ld" .pioenvs/bbcmicrobit/src/main.o -L/Users/ikravets/.platformio/platforms/nordicnrf51/ldscripts -L/Users/ikravets/.platformio/packages/framework-mbed/variant/NRF51_MICROBIT/mbed/TARGET_NRF51_MICROBIT/TOOLCHAIN_GCC_ARM -L.pioenvs/bbcmicrobit -Wl,--start-group -lstdc++ -lsupc++ -lnosys -lmbed -lc -lgcc .pioenvs/bbcmicrobit/libFrameworkMbed.a -lc -lgcc -lm .pioenvs/bbcmicrobit/lib/libnrf51-sdk.a .pioenvs/bbcmicrobit/lib/libble.a .pioenvs/bbcmicrobit/lib/libble-nrf51822.a .pioenvs/bbcmicrobit/lib/libmicrobit.a .pioenvs/bbcmicrobit/lib/libmicrobit-dal.a -Wl,--end-group
.pioenvs/bbcmicrobit/lib/libmicrobit-dal.a(MicroBitFiber.o): In function `schedule()':
MicroBitFiber.cpp:(.text._Z8schedulev+0x6e): undefined reference to `swap_context'
MicroBitFiber.cpp:(.text._Z8schedulev+0xb8): undefined reference to `save_context'
MicroBitFiber.cpp:(.text._Z8schedulev+0xc4): undefined reference to `restore_register_context'
MicroBitFiber.cpp:(.text._Z8schedulev+0x108): undefined reference to `swap_context'
.pioenvs/bbcmicrobit/lib/libmicrobit-dal.a(MicroBitFiber.o): In function `invoke(void (*)(void*), void*)':
MicroBitFiber.cpp:(.text._Z6invokePFvPvES_+0x20): undefined reference to `save_register_context'
collect2: error: ld returned 1 exit status
[.pioenvs/bbcmicrobit/firmware.elf] Error 1
```

palfrey, 2016-07-25:
@ivankravets Yeah, you're missing the little bit of assembler code from CortexContextSwitch.s. Multiple versions in https://github.com/lancaster-university/microbit-dal/tree/master/source/asm and you'll want CortexContextSwitch.s.gcc

palfrey, 2016-07-27:
@ivankravets How are you getting on with this?

ivankravets, 2016-07-27:
This issue depends on #500 and #533. We should fix these issues before.

Need to pre-built (or find somewhere) static `srec_cat` package for the all popular host OS (Win, macOS, Linux + Linux ARM).

Could you help us with this?  See here examples of our packages https://bintray.com/platformio/dl-packages/tool-openocd#files

palfrey, 2016-07-27:
Any examples of how the compilation is managed for existing tools? e.g. openocd? If so, I have access to all of those types of systems for making srec_cat.

ivankravets, 2016-07-27:
@palfrey we do that manually (download source code, install deps/libs, `make`)  😢  

We would be thankful if you share somewhere these packages.

ivankravets, 2016-07-28:
I have 1 error. See my log http://pastebin.com/Qc6nGQ1T

I hope you will fix it.

palfrey, 2016-07-29:
Will hopefully have a better look at this at the weekend, but meanwhile, to solve the createBLEInstance issue you'll need https://github.com/lancaster-university/nRF51822/blob/master/source/nRF5xn.cpp. The commit in my code that fixed this for me was https://github.com/palfrey/microbit-blink/commit/c40f90324111da6ec3d89fd86715093759725e4f

palfrey, 2016-07-30:
@ivankravets: When trying to download from https://cl.ly/1X1B2i461i31 I get "access denied"

ivankravets, 2016-07-30:
I've just updated links in the comments. Something is wrong with CloudApp service. Can you download now?

palfrey, 2016-07-30:
Attached is a ZIP file with Linux (x64), OS X and Windows binaries for srec_cat, as well as a build script that works on all 3. Going to get a Raspberry Pi booted up so I can do the ARM build as well....

[binaries.zip](https://github.com/platformio/platformio/files/391967/binaries.zip)

I can download the link from CloudApp now, but haven't had a good look at it yet

ivankravets, 2016-07-30:
@valeros please add `srec_cat` package to out repository.

ivankravets, 2016-07-30:
@palfrey thanks a lot for the binaries! 

I forgot to note that `platformio.ini` should be clean. The latest PIO3 development version should build libraries without problems. See https://github.com/platformio/platformio/issues/709#issuecomment-235966940

Our aim is to make working libraries and then need to create forks with `library.json` and extra scripts (if library requires it).

palfrey, 2016-07-30:
So the core problem with the version of my code from CloudApp is the following:

```
Library Dependency Map
|-- <Microbit> (/Users/palfrey/src/microbit-blink/microbit-blink-platform-3.0/lib/Microbit_ID361)
|   |-- <Microbit-DAL> (/Users/palfrey/src/microbit-blink/microbit-blink-platform-3.0/lib/Microbit-DAL_ID360)
|   |   |-- <nrf51-sdk> v2.2.0 (/Users/palfrey/src/microbit-blink/microbit-blink-platform-3.0/lib/nrf51-sdk-2.2.0-mb4)
|   |   |-- <ble> v2.5.0 (/Users/palfrey/src/microbit-blink/microbit-blink-platform-3.0/lib/BLE_API-2.5.0-mb3)
|   |   |   |-- <nrf51-sdk> v2.2.0 (/Users/palfrey/src/microbit-blink/microbit-blink-platform-3.0/lib/nrf51-sdk-2.2.0-mb4)
|   |-- <nrf51-sdk> v2.2.0 (/Users/palfrey/src/microbit-blink/microbit-blink-platform-3.0/lib/nrf51-sdk-2.2.0-mb4)
```
The dependency tree for Microbit-DAL should include ble-nrf51822 (i.e the lib in lib/nRF51822-2.5.0-mb6) but it doesn't, and various combinations of module/library.json editing don't seem to fix it. Any ideas?

ivankravets, 2016-07-30:
> The dependency tree for Microbit-DAL should include ble-nrf51822

Do you see any files in Microbit-DAL that depend on ble-nrf51822?

palfrey, 2016-07-30:
Yes.  `MicroBitBLEManager.cpp` from Microbit-DAL needs the `btle_set_gatt_table_size` function from `btle.cpp`. Also`BLE.cpp` from BLE_API needs `nRF5xn.cpp` for `createBLEInstance`, both in ble-nrf51822 and neither have it listed as a dependency.

ivankravets, 2016-07-30:
Cool, give me a few minutes to debug this machinery.

ivankravets, 2016-07-30:
Ok, this machinery tells me that there are no links to `ble-nrf51822` Is it possible? Ok, in a few words... The new Library Dependency Finder (LDF) has C Preprocessor behaviour. If a library depends (via include) on the other library, it will be added to build process.

Do you see any connections between Microbit-DAL and ble-nrf51822? The phrase " `MicroBitBLEManager.cpp`  needs the `btle_set_gatt_table_size` function" doesn't work here. LDF doesn't analyze prototypes. Only includes and macros (#if, ifdef, etc.). What is more, it can evaluate the condition for `#if...` and don't matter that PlatformIO doesn't call `gcc -E` for that. We use Python based implementation. 

----

If there is no link between libraries, you can tell build system to force library for build process
```ini
[env:***]
lib_force = ble-nrf51822
```

P.S Why `Microbit-DAL` doesn't include headers from ble-nrf51822?

palfrey, 2016-07-30:
So, `MicroBitBLEManager.cpp` for reasons I'm not sure about, has an explicit `uint32_t btle_set_gatt_table_size(uint32_t size);` instead of actual include header usage, which means that attempts to figure out that link automatically are probably going to fail!

`lib_force` fixes the problem, but can I do a `lib_force` equivalent in the library.json (to avoid the platformio.ini requirement)? Adding a `dependencies` line doesn't appear to work.

ivankravets, 2016-07-30:
> lib_force fixes the problem, but can I do a lib_force equivalent in the library.json (to avoid the platformio.ini requirement)? Adding a dependencies line doesn't appear to work.

I've not implemented it yet. OK, again, give me a few minutes and I'll implement it. Also, I don't see problems to handle `dependencies` from other manifests, like `module.json`. 

I'll answer here.

ivankravets, 2016-07-30:
Please update to the 3.0.0.dev14 and read http://docs.platformio.org/en/feature-platformio-30/librarymanager/config.html#dependencies

The mbed format for dependencies is allowed. Also, take a look at [this paragraph](http://docs.platformio.org/en/feature-platformio-30/faq.html#how-works-library-dependency-finder-ldf)

> If project or library contains own dependencies list (see dependencies), the LDF will not looking for dependencies in the source code. The specified libraries will be built automatically without check.

It means, then if dependencies are specified manually, LDF will be turnd OFF for this library. In this case, need to specify ALL dependencies or nothing (LDF will find automatically). It looks like "manual" and "auto" control 😊 

P.S: The last thing that we need here is to add support for `srec_cat` (@valeros will do that) and create `library.json` manifests for all microbit dependencies. With the new library manager we can use direct repository links (is not released yet), but I recommend to register all deps in the PlatformIO Library registry to avoid issue when git/hg is not installed on the user machine.

palfrey, 2016-07-30:
Once I updated the Microbit-DAL library.json to have the following block, it compiled ok
```
    "dependencies": {
        "nrf51-sdk": "2.2.0",
        "ble": "2.5.0",
        "ble-nrf51822": "2.5.0"
    },
```
One remaining item is the question of should all Micro:bit projects have the core "Microbit" library as an automagic dependency? Technically you *can* write code for them without it, but you're basically back to core mbed code at that point, and getting access to the peripherals is much harder. Also, the "Microbit" library is basically just a nice wrapper around the "Microbit-DAL" code, but all the standard Micro:bit C examples use that, so it's probably worth doing it. Can libraries be set as "always install this for this target type"?

Re: `srec_cat` support: I'm assuming that's both halves of that? Both adding the `tool-srec_cat` or whatever it's called package, and then the using of the microbit-targets (https://github.com/lancaster-university/microbit-targets) project bootloader/softdevice code to make a actually usable hex file?

ivankravets, 2016-07-30:
> Can libraries be set as "always install this for this target type"?

If you can create another build environment in `platformio.ini` without dependency to Microbit library. What is more, if LDF found library and you don't need it, just put it to `lib_ignore`.

> Re: srec_cat support: I'm assuming that's both halves of that? 

Yes, this will be built into a development platform. You don't need to do extra steps. Let's wait for @valeros 

P.S: Thanks a lot for the great help!!! 👍

palfrey, 2016-07-30:
> P.S: Thanks a lot for the great help!!!

Given the extent to which PlatformIO is a massive improvement on other things I've done with embedded systems (and I spent 4.5 years doing a PhD in embedded systems, so I know how frustrating things get), and the usefulness when I was doing some recent ESP8266 stuff, I'm glad to help out when I want something to work and it's almost there!

palfrey, 2016-07-31:
I've now compiled a Linux ARM binary (on a Raspberry Pi, so should work for the main ARM use case). Attached is a revised binaries.zip with that in, as well as a slightly updated version of the build script that works on all 4 platforms.

[binaries.zip](https://github.com/platformio/platformio/files/392397/binaries.zip)

palfrey, 2016-08-03:
I see that a588e88fec179ca7d090d245c78d70851c0d3898 has been merged into `feature/platformio-30` but even after upgrading to that, I'm not seeing the `srec_cat` kicking in yet. Are there any other changes I need to make to test this locally?

ivankravets, 2016-08-03:
@palfrey it has not been merged yet. PlatformIO 3.0 has decentralized architecture with development platforms. See https://github.com/platformio?utf8=✓&query=platform-

I'm working on the merging. Give me a few minutes. I'll reply here.

ivankravets, 2016-08-03:
Please upgrade to the latest PlatformIO 3.0 dev, remove `~/.platformio/platforms` folder and it should work now.

ivankravets, 2016-08-03:
Would be good to run `platformio update` to update all packages.

palfrey, 2016-08-03:
Did that, got the following (from `platformio run -v`, after LDF output):
```
/Users/palfrey/.platformio/packages/tool-sreccat/srec_cat -intel .pioenvs/bbcmicrobit/userfirmware.hex -intel -o .pioenvs/bbcmicrobit/firmware.hex -intel --line-length=44
arm-none-eabi-size -B -d .pioenvs/bbcmicrobit/firmware.elf
misplaced "-Intel" option
Usage: srec_cat [ <option>... ] <filename>...
srec_cat -Help
srec_cat -VERSion
srec_cat -LICense
scons: *** [.pioenvs/bbcmicrobit/firmware.hex] Error 1
```

I think it's still missing the softdevice firmware judging from the srec_info output (also that lines up with a missing argument before the first "-Intel")

valeros, 2016-08-04:
Hi @palfrey !
Please make sure that you are using latest `framework-mbed` package with file `target.json` here
`/Users/palfrey/.platformio/packages/framework-mbed`

palfrey, 2016-08-04:
Ah... Was missing the target.json entirely from there. Ran `platformio platforms update` and got framework-mbed @ 1.121.1 installed (did have 1.121.0). `platformio run -t upload --upload-port /Volumes/MICROBIT` now successfully installs a working Micro:bit binary!

ivankravets, 2016-08-04:
@palfrey what is your `platformio --version`? PlatformIO should update packages automatically when you upgrade  it.

palfrey, 2016-08-04:
3.0.0.dev19

ivankravets, 2016-08-04:
Thanks, that is strange. I'll take a look.

The last thing that we need is to create Microbit dependencies and register with http://platformio.org/lib

I'll do that when fix https://github.com/platformio/platformio/issues/413

palfrey, 2016-08-08:
Now that #413 is merged, I've got a new version of my blink project (which I've added a few bells and whistles to along the way...) at https://github.com/palfrey/microbit-blink/tree/platform-3.0. At the moment however the [platformio.ini](https://github.com/palfrey/microbit-blink/blob/platform-3.0/platformio.ini) has all of the dependencies listed, as without that it doesn't appear to be able to locate the others, when I think that all *should* be needed is the "Microbit" one. 

Given that one has a [library.json](https://github.com/palfrey/microbit/blob/platformio-3.0/library.json) with what looks like a perfectly good Microbit-DAL dependency, I think there's either a bug or some part of the library dependency code is missing?

ivankravets, 2016-08-09:
1. I don't recommend to use VCS because it requires to have installed client. The idea of PlatformIO Library Manager to work from a box without dependencies to system software.
2. PlatformIO Library Manager handles only `.library.json` manifest file from the package. This file creates by PaltformIO Library Crawler and fills with required information (library ID in the registry, other info). The Library Builder uses `library.json`, `module.json` or `library.properties` that depends on a library type.

The ideal solution is when all dependencies are located in PlatformIO Library Registry. In this case, Library Manager will use prepared archives by Library Crawler.

-------

In any case, I updated 2 manifests and now it should work with a simple `platformio.ini`
```ini
[env:bbcmicrobit]
platform = nordicnrf51
framework = mbed
board = bbcmicrobit
lib_deps = microbit@~2.0.0-rc4
```

palfrey, 2016-08-09:
Tried that, got the following. This is using current develop branch of PlatformIO

```
[Tue Aug  9 12:07:06 2016] Processing bbcmicrobit (platform: nordicnrf51, lib_deps: microbit@~2.0.0-rc4, board: bbcmicrobit, framework: mbed)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 compatible libraries
Looking for dependencies...
Error: Could not find `microbit-dal` dependency for `microbit` library
```

ivankravets, 2016-08-09:
Please remove `.piolibdeps` folder and `pio upgrade`

closed 2016-08-09 by ivankravets

palfrey, 2016-08-09:
Nice! Confirmed clean build with 3.0.0a3. I don't have my Micro:bit to hand to test, but will test this evening.

ivankravets, 2016-08-09:
Should we improve here something? 

P.S: Thanks a lot for the great help on this issue! 👍

palfrey, 2016-08-09:
Couple of things still outstanding:
* IIRC, upload location wasn't correctly detected (certainly on my Mac). I had to add "--upload-port /Volumes/MICROBIT" to make it work
* We're still using the "kitchen sink" bootloader/softdevice. These are fine for my purposes, but some larger Micro:bit projects may not work due to lack of space (e.g. MicroPython which appears to use the "no softdevice" option https://microbit-micropython.readthedocs.io/en/latest/devguide/flashfirmware.html)
* The upload process ends with the words "please reset your device", which isn't necessary on the Micro:bit as it automagically flashes and resets when you copy a new image onto it.

Overall, very happy. Will be doing a blog post on the matter in the nearby future, and will send you a link when done.

ivankravets, 2016-08-09:
1. Is fixed in https://github.com/platformio/platformio/commit/fe76192590d51e34da73ec6aedab68b5d376ba29
2. We don't support MicroPython and it's very bad :( We don't have time for that, but have an open issue https://github.com/platformio/platformio/issues/728 . Would be thankful if someone helps us with that. There is no problem to do that with PlatformIO 3.0. All development platforms with build scripts have been moved outside from PlatformIO Core. What is more, we are ready to provide full help in this way.
3. Fixed in https://github.com/platformio/platformio/commit/92e82941fefc99b5d17056b62d5af9972c3c9a6d . Or, maybe, you can propose better wording for https://github.com/platformio/platformio/blob/develop/platformio/builder/tools/pioupload.py#L144?

palfrey, 2016-08-09:
1. fe76192 works well
2. I was only using MicroPython as an example, but simply noting that larger programs like that will be excluded from being able to be built for the Micro:bit unless there's some sort of platform option of "use this different bootloader".
3. Does https://github.com/platformio/platformio/pull/744 work?

palfrey, 2016-08-28:
FYI, there's now a blog post up about all of this http://www.lshift.net/blog/2016/08/28/using-the-bbc-microbit-with-platformio/

[3 more replies]

commit 20be83ef (2016-08-09): Add support for BBC micro:bit board that is built on the ARM mbed and Nordic nrf51 platform // Resolve #709