platformio/platformio-core #710: Unable to run a simple blink led on Arduino Due (build upload but doesn't run on target)

akira215, 2016-07-02:
I successfully ran lots of project on Arduino Due using PlatformIO.
Since the last update, I'm unable to run any app on this board. Build and upload seems to works fine, but nothing happen after board reset. The same app (simple blink led) works fine with Arduino IDE.
I tried the  USB Native port, it doesn't work also.
In the other hand, everything works fine with PlatformIO on Arduino Leonardo board.
I can't figure out what's happening.

Any help appreciated.

Regards

------------------------------------------------------------------

### Configuration

**Ubuntu 16.04**:

**PlatformIO, version 2.11.0**

**BOSSA CLI version: 1.10601.0**
### Description of problem

#### Steps to Reproduce

1. Build & upload any simple project on Arduino due

### Actual Results
Nothing happen after board reset

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:due]
platform = atmelsam
framework = arduino
board = due
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() 
{  
  pinMode(13, OUTPUT);
}

void loop() {
  while(1)
  {
    digitalWrite(13,LOW);
    delay(1000);
    digitalWrite(13,HIGH);
    delay(1000);
  }

}
```

### Additional info

akira215, 2016-07-02:
Ok, working around this problem, my understanding is :
 - The .bin file generated by the building toolchain works fine.
 - The command for bossac seems to be incorrect, the --boot option should be enable to boot from FLASh (instead of ROM)
So basically, if I upload the firmware.bin file generated by platformIO build chain with : 
`bossac --info --port "ttyACM0" -e -w -v -b -U false .pioenvs/due/firmware.bin -R`
it works !!
The command generated by plaformIO-IDE is : 
`bossac --info --port "ttyACM0" --erase --write --verify --reset --debug -U false .pioenvs/due/firmware.bin`
As you can see the only difference is the -b parameter (--boot option)
I'll finally try to find where this command is generated inside the platformIO-IDE to hack this.

Regards

akira215, 2016-07-02:
I finally make it work by adding the `--boot` option in `platformio/builder/scripts/atmelsam.py` at line 169.
Not sure that it will work for all Atmel SAM processor, but I think there is a mistake in this file.
Could you please check this for the future releases ?

Regards

akira215, 2016-07-02:
The complete path of the file for linux user is :
`~/.atom/packages/platformio-ide/penv/local/lib/python2.7/site-packages/platformio/builder/scripts`

Regards

ivankravets, 2016-07-03:
@valeros is this issue caused by https://github.com/platformio/platformio/commit/486529b11251b2a1fb7929d9c7b9b3b0a5879520 ?

akira215, 2016-07-04:
Hi Ivan,
I don't think so because its on Arduino Due platform. The file to be changed is the following : `/platformio/builder/scripts/atmelsam.py`
On my computer `--boot` option is not set. On the repository, the file has this lines:
`if "sam3x8e" in BOARD_OPTIONS.get("build", {}).get("mcu", ""):
        env.Append(UPLOADERFLAGS=["--boot"])
`
It has been corrected by #710 issue
[https://github.com/platformio/platformio/commit/241ad8174fd767dd17fb86a3e8b86cd121685095](url)

Thanks for help anyway

Regards

ivankravets, 2016-07-04:
@akira215 this is a bug. Thanks a lot that reported it. We will release a new version as soon as possible.

Please test the latest development version where this issue is fixed http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-07-04 by ivankravets

akira215, 2016-07-05:
Hi Ivan,

Ok, I will try it by the end of this week. Regards

akira215, 2016-07-10:
Hi Ivan,

I installed the lastest development version (version 2.11.1.dev2) : The 
bug has been fixed, everything works fine.
I ll continue to developp using pio for SAMx µC, so I will be a good 
tester !

Thanks.

Regards

ivankravets, 2016-07-10:
Happy Coding with PlatformIO! 😄

akira215, 2016-07-10:
Thanks.

I may add an extra filter in the Serial Port Monitor, I can't fine my 
happiness in the 6 existings !

Regards

commit e8290054 (2016-07-04): * Fix missed ``--boot`` flag for the firmware uploader for ATSAM3X8E Cortex-M3 MCU based boards (Arduino Due, etc) // Resolve #710