platformio/platformio-core #711: Missing trailing '\' in generated platformio.pro file for Qt creator IDE

viktor-irafm, 2016-07-03:
### Configuration

**Operating system**:
linux amd64

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.11.0

### Description of problem

When used command: `platformio init --ide qtcreator` for generating files for Qt creator if more files are in the src directory the SOURCES section in the resulting platformio.pro file is not correctly formed (trailning \ are not included).

#### Steps to Reproduce

1. add more cpp files into src dir
2. execute `platformio init --ide qtcreator --board uno`

### Actual Results

```
SOURCES += \
    src/main.cpp
    src/file1.cpp
    src/file2.cpp
```

### Expected Results

```
SOURCES += \
    src/main.cpp \
    src/file1.cpp \
    src/file2.cpp
```

ivankravets, 2016-07-04:
Thanks a lot for the report. Could you re-test it with the latest http://docs.platformio.org/en/latest/installation.html#development-version ?

closed 2016-07-04 by ivankravets

viktor-irafm, 2016-07-04:
Thank you, now it is working correctly.

ivankravets, 2016-07-04:
Thanks for the report! 👍 

Please give PlatformIO repository a star ⭐ https://github.com/platformio/platformio/stargazers

commit 935f7cd5 (2016-07-04): Fix missing trailing ``\`` for the source files list when generate project for Qt Creator IDE // Resolve #711