platformio/platformio-core #712: Problem Arduino Yun bootloader and PlatformIO

nautilus525, 2016-07-04:
Dear PlatformIO team:

When I download in Yun board a .cpp including SoftwareSerial.h or AltSoftwareSerial.h libraries, some times I obtain the following Error message:

“Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding 
icons: +++ [upload] Error 1”

And another times, the result is success but when I open the Serial Monitor, it begins correctly but after some second it crashed, showing the following messages:

“Exception in thread rx:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Library/Python/2.7/site-packages/serial/tools/miniterm.py", line 403, in reader
    data = self.serial.read(self.serial.in_waiting or 1)

  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 505, in read
    raise SerialException('read failed: %s' % (e,))
SerialException: read failed: [Errno 6] Device not configured”

Same result if I launch Terminal instead of Serial Monitor.

The worst is that the Yun bootloader has been removed and the board is not recognized any more.

A workaround to recovery the board was to use Arduino IDE and download any Yun sketch using the WiFi port (it downloads both, the sketch and the bootloader too) ... but I can not use PlatformIO and I love it!

Please, I’ll appreciate any help, this configuration/workaround is not very convenient …

Thanks in advance,

Joaquin

ivankravets, 2016-07-04:
Please replace the contents of `~/.platformio/tool-avrdude` with the files located in this archive:
- [tool-avrdude-1.60001.1.zip](https://github.com/platformio/platformio/files/346676/tool-avrdude-1.60001.1.zip)

Did it resolve the issue?

nautilus525, 2016-07-05:
I'm impressed! Fixed up!
I copied the new files in /home/.platformio/packages/
I have done various test cases and now it works perfectly.
Great!  Thank you very much.

ivankravets, 2016-07-05:
Thanks a lot that reported. We will release new version as soon as possible.

closed 2016-07-05 by ivankravets