platformio/platformio-core #713: Misplaced header files in Qt creator project file

viktor-irafm, 2016-07-04:
### Configuration

**Operating system**:
linux amd64

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.11.1.dev0

### Description of problem

When used command: `platformio init --ide qtcreator --board uno` for generating files for Qt creator IDE if header files are in the src directory they are added into SOURCES variable in the resulting platformio.pro file but they should be added into HEADERS variable.

#### Steps to Reproduce

1. place header files into src directory
2. execute `platformio init --ide qtcreator --board uno`

### Actual Results

```
SOURCES += src/main.cpp
SOURCES += src/file2.cpp
SOURCES += src/file2.h
SOURCES += src/main.h
```

### Expected Results

```
SOURCES += src/main.cpp
SOURCES += src/file2.cpp

HEADERS += src/main.h
HEADERS += src/file2.h
```

ivankravets, 2016-07-04:
Please retest with http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-07-04 by ivankravets

viktor-irafm, 2016-07-05:
Yes! I may confirm it works, thanks.

commit 52f15020 (2016-07-04): Split source files to HEADERS and SOURCES when generate project for Qt Creator IDE // Resolve #713