platformio/platformio-core #715: Support for a _REVISION_ macro.

jasiek, 2016-07-09:
What kind of issue is this?

- [ ] Feature Request. 

Every now and then I build firmware and it's difficult for me to figure out, from the POV of the device what version I'm running. I'd be great if there was a predefined macro called `_REVISION_` which would get replaced with a Git/SVN/Hg revision of the software built. 

By default, if there are no changes in the working copy, the macro would get replaced with the commit revision.

If there are changes to the working tree, it could be replaced with either `UNKNOWN` or the commit revision and a character, for example `*`.

ivankravets, 2016-07-09:
1. Please read this article http://blog.squix.org/2016/06/esp8266-continuous-delivery-pipeline-push-to-production.html
2. `build_flags` option allows to run own command. You need:
```
[env:myenv]
build_flags = !echo "-D_REVISION_="$(git rev-parse HEAD)
```

See http://docs.platformio.org/en/latest/projectconf.html#build-flags and examples

ivankravets, 2016-07-09:
What is more, you can extend build system using http://docs.platformio.org/en/latest/projectconf.html#extra-script

```
env.Append(
    CPPDEFINES=["_REVISION_=" % SOME_PYTHON_VAR]
)
```

closed 2016-07-09 by ivankravets