platformio/platformio-core #716: Build error on OSX / ESP8266

andig, 2016-07-09:
### Configuration

**Operating system**: OSX

**PlatformIO Version** (`platformio --version`): 2.11.0

### Description of problem

Compiling esp8266 sketch gives error message about every second time:

	Assembler messages:
	Fatal error: can't create .pioenvs/esp12e/src/tmp_ino_to.o: No such file or directory
	scons: *** [.pioenvs/esp12e/src/tmp_ino_to.o] Error 1
	Assembler messages:
	Fatal error: can't create .pioenvs/esp12e/FrameworkArduino/Esp.o: No such file or directory
	scons: *** [.pioenvs/esp12e/FrameworkArduino/Esp.o] Error 1

	Assembler messages:
	Fatal error: can't create .pioenvs/esp12e/FrameworkArduino/FS.o: No such file or directory
	scons: *** [.pioenvs/esp12e/FrameworkArduino/FS.o] Error 1
	Assembler messages:
	Fatal error: can't create .pioenvs/esp12e/src/webserver.o: No such file or directory
	scons: *** [.pioenvs/esp12e/src/webserver.o] Error 1
	OSError: [Errno 2] No such file or directory: '/Users/andig/Documents/htdocs/vzero/.pioenvs/.sconsign.dbli
	te':
	File "/Users/andig/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1374:
	_exec_main(parser, values)
	File "/Users/andig/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1337:
	_main(parser)
	File "/Users/andig/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1102:
	nodes = _build_targets(fs, options, targets, target_top)
	File "/Users/andig/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1299:
	jobs.run(postfunc = jobs_postfunc)
	File "/Users/andig/.platformio/packages/tool-scons/script/../engine/SCons/Job.py", line 113:
	postfunc()
	File "/Users/andig/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1296:
	SCons.SConsign.write()
	File "/Users/andig/.platformio/packages/tool-scons/script/../engine/SCons/SConsign.py", line 109:
	syncmethod()
	File "/Users/andig/.platformio/packages/tool-scons/script/../engine/SCons/dblite.py", line 127:
	self._os_unlink(self._file_name)
	Exception OSError: OSError(2, 'No such file or directory') in <bound method dblite.__del__ of <SCons.dblit
	e.dblite object at 0x110764390>> ignored
	========================= [ ERROR ] Took 4.90 seconds =========================
	 
#### Steps to Reproduce

1. just run the build - it will fail the first time
2. run the build again - its fine

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2016-07-09:
1. Remove `.pioenvs` folder manually
2. Don't use `sudo`

andig, 2016-07-09:
Not using sudo. Never created such a folder. Does it make sense to remove it once?

ivankravets, 2016-07-09:
In this case, it looks very strange. Yes, you can delete this folder. It's cache folder.

Keep me in touch.

andig, 2016-07-09:
Mhm. Can't find where it actually is- not relative to the current sketch?

ivankravets, 2016-07-09:
See your log above: `/Users/andig/Documents/htdocs/vzero/.pioenvs/`

andig, 2016-07-09:
I see- only exists while the build is running. I've got a suspicion: have sublimetext open in parallel and can see the folder appear and vanish. I'm suspecting it might hold a lock for a short time. I'll close for now and reopen if I can observe this with just atom open.

closed 2016-07-09 by andig

thinkyhead, 2021-08-31:
That's interesting. I also use Sublime Text (4) and I have this same problem on every other build. I'll have to try it without Sublime open and see if it occurs as frequently.

ivankravets, 2021-08-31:
> I'll have to try it without Sublime open and see if it occurs as frequently.

Does it work without Sublime?