platformio/platformio-core #717: Request: Arduino ISP Support

sergiotomasello, 2016-07-11:
Could you add support to [Arduino ISP](http://www.arduino.org/products/shields/arduino-isp) by Arduino Srl? 

Following the lines to be included into `avrdude.conf` file:

```
programmer
id    = "arduinoisporg";
desc  = "Arduino ISP Programmer";
type  = "usbtiny";
connection_type = usb;
usbvid     = 0x2A03;
usbpid     = 0x0049;
;
```

Regards
Sergio

valeros, 2016-07-12:
Thanks, updated!

closed 2016-07-12 by valeros

tiborebner, 2017-07-19:
I still don't see the entry with `id = arduinoisporg` in my PlatformIO-specific `avrdude.conf`. What happened to that? I had to put it in there manually for it to work with my ISP programmer.

ivankravets, 2017-07-20:
@tiborebner Please file new issue here https://github.com/platformio/platform-atmelavr/issues

ivankravets, 2017-07-20:
@tiborebner Please run `pio update`, the `avrdude` package was updated. Thanks.