platformio/platformio-core #722: Cannot build Teensy20 project with USB Type: "MIDI"

venelinpetrov, 2016-07-17:
### Configuration

**Operating system**:
Windows 7 Professional

**PlatformIO Version**:
2.11.1
### Description of problem
The ```build_flags = -USB_MIDI``` flag does not seem to take effect, because midi specific methods are not recognized during compilation (e.g. ```usbMIDI.sendControlChange();```)

### If problems with PlatformIO Build System:

1. Initialize teensy20 project. You should have the following config:

  ```ini
[env:teensy20]
platform = teensy
framework = arduino
board = teensy20
```
2. Add this entry to platformio.ini

  ```ini
build_flags = -USB_MIDI
# or build_flags = -DUSB_MIDI
```
3. Execute

  ```
pio run
```

### Actual Results
Exception is thrown saying that ```midiUSB``` is not recognized

### Expected Results
Successful compilation

ivankravets, 2016-07-18:
@valeros please re-assign this issue to me when will fix it. I'll add this info to docs.

ivankravets, 2016-07-18:
Docs: http://docs.platformio.org/en/latest/platforms/teensy.html#teensy-2-0-and-usb

closed 2016-07-18 by ivankravets

venelinpetrov, 2016-07-18:
Just to avoid any misunderstanding I would like to underline that the problem (and the fix also) does not seem to depend on the teensy board version. I tested it now with teensy 3.0 project and the behavior is the exact same. You can unmention the version in the documentation if you like.

ivankravets, 2016-07-18:
Thanks a lot! I've just updated the docs.

P.S: Does this build work with Teensy 2.0?

venelinpetrov, 2016-07-18:
Yes, testing with the latest development version, the code compiles and uploads successfully to a physical board. Also needed to add the ```build_flags = -D USB_MIDI``` entry as the documentation suggests.
Thanks for the effort, really. Actually, I came up with a similar solution today, but couldn't find time to dig enough in the code in order to verify it.

ivankravets, 2016-07-18:
Thank you, that reported!

Happy coding! 😊

commit d7fa2551 (2016-07-18): Implement Teensy 2.0 USB functionality (HID, SERIAL_HID, DISK, MIDI, etc.) // Resolve #722