platformio/platformio-core #724: Linter-gcc errors using generated .gcc-flags.json using pio init --ide atom

gargamel007, 2016-07-17:
Hi there ! 
------------------------------------------------------------------

### Configuration

**Operating system**: Linux Mint
**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.11.1

### Description of problem
I'm using command line platformio version and a custom Atom IDE with linter-gcc installed.
It seems there is a bug in the generated .gcc-flags.json that I generated using pio init --ide atom.
As a result, as soon as I import <ESP8266WiFi.h> in my .h file, I'm getting lots of gcc errors like this : 
`GCC error 'shared_ptr' in namespace 'std' does not name a typeat line 41 col 9 in /~/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.h`
`GCC error 'WiFiEventHandler' does not name a typeat line 57 col 9 in /~/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.h`

The project compiles fine using `pio run`
The linter errors do not appear on if I do the same import directly in .cpp files.

So in my .gcc-flags.json, I tried changing 
"gccDefaultCFlags": "-fsyntax-only -std=**gnu99** -Wpo...
to 
"gccDefaultCFlags": "-fsyntax-only -std=**c++11** -Wpo...

It now matches gccDefaultCppFlags and fixes all the errors I had before on the linter !

Is this the right way to fix this ? 
Thanks.

ivankravets, 2016-07-17:
Your solution is good. 

`gccDefaultCFlags` this is GCC flags for C/H files.  If you use "C++" features, the header file should be with `.hpp` extension. For `.hpp` files the `gccDefaultCppFlags` will be used.

closed 2016-07-17 by ivankravets

gargamel007, 2016-07-17:
Thanks. Is it better for me to rename my .h file to .hpp, or is it better to keep the gccDefaultCFlags fix ?
In the second case why not fix the Flags generated by platfromio ?

gargamel007, 2016-07-17:
I'm asking this because lots of libs for ESP8266 seem to use .h extensions in place of .hpp ones :(

ivankravets, 2016-07-17:
> In the second case why not fix the Flags generated by platfromio ?

I understand all problems. We don't depend on developers source code. We provide all information for Atom GCC Linter plugin. It decides which flags to use for specific files. If we put C++ flags to "C" files, it will break our linter.

The key benefit of PlatformIO IDE that is uses data from PlatformIO Build System. We don't use CTAG or other static -pre-generators (a lot of IDEs do that). With PlatformIO IDE, your code will be linted with the real build environment (the same as press "Build" command).

gargamel007, 2016-07-17:
Oh my bad. I thought PlatformIO IDE also uses the linter-gcc plugin like I do here with the .gcc-flags.json file. There might be another trick that I'm missing to get the same results using a linter and the "pio run" build system which works using .h or .hpp files.
I'll have a look in the IDE code.

ivankravets, 2016-07-17:
PlatformIO IDE uses `linter-gcc` plugin but provides own data via `.gcc-flags.json`.

gargamel007, 2016-07-17:
Let me know if I'm asking to many questions. I'm still new to PlatformIO and see this as a good learning opportunity :)

As I see in the IDE code, the .gcc-flags.json is generated using : 
`'platformio', ['-f', '-c', 'atom', 'init', '--ide', 'atom', '-b', config[section].board],`
my .gcc-flags.json was generated using : `pio init --ide atom` on the command line... which should do the same.
That files had errors .h files using the linter-gcc plugin. Is there another trick I am missing ?

ivankravets, 2016-07-17:
Linter GCC plugin treats ".h" files as "C" file, not "C++". In this case, it uses `gccDefaultCFlags` flags 😄

gargamel007, 2016-07-17:
OK thanks. So if I use .h files in PIO IDE, it will not fix my issue and the linter will behave differently than the build system. 

I'll use .hpp files for now :)

ivankravets, 2016-07-17:
Ok, I got what do you want to hear from me...

Linter GCC checks each file individually. That is why it differentiates  H/HPP. It doesn't compile source code. 

Now, take a look at PlatformIO Build System. If you use "H" files in CPP files, then GCC Preprocessor will treat these files as "HPP" (will apply C++ flags).

Is it clearer now? I don't know how to resolve this issue better :(

gargamel007, 2016-07-17:
Yup ! I get it. Thanks a lot.