platformio/platformio-core #727: Buggy compile on Arduino Leonardo

golyalpha, 2016-07-22:
### Configuration

**Win10 PRO x64**:

**PlatfromIO 2.11.1** (`platformio --version`):

### Description of problem
My .cpp code from Atom doesn't compile due to Serial not being defined and there us also comparison between signed and unsigned int (not sure if that would compile, but Arduino IDE has no problem).
The problem is, I've got two .cpp files: 
  Authenticator.cpp (has bunch of Serial and works flawlessly)
  MemoryWipe.cpp (fails to compile)

#### Steps to Reproduce

1. Create Authenticator.cpp
2. Create MemoryWipe.cpp
3. Attempt build/upload

### Actual Results
src\MemoryWipe.cpp:4:3: error: 'Serial' was not declared in this scope
and

¨scons: *** [.pioenvs\leonardo\src\MemoryWipe.o] Error 1
### Expected Results
Build successfull

**The content of `platformio.ini`:**
```ini
[env:leonardo]
platform = atmelavr
framework = arduino
board = leonardo
```

**Source file to reproduce issue:**
```cpp
Authenticator.cpp
#include "EEPROM.h"
#include "Keyboard.h"

String serialInput;
byte result;
String userName;
String passWord;
int writeProtectAdd = 0;
int workstationLogonAdd = 1;
int contentAdd = 2;
struct userLogon{
  String user;
  String pass;
};
userLogon userLogin;
String pass;

void setup() {
  Serial.begin(9600);
  Keyboard.begin();
  if (EEPROM.read(workstationLogonAdd) == byte(255)){
    EEPROM.get(contentAdd, userLogin);
    Keyboard.print(userLogin.user);
    Keyboard.press(KEY_TAB);
    Keyboard.releaseAll();
    Keyboard.println(userLogin.pass);
  }
  else{
    EEPROM.get(contentAdd, pass);
    Keyboard.println(pass);
  }
}

void loop() {
  if (EEPROM.read(writeProtectAdd) == 0xFF){
    while(!Serial){}
    Serial.println("XPen is write protected, you cannot change written values.");
    while(true){}
  }
  else{
    while(!Serial){}
    Serial.println("Write protect?");
    while (!Serial.available()) {}
    serialInput = Serial.read();
    if (serialInput == "49"){
      result = 0xFF;
      Serial.println("Saw " + serialInput);
    }
    else {
      Serial.println("Saw " + serialInput);
      result = 0x00;
    }
    EEPROM.write(writeProtectAdd, result);
    Serial.println("User-based login?");
    while (!Serial.available()) {}
    serialInput = Serial.read();
    if (serialInput == "49"){
      result = 0xFF;
      Serial.println("Saw " + serialInput);
    }
    else {
      result = 0x00;
      Serial.println("Saw " + serialInput);
    }
    EEPROM.write(workstationLogonAdd, Serial.read());
    if (result == 0xFF){
      Serial.println("Username:");
      while (!Serial.available()) {}
      while (Serial.available()){
        userName = userName + Serial.read();
      }
      Serial.println("Password:");
      while (!Serial.available()) {}
      while (Serial.available()) {
        passWord = passWord + Serial.read();
      }
      userLogin.user = userName;
      userLogin.pass = passWord;
      EEPROM.put(contentAdd, userLogin);
    }
    else {
      Serial.println("Password:");
      while (!Serial.available()) {}
      while (Serial.available()) {
        passWord = passWord + Serial.read();
        EEPROM.put(contentAdd, passWord);
      }
    }
    while (true) {}
  }
}

MemoryWipe.cpp
#include "EEPROM.h"

void setup(){
  Serial.begin(9600);
  while (!Serial) {}
  for (int i = 0 ; i < EEPROM.length() ; i++) {
    EEPROM.write(i, 0);
  }
  Serial.println("Wipe complete.");
}

void loop(){}

```

### Additional info
I have VS 2015 Comunity Edition installed.

ivankravets, 2016-07-22:
Looks like you forgot to add `#include <Arduino.h>` before all includes

golyalpha, 2016-07-22:
Never needed to... As noted Authenticator.cpp compiles just fine without it.

ivankravets, 2016-07-22:
Try to add to MemoryWipe.cpp

golyalpha, 2016-07-22:
Question, what are the rules for PlatformIO handling multiple cpp files in output?

ivankravets, 2016-07-22:
Sorry, I don't understand your question. PlatformIO builds all source files located in `src_dir`. If you want to skip some files, see http://docs.platformio.org/en/latest/projectconf.html#src-filter

golyalpha, 2016-07-22:
No, that's not the problem, I want it to build both files, but it seems to me like it tries to combine them into one hex file and it trips up on multiple definitions of setup() and loop() now.

golyalpha, 2016-07-22:
I filtered it out using src_filter, thanks.

closed 2016-07-22 by golyalpha