platformio/platformio-core #729: How to compile Wiring Pi to use for cross compilation on a mac

kanedo, 2016-07-26:
I want to compile my raspberry pi projects on my mac. Whenever I try to setup a project for this using 
```
pio init -b raspberrypi_2b --ide clion
```
and then 
```
pio run
```
it tells me 
```
Error: PlatformIO does not support temporary cross-compilation for WiringPi framework. Please run PlatformIO directly on Raspberry Pi
```
but I cant find any information on how to actually to this and where to place the compiled library on my mac. What is the way to do this?
thanks

ivankravets, 2016-07-27:
> but I cant find any information on how to actually to this and where to place the compiled library on my mac

Do you have cross toolchain for RPi with peripheral support? We have basic toolchain for ARM arch but it doesn't support perepherial :(

ivankravets, 2016-12-26:
Temporary solution is to use http://docs.platformio.org/en/latest/plus/pio-remote.html

ivankravets, 2018-06-18:
Moved to https://github.com/platformio/platform-linux_arm/issues/2

closed 2018-06-18 by ivankravets