platformio/platformio-core #730: Multi-file arduino project does not build

saschaludwig, 2016-07-26:
Here is a multi-file project which has problems building in platformio ide 1.3.6 cli 2.11.1:
https://github.com/ESP8266nu/ESPEasy

I get quite a number of errors like this:
`/Users/sl/ownCloud/Arduino/ESPEasy/src/_P006_BMP085.ino:54:39: error: 'Plugin_006_bmp085_begin' was not declared
 in this scope`

Could it be there is a problem with underscores in the .ino filenames?
How could this be fixed?

Thanks, Sascha

palfrey, 2016-07-30:
You need to declare functions in C before you use them, or in this particular case, re-order the functions so that you start with functions in a file that don't need any of your other ones, then the ones that only need the ones you've declared so far, and so on.

http://stackoverflow.com/questions/14307087/whats-wrong-with-this-arduino-code has a good example of the type of thing.

ivankravets, 2016-07-30:
PlatformIO Build System has own INO to CPP auto converter. See
- https://github.com/platformio/platformio/blob/develop/platformio/builder/tools/piomisc.py#L26

It looks like it doesn't convert code very well. I'll look on it later.

ivankravets, 2016-09-01:
Please re-test with PlatformIO 3.0 http://docs.platformio.org/en/stable/installation.html#development-version

closed 2016-09-01 by ivankravets

saschaludwig, 2016-09-02:
Tested with PlatformIO 3.0: code does compile now.
Thank you very much! Awesome work