platformio/platformio-core #732: Cannot upload to SAMD21_XPRO with Mbed framework

deladriere, 2016-07-27:
I can see the serial port with ( pio serial ports list ) where it's connected

/dev/cu.usbmodem1a122
Hardware ID: USB VID:PID=03EB:2111 SER=ATML2130021800004030 LOCATION=26-1.2
Description: EDBG CMSIS-DAP

if I use try to use a specific port like : 
pio run --target upload --upload-port /dev/cu.usbmodem1a122

it reports a success but it's a lie : the code is not uploaded

MacBook-Pro:SAMD21_XPRO ***$ pio run --target upload --upload-port /dev/cu.usbmodem1a122
[Tue Jul 26 14:04:48 2016] Processing samd21_xpro (platform: atmelsam, board: samd21_xpro, framework: mbed)

MethodWrapper(["upload"], [".pioenvs/samd21_xpro/firmware.bin"])
Looking for upload port/disk...
Manually specified: /dev/cu.usbmodem1a122
BeforeUpload(["upload"], [".pioenvs/samd21_xpro/firmware.bin"])
============================================== [SUCCESS] Took 0.89 seconds

even worse : I can specify ANY USB port and it reports a success too !

what am I doing wrong ?
thx

Mac OSX Yosemite
PlatformIO, version 2.11.1
PlatformIO IDE 1.3.6

valeros, 2016-07-29:
Hi @deladriere !
SAMD21-XPRO does not support programming via media disk.
Could you please try an alternative upload method via OpenOCD?
Install OpenOCD package:
```
pio platforms install atmelsam --with-package=tool-openocd
```
Compile a simple blink project project:
```
pio run
```
Run this command in terminal the root directory of your project:
```
/home/user/.platformio/packages/tool-openocd/bin/openocd" -d2 -s /home/user/.platformio/packages/tool-openocd/share/openocd/scripts -f /home/user/.platformio/packages/tool-openocd/share/openocd/scripts/board/atmel_samd21_xplained_pro.cfg -c "program {{.pioenvs/samd21_xpro/firmware.bin}} verify reset shutdown"
```

deladriere, 2016-07-30:
will this affect my SAM Arduino install : I work also (mainly) with Arduino Zero I don't want to loose this

valeros, 2016-07-31:
@deladriere It's only an uploading tool, it won't affect your Zero board.

deladriere, 2016-08-01:
I tried this command but it just gave me a blinking caret prompt
">_"

valeros, 2016-08-01:
OK, let's try easier command:
```
/home/user/.platformio/packages/tool-openocd/bin/openocd -d2 -s /home/user/.platformio/packages/tool-openocd/share/openocd/scripts -f /home/user/.platformio/packages/tool-openocd/share/openocd/scripts/board/atmel_samd21_xplained_pro.cfg
```

deladriere, 2016-08-01:
terminal response : 
 bash: /home/user/.platformio/packages/tool-openocd/bin/openocd: No such file or directory

valeros, 2016-08-01:
Please use your login name instead of `user` in this command.

deladriere, 2016-08-01:
same results I needed /User/loginname/
then i gives : 
Open On-Chip Debugger 0.9.0-gd4b7679 (2015-06-10-19:16)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.1A.00FB
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21j18.cpu: hardware has 4 breakpoints, 2 watchpoints
cursor blinking / nothing happens on board

valeros, 2016-08-01:
Nice, OpenOCD has recognised your board. Let's try this from the project's root:
```
/User/loginname/.platformio/packages/tool-openocd/bin/openocd -d2 -s /User/loginname/.platformio/packages/tool-openocd/share/openocd/scripts -f /User/loginname/.platformio/packages/tool-openocd/share/openocd/scripts/board/atmel_samd21_xplained_pro.cfg -c "program {{.pioenvs/samd21_xpro/firmware.bin}} verify reset; shutdown"
```

deladriere, 2016-08-01:
Open On-Chip Debugger 0.9.0-gd4b7679 (2015-06-10-19:16)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Error: unable to open CMSIS-DAP device 0x3eb:0x2111
Error: No Valid JTAG Interface Configured.
MacBook-Pro:SAMD21_XPRO jldeladriere$

deladriere, 2016-08-01:
After unplugged it works ! (I guess It was stuck from previous command)

        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.1A.00FB
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477

Info : at91samd21j18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread
xPSR: 0x61000000 pc: 0x0000019c msp: 0x20008000
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21J18A (256KB Flash, 32KB RAM)
wrote 16384 bytes from file .pioenvs/samd21_xpro/firmware.bin in 2.230784s (7.172 KiB/s)
** Programming Finished **
** Verify Started **
verified 6052 bytes in 0.533150s (11.085 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

valeros, 2016-08-01:
Is the board blinking as you expected?

deladriere, 2016-08-01:
yes !

valeros, 2016-08-01:
OK, thanks for the testing, we will add OpenOCD as the uploader in next release.

deladriere, 2016-08-01:
cool !

deladriere, 2016-08-01:
can we dream of using OpenOCD to debug also ?

valeros, 2016-08-01:
Yes, we are planning to add the debug feature in PlatformIO 3.0

valeros, 2016-08-02:
Hi @deladriere! I've just committed a fix for your board. Could you please install PlatformIO from dev branch and try to program your board?
```
pip install https://github.com/platformio/platformio/archive/develop.zip
platformio run -t upload
```
Thanks!

deladriere, 2016-08-04:
upgraded to 2.11.2
IDE 1.4.0
uploading stuck after 
Open On-Chip Debugger 0.9.0-gd4b7679 (2015-06-10-19:16)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
openocd.bin: option requires an argument -- c
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.1A.00FB
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21j18.cpu: hardware has 4 breakpoints, 2 watchpoints

nothing happens
if I press reset on the board 

Polling target at91samd21j18.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Polling target at91samd21j18.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 300ms

endless loop

deladriere, 2016-08-04:
Just in case upload works fine on Arduino IDE with opencd

uploads with 

Users/***/Library/Arduino15/packages/arduino/tools/openocd/0.9.0-arduino/bin/openocd -d2 -s /Users/j***/Library/Arduino15/packages/arduino/tools/openocd/0.9.0-arduino/share/openocd/scripts/ -f /Users/***/Library/Arduino15/packages/atmel-samd21-xpros/hardware/samd/0.2.0/variants/atmel_samd21_xpro_v1/openocd_scripts/variant_atmel_edbg.cfg -c telnet_port disabled; program {{/var/folders/v5/zmn08th96gz7w0g4sb04v4wc0000gn/T/build91ce36a0224b18fec82b5f9926025751.tmp/Blink.ino.bin}} verify reset 0x00002000; shutdown 

Pio with

/Users/***/.platformio/packages/tool-openocd/bin/openocd" -d2 -f atmel_samd21_xplained_pro.cfg -s /Users/***/.platformio/packages/tool-openocd/share/openocd/scripts -s /Users/***/.platformio/packages/tool-openocd/sha
re/openocd/scripts/board -c "telnet_port disabled; program {{.pioenvs/samd21_xpro/firmware.bin}} verify reset shutdown"

deladriere, 2016-08-04:
I think the ; is missing after the reset keyword

valeros, 2016-08-04:
Thanks @deladriere, please switch to development branch in PlatformIO IDE:
_Menu: PlatformIO > Settings > PlatformIO IDE > Use development version of PlatformIO_
_Menu: PlatformIO > Upgrade PlatformIO_

deladriere, 2016-08-04:
done : _Menu: PlatformIO > Settings > PlatformIO IDE > Use development version of PlatformIO_

I guess you mean Update platform and libraries

doesn't look to be updating Ide still 1.4.0

upload still stuck at 
Info : at91samd21j18.cpu: hardware has 4 breakpoints, 2 watchpoints

ivankravets, 2016-08-04:
> I guess you mean Update platform and libraries

Upgrade.... Not update.

deladriere, 2016-08-04:
I did too :
```
 You're up-to-date!
 PlatformIO 2.11.2 is currently the newest version available.
```

Still same (ps also needs to kill opencd.bin between test to be able to re upload)

ivankravets, 2016-08-04:
Hm... Please open PlatformIO IDE Terminal and run `pip install -U https://github.com/platformio/platformio/archive/develop.zip`. The version will be the same, but issue should be fixed.

deladriere, 2016-08-04:
same 
still waiting after
Info : at91samd21j18.cpu: hardware has 4 breakpoints, 2 watchpoints
I notice the command now ends with 
`verify reset ; shutdown`
should it not be 
`verify reset; shutdown`

deladriere, 2016-08-04:
fyi
Arduino uploads fine but if I copy the upload command in the Terminal it stucks at the same place

/Users/***/Library/Arduino15/packages/arduino/tools/openocd/0.9.0-arduino/bin/openocd -d2 -s /Users/***/Library/Arduino15/packages/arduino/tools/openocd/0.9.0-arduino/share/openocd/scripts/ -f /Users/***/Library/Arduino15/packages/atmel-samd21-xpros/hardware/samd/0.2.0/variants/atmel_samd21_xpro_v1/openocd_scripts/variant_atmel_edbg.cfg -c telnet_port disabled; program {{/var/folders/v5/zmn08th96gz7w0g4sb04v4wc0000gn/T/build91ce36a0224b18fec82b5f9926025751.tmp/Blink.ino.bin}} verify reset 0x00002000; shutdown 

Info : at91samd21j18.cpu: hardware has 4 breakpoints, 2 watchpoints

valeros, 2016-08-04:
Hm, this issue occurs only in MacOS for some reason.
Could you please try to remove "-d2" debug flag from the command?
```
"/Users/user/.platformio/packages/tool-openocd/bin/openocd" -f atmel_samd21_xplained_pro.cfg -s /Users/user/.platformio/packages/tool-openocd/share/openocd/scripts -s /Users/user/.platformio/packages/tool-openocd/share/openocd/scripts/board -c "telnet_port disabled; program {{.pioenvs/samd21_xpro/firmware.bin}} verify reset ; shutdown"
```

deladriere, 2016-08-04:
the command you gave me 3 days ago works

/Users/****/.platformio/packages/tool-openocd/bin/openocd -d2 -s /Users/****/.platformio/packages/tool-openocd/share/openocd/scripts -f /Users/****/.platformio/packages/tool-openocd/share/openocd/scripts/board/atmel_samd21_xplained_pro.cfg -c "program {{.pioenvs/samd21_xpro/firmware.bin}} verify reset; shutdown"

the command you just gave me works also if I remove some double quote

/Users/****/.platformio/packages/tool-openocd/bin/openocd -f atmel_samd21_xplained_pro.cfg -s /Users/****/.platformio/packages/tool-openocd/share/openocd/scripts -s /Users/****/.platformio/packages/tool-openocd/share/openocd/scripts/board -c "telnet_port disabled; program {{.pioenvs/samd21_xpro/firmware.bin}} verify reset ; shutdow"

valeros, 2016-08-04:
@deladriere Sorry to bother you again, I've just committed some changes so could you try to flash your board with latest development branch?
```
pip install -U https://github.com/platformio/platformio/archive/develop.zip
```

deladriere, 2016-08-05:
It works now !
Thx a lot 
In fact today I tried it on an Arduino Zero because I didn't have the Samd21_xpro with me. 
Error was solved only after patching to the latest development branch as requested
I guess it should work too on the Samd21_xpro as they have both the same Atmel EDBG interface
I will test this later.

BRW it's cool to be able to code in mbed  on Arduino Zero too (^_^)

closed 2016-08-08 by ivankravets

ivankravets, 2016-08-08:
@deladriere PlatformIO 3.0 has been merged into `develop` branch where this issue is fixed. Could you try to re-install the latest development version and re-test it? Thanks.

http://docs.platformio.org/en/latest/installation.html#development-version

deladriere, 2016-08-23:
Tested with IDE 1.4.0 and CLI 3.0.0a6 : it works fine thx !

ivankravets, 2016-08-23:
@deladriere Thanks 👍

commit 855c28d9 (2016-08-02): Fix firmware uploading to Atmel SAMD21-XPRO board using ARM mbed framework // Resolve #732