platformio/platformio-core #736: NodeMCU builds fail after 2.11.1 upgrade

webmonger, 2016-07-28:
[ x ] Bug report. 

------------------------------------------------------------------

### Configuration

**Operating system**: OS X - El Capitan

**PlatformIO Version** (`platformio --version`): PlatformIO, version 2.11.1

### Description of problem
Building any code against a NodeMCU board fails with generic C++ errors e.g.

scons: *** [.pioenvs/nodemcu/src/main.o] Error 1
In file included from .pioenvs/nodemcu/FrameworkArduino/Esp.cpp:22:0:
.pioenvs/nodemcu/FrameworkArduino/flash_utils.h:28:19: error: 'uint32_t' was not declared in this scope
int SPIEraseBlock(uint32_t block);
^

#### Steps to Reproduce

1. Using Atom Menu > PlatformIO > Project Examples > Selected project > espressif/esp8266-webserver
2. Check to make sure platformio.ini contains:
[env:nodemcu]
platform = espressif
framework = arduino
board = nodemcu
3. PlatformIO > Build
4. Build fails

### Actual Results
```ini
Processing nodemcu (platform: espressif, board: nodemcu, framework: arduino)
--------------------------------------------------------------------------------
xtensa-lx106-elf-ar rcs .pioenvs/nodemcu/libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/src/main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-

sections -fdata-sections -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=021101 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DARDUINO=20300
-U__STRICT_ANSI__ -I/Users/jc/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I/Users/jc/.platformio/packages/framework-arduinoesp
ressif/tools/sdk/lwip/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant -Isrc src/main.cpp
xtensa-lx106-elf-ranlib .pioenvs/nodemcu/libFrameworkArduinoVariant.a
src/main.cpp: In function 'void setup()':
src/main.cpp:12:11: error: 'LED_BUILTIN' was not declared in this scope
pinMode(LED_BUILTIN, OUTPUT);
^
src/main.cpp:12:24: error: 'OUTPUT' was not declared in this scope
pinMode(LED_BUILTIN, OUTPUT);
^
src/main.cpp:12:30: error: 'pinMode' was not declared in this scope
pinMode(LED_BUILTIN, OUTPUT);
^
src/main.cpp: In function 'void loop()':
src/main.cpp:18:16: error: 'LED_BUILTIN' was not declared in this scope
digitalWrite(LED_BUILTIN, HIGH);
^

src/main.cpp:18:29: error: 'HIGH' was not declared in this scope
digitalWrite(LED_BUILTIN, HIGH);
^
src/main.cpp:18:33: error: 'digitalWrite' was not declared in this scope
digitalWrite(LED_BUILTIN, HIGH);
^
src/main.cpp:20:13: error: 'delay' was not declared in this scope
delay(1000);
^
src/main.cpp:22:29: error: 'LOW' was not declared in this scope
digitalWrite(LED_BUILTIN, LOW);
^
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/FrameworkArduino/Esp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4
 -ffunction-sections -fdata-sections -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=021101 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DAR
[... 509 lines cut ...]
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (mdns.begin("esp8266", WiFi.localIP())) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}
```

### Additional info

ivankravets, 2016-07-28:
I've just tested on my macOS and didn't find any problems. Could you try to remove `~/.platformio` folder and try again?

`rm -rf ~/.platformio`

webmonger, 2016-07-28:
Thanks, that worked

closed 2016-07-28 by webmonger