platformio/platformio-core #737: Platformio doesn't fetch samd core files for Servo Library

bhavyangupta, 2016-07-28:
### Configuration

**Operating system**:
Ubuntu 14.04

**PlatformIO Version** (`platformio --version`):
version 2.11.2.dev2

### Description of problem
I am trying to build some code with the Servo library for Sparkfun's samd21 mini breakout board.
However, the code fails to compile. The errors I get are listed below:

### Actual Results

```bash
In file included from src/blink_main.cpp:2:0:
.pioenvs/zero/Servo/Servo.h:67:2: error: #error "This library only supports boards with an AVR or SAM processor."
#error "This library only supports boards with an AVR or SAM processor."
^
scons: *** [.pioenvs/zero/src/blink_main.o] Error 1
===================================================== [ ERROR ] Took 1.46 seconds =====================================================
```

**The content of `platformio.ini`:**
```ini
[env:zero]
platform=atmelsam
framework=arduino
board=sparkfun_samd21_mini_usb
build_flags=-I include/ -I include/ros_lib 
lib_use=Wire, Servo```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include  <Servo.h>
void setup() {
}
void loop() {
}
```

### Additional info

The contents of the Servo directory for my environment are:
``` bash
Servo/
├── avr
│   └── ServoTimers.h
├── sam
│   └── ServoTimers.h
└── Servo.h
2 directories, 3 files
```
I know  that this is not correct since arduino 1.6.9 libraries contains a samd/ folder as well.
Is this an issue with my workspace or the tool?

ivankravets, 2016-07-28:
@valeros please re-pack Arduino framework for SAM and fix this library `framework-arduinosam/libraries/Servo`. Need to remove `Servo/src/avr` folder.

@bhavyangupta this issue is fixed in PlatformIO 3.0. It has C Processor behaviour and understands macros. Temporary solution for you is to remove manually `~/.platformio/packages/framework-arduinosam/libraries/Servo/src/avr` folder.

bhavyangupta, 2016-07-28:
I removed the avr folder as directed, but the code doesn't compile and gives the same error. Are there any other things that I can try?

ivankravets, 2016-07-28:
Sorry, that is not a bug. 

@valeros please ignore my post above. Code compiles very well and AVR code is under `#ifdef`.

ivankravets, 2016-07-28:
The issue is linked with wrong board manifest. 

Temporary solution is to add SAM define. We will fix it in the next release:
```ini
[env:sparkfun_samd21_mini_usb]
platform = atmelsam
framework = arduino
board = sparkfun_samd21_mini_usb
build_flags = -DARDUINO_ARCH_SAM
```

ivankravets, 2016-07-28:
@valeros should we add `-DARDUINO_ARCH_SAM` for all `SAMD` based boards?

bhavyangupta, 2016-07-28:
Thank you for responding so promptly!  I really appreciate your help. I added the build switch, it now gives me the following errors:

```
[Thu Jul 28 15:04:26 2016] Processing zero (platform: atmelsam, framework: arduino, board: sparkfun_samd21_mini_usb, lib_use: Wire, Servo, build_flags: -I include/ -I include/ros_lib -DARDUINO_ARCH_SAM)
---------------------------------------------------------------------------------------------------------------------------------------
....
.pioenvs/zero/Servo/sam/ServoTimers.h:52:29: error: 'TC1' was not declared in this scope
......
.pioenvs/zero/Servo/sam/ServoTimers.h:59:29: error: 'TC1' was not declared in this scope
......
.pioenvs/zero/Servo/sam/ServoTimers.h:66:29: error: 'TC1' was not declared in this scope
......
.pioenvs/zero/Servo/sam/ServoTimers.h:73:29: error: 'TC0' was not declared in this scope
......
.pioenvs/zero/Servo/sam/ServoTimers.h:80:29: error: 'TC0' was not declared in this scope
......
.pioenvs/zero/Servo/sam/Servo.cpp:79:9: error: 'union Tc' has no member named 'TC_CHANNEL'
..
.pioenvs/zero/Servo/sam/Servo.cpp:81:13: error: 'union Tc' has no member named 'TC_CHANNEL'
..
.pioenvs/zero/Servo/sam/Servo.cpp:81:43: error: 'TC_CCR_SWTRG' was not declared in this scope
..
.pioenvs/zero/Servo/sam/Servo.cpp:90:13: error: 'union Tc' has no member named 'TC_CHANNEL'
..

```

ivankravets, 2016-07-28:
@bhavyangupta please remove this `lib_use = Wire, Servo`

bhavyangupta, 2016-07-28:
I did that and there is no change in the build output. Same errors.

ivankravets, 2016-07-28:
Could you try my `platformio.ini` above? https://github.com/platformio/platformio/issues/737#issuecomment-236006485

valeros, 2016-07-28:
HI @bhavyangupta !
Looks like the implementation for samd MCUs is missed in this library.
Could you please try to use [this version](https://github.com/arduino-libraries/Servo)? Just unzip it to the `lib` folder in your project. Does it work now?

bhavyangupta, 2016-07-28:
@ivankravets 
Here's the platformio.ini I tried ( no changes to output):

```
[env:sparkfun_samd21_mini_usb]
platform=atmelsam
framework=arduino
board=sparkfun_samd21_mini_usb
# build_flags=-I include/ -I include/ros_lib -DSERIAL_BUFFER_SIZE=10000
# build_flags=-I include/ -I include/ros_lib
build_flags=-I include/ -I include/ros_lib -DARDUINO_ARCH_SAM
# lib_use=Wire, Servo
```

@valeros 
I agree with your observation. However, when I put the Servo folder from the repo in the lib folder, the errors are still there. I can see that the tool has downloaded the Servo library and ignored the one the lib folder. Here's the directory tree for your reference:

```bash
.
├── include
│   └── ros_lib
├── lib
│   ├── readme.txt
│   └── Servo
├── lo
├── log
├── main.cpp
├── platformio.ini
└── src
    ├── blink_main.cpp
    ├── duration.cpp
    └── time.cpp
5 directories, 8 files

sparkfun_samd21_mini_usb/
├── FrameworkArduino
├── FrameworkArduinoVariant
├── FrameworkCMSISInc
├── FrameworkDeviceInc
├── libFrameworkArduino.a
├── libFrameworkArduinoVariant.a
├── Servo
└── src
6 directories, 2 files
```

valeros, 2016-07-28:
Hm, OK let's replace the old `Servo` lib with new one here `~/.platformio/packages/framework-arduinosam/libraries/Servo/`

bhavyangupta, 2016-07-28:
Tried that. No change in the output. Is there a way to disable the downloading of Servo library? I can see that it gets the library from somewhere after I remove the Servo from the .pioenvs/sparkfun_samd21_mini_usb/ folder.

ivankravets, 2016-07-28:
@bhavyangupta please read warning block http://docs.platformio.org/en/stable/projectconf.html#envs-dir

@bhavyangupta Could you share somewhere full project that doesn't work?

bhavyangupta, 2016-07-28:
@ivankravets  [Here](https://github.com/bhavyangupta/platformio_scratchpad) you go.

I was deleting the folder to "check" if it actually fetches a new Servo library if one is already there in the ~/.platformio/packages/framework-arduinosam/libraries/Servo/ path.

ivankravets, 2016-07-28:
Sorry, that was my mistake. Don't need to use `build_flags= -DARDUINO_ARCH_SAM`. The `Servo` library contains own SAMD implementation. 

Just place [Servo](https://github.com/arduino-libraries/Servo) library to `lib` folder of the project.

### `platformio.ini`
```ini
[env:sparkfun_samd21_mini_usb]
platform=atmelsam
framework=arduino
board=sparkfun_samd21_mini_usb
# build_flags=-I include/ -I include/ros_lib -DSERIAL_BUFFER_SIZE=10000
# build_flags=-I include/ -I include/ros_lib
#build_flags= -DARDUINO_ARCH_SAM
# lib_use=Wire, Servo
src_filter = +<blink_main.cpp>
```

P.S: Be careful, I filtered source files just to the one file (because I don't have `ros_lib` headers)

### Structure
```
tree -L 3 .
├── lib
│   ├── Servo-master
│   │   ├── README.adoc
│   │   ├── examples
│   │   ├── keywords.txt
│   │   ├── library.properties
│   │   └── src
│   └── readme.txt
├── platformio.ini
└── src
    ├── blink_main.cpp
    ├── duration.cpp
    └── time.cpp

5 directories, 8 files
```

ivankravets, 2016-07-28:
@valeros please update Servo library in `frameworksam` package. It should contain SAMD implementation.

bhavyangupta, 2016-07-28:
It works! Thanks a lot!

ivankravets, 2016-07-28:
Thanks a lot that reported.

Please don't close this issue. @valeros will update library and close it.

valeros, 2016-07-29:
Updated!

closed 2016-07-29 by valeros