platformio/platformio-core #739: Platformio erases bootloader on samd21

bhavyangupta, 2016-08-01:
**Operating system**:
Ubuntu 14.04

**PlatformIO Version** (`platformio --version`):
version 2.11.2.dev2

### Description of problem
I uploaded some code on [Sparkfun's samd21 mini breakout board](https://www.sparkfun.com/products/13664) and the tool erased the bootloader. The
device does not run the code nor does it accept code anymore.

**The content of `platformio.ini`:**
```ini
[env:zero]
platform=atmelsam
framework=arduino
board=sparkfun_samd21_mini_usb
# Make sure RingBuffer.h doesn't have SERIAL_BUFFER_SIZE defined
build_flags = -I include/ -I include/ros_lib  -DSERIAL_BUFFER_SIZE=5000 

### Additional info

The code being uploaded was fairly large and the "blink" example uploaded correctly without issues. This problem was replicated on two separate boards.

valeros, 2016-08-01:
Hi @bhavyangupta !
I've just uploaded a simple blink project to `sparkfun_samd21_mini_usb` and it works well.
Does your code work as expected in Arduino IDE? I don't think we could completely erase the bootloder, but could you try to press the Reset button two times very quickly?

bhavyangupta, 2016-08-01:
Hi @valeros 

I agree, a simple blink code uploads well on my system as well. I have tried pressing the reset button twice and the device doesn't enter the bootloader mode anymore. Since this has happened with two boards, I think there is some issue with the tool. I am currently testing if the IDE is able to upload the same code to a new board that "bricked" the other two boards.

valeros, 2016-08-01:
Hm, that's not good. Where can I find a project to reproduce the issue?

bhavyangupta, 2016-08-02:
I just verified that the code uploads correctly when I use a Windows pc and upload via platformio's Atom integration. When I try to upload the same code on linux it gives "device not found " error. Is there a reason why same code would upload correctly via windows but not through linux?

ivankravets, 2016-08-02:
@bhavyangupta have you installed https://github.com/platformio/platformio/blob/develop/scripts/99-platformio-udev.rules ?

bhavyangupta, 2016-08-02:
@ivankravets The udev rules are installed. I tried uploading code from a Windows machine and OSx and it works perfectly with Atom integration. On my laptop, it fails when I use the IDE and the CLI with following error message:

```
[Tue Aug  2 18:03:02 2016] Processing zero (platform: atmelsam, board: sparkfun_samd21_mini_usb, framework: arduino, build_flags: -I include/ -I include/ros_lib)
----------------------------------------------------------------------------------------------------------------------------------------
MethodWrapper(["upload"], [".pioenvs/zero/firmware.bin"])
Looking for upload port/disk...

Warning! Please install `99-platformio-udev.rules` and check that your board's PID and VID are listed in the rules.
https://raw.githubusercontent.com/platformio/platformio/develop/scripts/99-platformio-udev.rules

Auto-detected: /dev/ttyACM0
BeforeUpload(["upload"], [".pioenvs/zero/firmware.bin"])
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Waiting for the new upload port...
"/home/bhavya/.platformio/packages/tool-bossac/bossac" --info --port "ttyACM0" --erase --write --verify --reset --debug -U true .pioenvs/zero/firmware.bin
No device found on ttyACM0
scons: *** [upload] Error 1
===================================================== [ ERROR ] Took 3.39 seconds =====================================================
```
Another observation is that the code **does** upload successfully if I try multiple times. This seems like the delay between `Waiting for the new upload port` and the `bossac` command isn't long enough so the new device doesn't enumerate sometimes - causing upload failure.

ivankravets, 2016-08-08:
When you see "No device found on ttyACM0", could show me an output of `pio serialports list`?

bhavyangupta, 2016-08-09:
@ivankravets  Here you go:
```
/dev/ttyACM0
------------
Hardware ID: USB VID:PID=1B4F:0D21 LOCATION=1-3
Description: ttyACM0
```

ivankravets, 2016-08-15:
Could you try the latest development version?

```
pip install -U https://github.com/platformio/platformio/archive/feature/issue-739-samd21.zip
```

Thanks.

ivankravets, 2016-08-25:
@bhavyangupta do you have any news? Could you try https://github.com/platformio/platformio/issues/739#issuecomment-239869318?

bhavyangupta, 2016-08-25:
@ivankravets  Sorry I haven't been able to check this out yet, however, I have not had the problem ever again - and the tool has been working perfectly since then. Should I close this issue? I will re - open if the problem re-appears?

ivankravets, 2016-08-25:
Yep, please close it and reopen when will have problems.

bhavyangupta, 2016-08-25:
Thanks a lot for the help along the way guys! This is a great tool and has made my life so much easier!

closed 2016-08-25 by bhavyangupta