platformio/platformio-core #740: Libraries in lib folder aren't recognised

ChrisAlphabet, 2016-08-05:
**Operating system**:
Windows 10

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.0.0.dev19

### Description of problem
The project specific libraries in the lib folder are no longer recognised. If I move them to where I have defined ExternalLibs that are not project specific they are recognised.

ivankravets, 2016-08-05:
Run `pio run --verbose`. Do you see some "yellow warnings"?

ChrisAlphabet, 2016-08-07:
Nah, no yellow warnings. The only yellow text is "compilation terminated" after I get the fatal error saying "No such file or directory"

Reverting back to PlatformIO version 2.11.2 works but I'm trying to get unit testing up and running

ChrisAlphabet, 2016-08-08:
Interestingly, if I create another project and copy everything into lib it works but it doesn't find the external libraries.

When I add the external library directory to the project configuration file it finds the external libraries but not the ones in lib.

Am I doing something wrong when creating library directories? Will version 3 work differently to version 2?

ivankravets, 2016-08-08:
Yes, PlatformIO 3 has big difference in comparison to PlatformIO 2. We will write "migration" section in the docs. 

Shortly, you need http://docs.platformio.org/en/feature-platformio-30/projectconf.html#lib-extra-dirs

ChrisAlphabet, 2016-08-08:
Ok, great. What do I have to change in the platformio.ini file to use lib_extra_dirs?

This is what I tried.

Cheers

```[env:nodemcu]
platform = espressif
framework = arduino
board = nodemcu

[platformio]
lib_extra_dirs = ../../../ExternalLibs

ivankravets, 2016-08-08:
This is normal that it doesn't work. The `lib_extra_dirs` option should be applied to `[env:]` group.

closed 2016-08-08 by ivankravets

ChrisAlphabet, 2016-08-15:
Hi, I know this tread is closed but moving the `lib_extra_dirs` option to the `[:env]` group didn't work with the way I was specifying the path. 

To get it to work I needed to put the full path

`[:env]`
`lib_extra_dirs = C:/full/path/to/library/folder`

instead of 
`[:env]`
`lib_extra_dirs = ../../library`

Cheers!

ivankravets, 2016-08-15:
@ChrisAlphabet please open separate issue with that. Thanks.

commit 80c7a8d1 (2016-08-08): Warn about unknown options in project configuration file // Resolve #740
commit 0b749c38 (2016-08-08): Warn about unknown options in project configuration file // Resolve #740