platformio/platformio-core #741: Problems with installer .exe

maichaell, 2016-08-05:
Hi,

I'm having problems installing the .exe from downloads section. Here's the log:

```
2016-08-05 09:39:59> Program: Starting Squirrel Updater: --install .
2016-08-05 09:39:59> Program: Starting install, writing to D:\Users\lainemi\AppData\Local\SquirrelTemp
2016-08-05 09:39:59> Program: About to install to: D:\Users\lainemi\AppData\Local\atom
2016-08-05 09:39:59> CheckForUpdateImpl: Failed to load local releases, starting from scratch: System.IO.DirectoryNotFoundException: Could not find a part of the path 'D:\Users\lainemi\AppData\Local\atom\packages\RELEASES'.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share)
   at Squirrel.Utility.LoadLocalReleases(String localReleaseFile)
   at Squirrel.UpdateManager.CheckForUpdateImpl.<CheckForUpdate>d__3f.MoveNext()
2016-08-05 09:39:59> CheckForUpdateImpl: Reading RELEASES file from D:\Users\lainemi\AppData\Local\SquirrelTemp
2016-08-05 09:39:59> CheckForUpdateImpl: First run or local directory is corrupt, starting from scratch
2016-08-05 09:39:59> ApplyReleasesImpl: Writing files to app directory: D:\Users\lainemi\AppData\Local\atom\app-1.8.0
2016-08-05 09:40:13> IEnableLogger: Failed to install package to app dir: System.AggregateException: One or more errors occurred. ---> System.IO.IOException: Access to the path 'resources' is denied.
   at System.IO.DirectoryInfo.MoveTo(String destDirName)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<>c__DisplayClassb4.<>c__DisplayClassb9.<>c__DisplayClassbb.<installPackageToAppDir>b__b2(DirectoryInfo subdir)
   at Squirrel.Utility.<>c__DisplayClass18`1.<>c__DisplayClass1a.<ForEachAsync>b__17()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Squirrel.Utility.<>c__DisplayClass1e`1.<>c__DisplayClass20.<<ForEachAsync>b__1d>d__22.MoveNext()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<>c__DisplayClassb4.<>c__DisplayClassb9.<installPackageToAppDir>b__b1(DirectoryInfo ld)
   at Squirrel.EnumerableExtensions.ForEach[TSource](IEnumerable`1 source, Action`1 onNext)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<>c__DisplayClassb4.<<installPackageToAppDir>b__ad>d__bd.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Squirrel.Utility.<LogIfThrows>d__50`1.MoveNext()
---> (Inner Exception #0) System.IO.IOException: Access to the path 'resources' is denied.
   at System.IO.DirectoryInfo.MoveTo(String destDirName)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<>c__DisplayClassb4.<>c__DisplayClassb9.<>c__DisplayClassbb.<installPackageToAppDir>b__b2(DirectoryInfo subdir)
   at Squirrel.Utility.<>c__DisplayClass18`1.<>c__DisplayClass1a.<ForEachAsync>b__17()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Squirrel.Utility.<>c__DisplayClass1e`1.<>c__DisplayClass20.<<ForEachAsync>b__1d>d__22.MoveNext()<---

2016-08-05 09:40:13> Unhandled exception: System.AggregateException: One or more errors occurred. ---> System.AggregateException: One or more errors occurred. ---> System.IO.IOException: Access to the path 'resources' is denied.
   at System.IO.DirectoryInfo.MoveTo(String destDirName)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<>c__DisplayClassb4.<>c__DisplayClassb9.<>c__DisplayClassbb.<installPackageToAppDir>b__b2(DirectoryInfo subdir)
   at Squirrel.Utility.<>c__DisplayClass18`1.<>c__DisplayClass1a.<ForEachAsync>b__17()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Squirrel.Utility.<>c__DisplayClass1e`1.<>c__DisplayClass20.<<ForEachAsync>b__1d>d__22.MoveNext()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<>c__DisplayClassb4.<>c__DisplayClassb9.<installPackageToAppDir>b__b1(DirectoryInfo ld)
   at Squirrel.EnumerableExtensions.ForEach[TSource](IEnumerable`1 source, Action`1 onNext)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<>c__DisplayClassb4.<<installPackageToAppDir>b__ad>d__bd.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Squirrel.Utility.<LogIfThrows>d__50`1.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
[... 170 lines cut ...]
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<ApplyReleases>d__7a.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Squirrel.UpdateManager.<FullInstall>d__f.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   at Squirrel.Update.Program.<Install>d__38.MoveNext()
---> (Inner Exception #0) System.IO.IOException: Access to the path 'resources' is denied.
   at System.IO.DirectoryInfo.MoveTo(String destDirName)
   at Squirrel.UpdateManager.ApplyReleasesImpl.<>c__DisplayClassb4.<>c__DisplayClassb9.<>c__DisplayClassbb.<installPackageToAppDir>b__b2(DirectoryInfo subdir)
   at Squirrel.Utility.<>c__DisplayClass18`1.<>c__DisplayClass1a.<ForEachAsync>b__17()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Squirrel.Utility.<>c__DisplayClass1e`1.<>c__DisplayClass20.<<ForEachAsync>b__1d>d__22.MoveNext()<---
<---

```

ivankravets, 2016-08-05:
Please close IDE and download the latest Atom from http://atom.io. Install it.

Something is wrong with Atom's "auto-update" feature.

MrMartinZockt, 2016-08-05:
I had the same Problem last night. After I closed my antivirus program (Kaspersky) everything worked fine.

AtomeSetup.exe alone had the same problem

ivankravets, 2016-08-05:
Try to fully uninstall Atom:
- https://community.platformio.org/t/platformio-is-not-recognized-as-an-internal-or-external-command/431/20?u=ivankravets

Then, please use the latest from http://atom.io , go to "Settings > Install" and search for "platformio-ide", install it.

maichaell, 2016-08-05:
This is my first time install ever. I do have an antivirus program but I can't shut it down (company policy)

I'll try installing http://atom.io , "platformio-ide" - thanks.

ivankravets, 2016-08-25:
Please reopen if you need a help

closed 2016-08-25 by ivankravets