platformio/platformio-core #743: platformio test command fails

ohhorob, 2016-08-08:
[ ] Bug report. If you’ve found a bug, please provide information below.

**Operating system**: Mac OS X 10.11.5

**PlatformIO Version** (`platformio --version`): `PlatformIO, version 3.0.0a1`

### Description of problem

Running the test command fails with this backtrace:

```
$ platformio test
Error: Traceback (most recent call last):
  File "/Users/home/pio3dev/lib/python2.7/site-packages/platformio/__main__.py", line 105, in main
    cli(None, None, None)
  File "/Users/home/pio3dev/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/home/pio3dev/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(cox)
  File "/Users/home/pio3dev/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/home/pio3dev/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/home/pio3dev/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/home/pio3dev/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/home/pio3dev/lib/python2.7/site-packages/platformio/commands/test.py", line 48, in cli
    assert check_project_envs(project_dir, environment)
  File "/Users/home/pio3dev/lib/python2.7/site-packages/platformio/commands/run.py", line 299, in check_project_envs
    if not config.sections():
AttributeError: 'str' object has no attribute 'sections'
```

**The content of `platformio.ini`:**
```ini
#
# Project Configuration File
#
# A detailed documentation with the EXAMPLES is located here:
# http://docs.platformio.org/en/latest/projectconf.html
#

# A sign `#` at the beginning of the line indicates a comment
# Comment lines are ignored.
#[platformio]
#

# Simple and base environment
# [env:mybaseenv]
# platform = %INSTALLED_PLATFORM_NAME_HERE%
# framework =
# board =
#
# Automatic targets - enable auto-uploading
# targets = upload

[env:teensy31]
platform = teensy
framework = arduino
board = teensy31
test_dir = digihead_tests

# Pull in the PlatformIO registered libraries
#   12 => Adafruit-ST7735
#   13 => Adafruit-GFX
#  129 => Encoder
#  291 => ClickEncoder ? not sure if I actually need this yet
# 1106 => Bounce2
#  416 => TinyGPS
lib_install = 12, 13, 129, 291, 1106, 416
#lib_extra_dirs = /Users/home/.platformio/lib
#lib_deep_search = true
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

This error occurs with and without the `test_dir = digihead_tests` property. And it doesn't seem to matter where I put it. I've tried under a top-level `[platformio]` stanza, and under the `[env:teensy31]` one.

ivankravets, 2016-08-08:
Thanks a lot for the report and that help us to test PIO3.

closed 2016-08-08 by ivankravets

ohhorob, 2016-08-08:
Cheers.

I pulled the update from develop, and `platformio test` compiles cleanly with
```ini
[platformio]
test_dir = digihead_tests
```

I don't have a teensy with me to verify running the tests yet.

ivankravets, 2016-08-08:
> I pulled the update from develop

Next time please try `pio upgrade` command, it should upgrade current `develop` to the "latest" `develop` automatically.

commit 70df551e (2016-08-08): Fix platformio test command // Resolve #743