platformio/platformio-core #746: Some analog ports missing on atmelsam (adafruit_feather_m0_usb)

bcr, 2016-08-12:
What kind of issue is this?

- [X] Bug report. If you’ve found a bug, please provide information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
```
Blakes-MacBook:~ blake$ uname -a
Darwin Blakes-MacBook 15.6.0 Darwin Kernel Version 15.6.0: Thu Jun 23 18:25:34 PDT 2016; root:xnu-3248.60.10~1/RELEASE_X86_64 x86_64
Blakes-MacBook:~ blake$ sw_vers -productVersion
10.11.6
```
**PlatformIO Version** (`platformio --version`):
```
Blakes-MacBook:~ blake$ platformio --version
PlatformIO, version 2.11.2
```
### Description of problem
There appear to be some special analog input ports that are not defined in the PlatformIO environment, but are defined in the Adafruit boards for the Arduino IDE. I am specifically using the Adafruit Feather M0 Bluefruit LE and the desired pin is analog pin 7 which is defined in the Arduino IDE as `A7`.

The Adafruit boards I installed were from https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

#### Steps to Reproduce

1. Attempt to access analog pin `A7`. For instance, `analogRead(A7);`

### Actual Results

```
src/main.cpp:69:16: error: 'A7' was not declared in this scope
[ ERROR ] Took 6.95 seconds
```

### Expected Results
```
[SUCCESS] Took 7.87 seconds
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
#
# PlatformIO Project Configuration File
#
# Please make sure to read documentation with examples first
# http://docs.platformio.org/en/stable/projectconf.html
#
[env:adafruit_feather_m0_usb]
platform = atmelsam
framework = arduino
board = adafruit_feather_m0_usb
```

**Source file to reproduce issue:**
```cpp
void setup() {
  // put your setup code here, to run once:
  analogRead(A7);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Additional info

[Adafruit Feather M0 Bluefruit LE tutorial](https://learn.adafruit.com/adafruit-feather-m0-bluefruit-le/overview)

bcr, 2016-08-12:
I haven't started digging into the code yet, this may have been an issue with the Arduino code at one point based on [this forum post](https://forum.arduino.cc/index.php?topic=363393.0)

ivankravets, 2016-08-18:
Please move this issue to appropriate repository https://github.com/platformio/platform-atmelsam/issues and close. Thanks.

bcr, 2016-08-18:
Moved to platformio/platform-atmelsam#2

closed 2016-08-18 by bcr