platformio/platformio-core #748: __AVR_ATxmega* is defined on non xmega boards

jrobeson, 2016-08-14:
### Configuration

**Operating system**: Fedora 24
**PlatformIO Version** (`platformio --version`): 2.11.2 (via platformio IDE)

```sh
platformio lib list
[ ID  ] Name             Compatibility         "Authors": Description
---------------------------------------------------------------------------------------------------------------------------------------
[ 433 ] RF24             arduino, atmelavr, atmelsam, teensy "TMRh20": Optimized High Speed Driver for nRF24L01(+) 2.4GHz Wireless Transceiver
``
### Description of problem
When trying to build the [Getting Started](https://github.com/TMRh20/RF24/blob/master/examples/GettingStarted/GettingStarted.ino) example it fails.

#### Steps to Reproduce

1. platformio lib install 433
2. copy the getting started example into main.cpp
3. add `#include "Arduino.h"` to the top
4. run `platformio run`

### Actual Results

```sh
 ~/w/l/s/arduino_target  platformio run
[Thu Aug 11 23:43:37 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/main.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility -Isrc src/main.cpp
avr-g++ -o .pioenvs/uno/RF24_ID433/RF24.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/RF24.cpp
avr-gcc -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c
avr-g++ -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp
avr-gcc -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.c
In file included from .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.h:20:0,
from .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:9:
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.h:83:1: error: 'PORT_t' does not name a type
PORT_t * GPIO_getPort(int pinnum, uint8_t * pin_bm);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In static member function 'static void GPIO::open(int, int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In function 'int read(int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In static member function 'static void GPIO::write(int, int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In function 'int read(int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:35:1: warning: control reaches end of non-void function [-Wreturn-type]
}
[... 64 lines cut ...]
```

### Expected Results

It should build successfully
### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
framework = arduino
board = uno

```

**Source file to reproduce issue:**
https://raw.githubusercontent.com/TMRh20/RF24/master/examples/GettingStarted/GettingStarted.ino

### Additional info

The example compiles correctly in the arduino ide.

Comment from maintainer: https://github.com/TMRh20/RF24/issues/275
Other person who has the same build issue:  https://community.platformio.org/t/show-or-get-older-versions-of-libraries/569

MitsuhideA, 2016-08-14:
I think someone had the same problem with another library : https://github.com/platformio/platformio/issues/298#issuecomment-144711939

It will be resolved with PlatformIO 3.0 thanks to this #289

ivankravets, 2016-08-14:
@valeros please update `library.json` following PlatformIO 3.0 guide for this library http://platformio.org/lib/show/433/RF24

ivankravets, 2016-08-15:
Please switch to http://docs.platformio.org/en/stable/installation.html#development-version

Uninstall previous version of RF24

closed 2016-08-15 by ivankravets