platformio/platformio-core #751: avrdude binary downloaded by platformio relies on specific libncurses.so version

thozza, 2016-08-18:
### Configuration

**Operating system**:
Fedora 24

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.11.2

### Description of problem
It is not possible to upload compiled Arduino sketch to the board, because avrdude binary downloaded by platformio relies on specific libncurses.so version.

#### Steps to Reproduce

1. compile any Arduino sketch on Fedora 24 and try to upload it to the board

### Actual Results
Upload fails with:
```
Auto-detected: /dev/ttyACM0
"/home/xxx/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328p -C "/home/xxx/.platformio/packages/tool-avrdude/avrdude.conf" -c arduino -b 115200 -P "/dev/ttyACM0" -D -U flash:w:.pioenvs/arduino_uno/firmware.hex:i
/home/xxx/.platformio/packages/tool-avrdude/avrdude: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
scons: *** [upload] Error 127
```

### Expected Results
upload should pass successfully

### Additional info
platformio should ideally use the tools available on the system and only if some are missing try to download the binaries. Another option would be for platformio to compile the tools from sources or to link the binaries statically. Downloading binary which was dynamically linked with some library is error prone and not very portable.

On my system I have:
```
/usr/lib/libncurses.so.6
/usr/lib/libncurses.so.6.0
```

ivankravets, 2016-08-18:
Please move this issue to appropriate repository https://github.com/platformio/platform-atmelavr/issues. Thanks.

thozza, 2016-08-18:
https://github.com/platformio/platform-atmelavr/issues/6

closed 2016-08-18 by thozza

thozza, 2016-08-18:
@ivankravets no problem :smile: