platformio/platformio-core #752: Framework Incompatible Libraries when Native Unit Testing

ChrisAlphabet, 2016-08-19:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** 3.0.0a5

### Description of problem

I can't get the library files to be recognised when running local unit tests. 

If I run `pio test -e local -v` I get all my libraries listed as "Framework Incompatible Library"

How can I get this to run properly?

ivankravets, 2016-08-19:
Are these ignored libraries compatible with native/desktop development platform?

ChrisAlphabet, 2016-08-20:
I'm not really sure. What makes a library compatible?

I had a weird experience where I made a dummy class with some dummy functions and it was compatible. I then copied everything from one of my classes in and everything was compatible. But I haven't been able to reliably reproduce it. 

Can you point me where to dive deeper into assessing library compatibility for native/desktop?

I couldnt get native or windows_x86 to work.

Cheers

ivankravets, 2016-08-21:
We have cross example for Unit Testing https://github.com/platformio/platformio-examples/tree/develop/unit-testing/calculator

Does it work for you?

P.S: Docs http://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-compat-mode

ChrisAlphabet, 2016-08-21:
Yeah, the examples work for me without any problems

ivankravets, 2016-08-22:
Could you explain how to reproduce this issue? Please note that you can't compile non-host compatible libraries with `native` development platform. I mean, that you can't test "Arduino"-based library on the desktop machine. We don't emulate hardware.

ivankravets, 2016-08-25:
Please reopen if you need a help

closed 2016-08-25 by ivankravets