platformio/platformio-core #753: Can't Ignore More Than One Test

ChrisAlphabet, 2016-08-19:
Windows 10, version 3.0.0a5
Can't ignore multiple tests at one time.

For example

`[env:mybaseenv]`
`test_ignore = test_a, test_b, test_c`

doesn't ignore any tests

and

`[env:mybaseenv]`
`test_ignore = test_a`
`test_ignore = test_b`

only ignore test_b

It would be great to just run a single test rather than the whole test suite.

Thanks in advance!

ivankravets, 2016-08-19:
See docs http://docs.platformio.org/en/latest/projectconf.html#test-ignore

Should I change separator to `,` to avoid the same issue?

ChrisAlphabet, 2016-08-19:
Ahh, I thought I checked the docs but I must've skimmed. My Bad! :)  

I think a comma separator is a bit more intuitive, and it's consistent with

`lib_extra_dirs = path/to/lib1, path/to/lib2`

Cheers!

ivankravets, 2016-08-19:
Thanks, I'll change :)

ivankravets, 2016-08-19:
I see that you use `pio test`. Could you share some feedback? What we should improve? Thanks.

ChrisAlphabet, 2016-08-20:
Sure thing. I'm a pretty big fan of `pio test`! 

One of the things that could be improved is how much text is dumped into the console. If you run a bunch of tests the console fills up pretty quickly and you can't scroll back up to see which test failed. 

At the moment I'm putting all the console data in a `.txt` file so I can get everything using `pio test -e myenv -v > test_results.txt` which works ok cause i can `ctrl+f failed` and find what failed.

Maybe a default test logging feature would be handy? What do you think?

Another thing is that running an entire test suite takes a pretty long time. I'm using the `nodemcu` board so it might just be a slow platform but getting the whole process down to a few seconds would be great.

Also, I am having trouble getting native tests running. (see #752 ) Maybe a bit more documentation on getting native tests running would be helpful.

Other than that, it works great! I've been using Unity for the test framework but will be looking at getting Ceedling going probs next week-ish so I'll post any more feedback either in the community forum.

Love your work! Anything I can do to help out with PIO? Would love to contribute, where can I get started?

Cheers :)

ivankravets, 2016-08-21:
Thanks! 👍 

Please test new PlatformIO 3.0.0a6. Is it better?

closed 2016-08-21 by ivankravets

ChrisAlphabet, 2016-08-21:
That is much better! 

For anyone curious 

`pio test` will just print building, uploading, testing and test results

`pio test -v` will print everything to the console

Thanks!

commit f5b4cb0a (2016-08-21): Refactor "test_ignore" to accepts test names separated with comma // Resolve #753