platformio/platformio-core #754: Cannot compile using Arduino Library inside lib/

eni23, 2016-08-20:
Hello

Im trying to use Platformio with this Library https://github.com/sumotoy/TFT_ST7735/tree/1.0p1
Unfortunately the compilation fails. The Library works without problems under Arduino.

### Configuration

**Operating system**: Arch Linux amd64
**PlatformIO Version**:  2.11.2

### Description of problem

Compilation fails with a lot of errors:
```
.pioenvs/nodemcuv2/TFT_ST7735/TFT_ST7735.h:79:32: fatal error: _fonts/defaultFont.c: No such file or directory
#include _ST7735_DEF_FONT_PATH
^
compilation terminated.
```

#### Steps to Reproduce

1. clone https://github.com/sumotoy/TFT_ST7735 to lib/-directory and checkout `1.0p1`
2. Include the library according to an example
3. Try to compile

### Actual Results

`compilation terminated`

### Expected Results

Compilation works

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nodemcuv2]
platform = espressif
framework = arduino
board = nodemcuv2
```

**Source file to reproduce issue:**
```cpp
#include <SPI.h>
#include <TFT_ST7735.h>
#define __CS  16  //(D0)
#define __DC  5   //(D1)
#define __RST 4   //(D2)
TFT_ST7735 tft = TFT_ST7735(__CS, __DC, __RST);
void setup() {
  tft.begin();
  tft.print("hello platformio");
}
void loop(void) { }
```

### Additional info

The trick with the following inside platformio does not works:
```ini
build_flags = -Ilib/TFT_ST7735
lib_ignore = TFT_ST7735
```

It compiles but cannot link the final hex:
```
.pioenvs/nodemcuv2/src/tmp_ino_to.o:(.text.setup+0x10): undefined reference to `TFT_ST7735::begin(bool)'
.pioenvs/nodemcuv2/src/tmp_ino_to.o: In function `setup':
tmp_ino_to.cpp:(.text.setup+0x3e): undefined reference to `TFT_ST7735::begin(bool)'
.pioenvs/nodemcuv2/src/tmp_ino_to.o:(.text.startup._GLOBAL__sub_I_tft+0x1c): undefined reference to `TFT_ST7735::TFT_ST7735(unsigned char, unsigned char, unsigned char)'
.pioenvs/nodemcuv2/src/tmp_ino_to.o: In function `_GLOBAL__sub_I_tft':
tmp_ino_to.cpp:(.text.startup._GLOBAL__sub_I_tft+0x46): undefined reference to `TFT_ST7735::TFT_ST7735(unsigned char, unsigned char, unsigned char)'
```

if i comment out the `lib_ignore` i get different errors (a lot of them):
```
.pioenvs/nodemcuv2/TFT_ST7735/_fonts/Terminal_9.c:39:1: error: unknown type name '_smCharType'
const _smCharType image_data_Terminal_9_0x20[1]
^
.pioenvs/nodemcuv2/TFT_ST7735/_fonts/Terminal_9.c:45:1: error: unknown type name 'tImage'
const tImage Terminal_9_0x20
```

What can i do? I would love to use platformio for my project, but i need that library.

ivankravets, 2016-08-20:
Could you try http://docs.platformio.org/en/stable/installation.html#development-version ?

eni23, 2016-08-20:
Yes, with Platformio `version 3.0.0a5` it compiles.

ivankravets, 2016-08-21:
I close this issue because it could not be resolved in PlatformIO 2.0 and is fixed in PlatformIO 3.0 that is planned for release next week. Thanks.

closed 2016-08-21 by ivankravets

eni23, 2016-08-21:
For me this is fine. I like the output of pio 3.0 a lot since i'm using Makefiles.Thanks for your time and this awesome project.