platformio/platformio-core #755: ESP8266 OTA Does not work

williamesp2015, 2016-08-20:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform. All issues related to Development Platform should be reported to appropriate repository. Search it using link below
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

williamesp2015, 2016-08-20:
I am using simple OTA Arduino sketch and after upload by serial port working fine, then I added upload_port = 192.168.0.255 to platformio.ini and then upload but terminal shows:
MethodWrapper(["upload"], [".pioenvs\nodemcuv2\firmware.bin"])
Looking for upload port/disk...
Manually specified: 192.168.1.159
"c:\users\am\.atom\packages\platformio-ide\penv\scripts\python.exe" "C:\Users\AM\.platformio\packages\framework-arduinoespressif\tools\esp
ota.py" --debug --progress -i 192.168.1.159 -f .pioenvs\nodemcuv2\firmware.bin
23:10:04 [DEBUG]: Options: {'esp_ip': '192.168.1.159', 'host_port': 19448, 'image': '.pioenvs\\nodemcuv2\\firmware.bin', 'host_ip': '0.0.0
.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
23:10:04 [INFO]: Starting on 0.0.0.0:19448
23:10:04 [INFO]: Upload size: 247600
23:10:04 [INFO]: Sending invitation to: 192.168.1.159
23:10:14 [ERROR]: No Answer
scons: *** [upload] Error 1

ivankravets, 2016-08-20:
> then I added upload_port = 192.168.0.255 to platformio.ini
 
and 

> Manually specified: 192.168.1.159

:shipit:

ivankravets, 2016-08-20:
Please read docs for [Over-the-Air (OTA) update](http://docs.platformio.org/en/stable/platforms/espressif.html#over-the-air-ota-update).

If you still have problems, please reopen this issue.

closed 2016-08-20 by ivankravets

williamesp2015, 2016-08-20:
@ivankravets  thank you. I have done extensive study on OTA including that doc but the error is unusual. My PlatformIo is uptodate (2.11.2) and my ESP IP is 192.168.1.159

ivankravets, 2016-08-20:
Please fill all information here https://github.com/platformio/platformio/issues/755#issue-172284659

We need `platformio.ini`

williamesp2015, 2016-08-21:
Operating system: Windows 10
IDE:1.4.0  CLI:2.11.2 
platformio.ini
platform = espressif
framework = arduino
board = nodemcuv2
upload_port = 192.168.1.159

williamesp2015, 2016-08-21:
I tested https://github.com/AdySan/ESPSocket and its OT working.