platformio/platformio-core #756: ESP8266 OTA Does not work

williamesp2015, 2016-08-20:
I am using simple OTA Arduino sketch and after upload by serial port working fine and my chip ID is 192.168.1.159, then I added upload_port = 192.168.1.159 to platformio.ini and then upload but terminal shows:
MethodWrapper(["upload"], [".pioenvs\nodemcuv2\firmware.bin"])
Looking for upload port/disk...
Manually specified: 192.168.1.159
"c:\users\am.atom\packages\platformio-ide\penv\scripts\python.exe" "C:\Users\AM.platformio\packages\framework-arduinoespressif\tools\esp
ota.py" --debug --progress -i 192.168.1.159 -f .pioenvs\nodemcuv2\firmware.bin
23:10:04 [DEBUG]: Options: {'esp_ip': '192.168.1.159', 'host_port': 19448, 'image': '.pioenvs\nodemcuv2\firmware.bin', 'host_ip': '0.0.0
.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
23:10:04 [INFO]: Starting on 0.0.0.0:19448
23:10:04 [INFO]: Upload size: 247600
23:10:04 [INFO]: Sending invitation to: 192.168.1.159
23:10:14 [ERROR]: No Answer
scons: *** [upload] Error 1

ivankravets, 2016-08-20:
Something is wrong with your network/firewall. Please read all comments here http://www.penninkhof.com/2015/12/1610-over-the-air-esp8266-programming-using-platformio/

williamesp2015, 2016-08-20:
Thank you but Windows 10 Firewall is not the issue.

ivankravets, 2016-08-21:
Did you upload this sketch [BasicOTA.in](https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino) before testing OTA?

ivankravets, 2016-08-25:
Please reopen if you need a help

closed 2016-08-25 by ivankravets