platformio/platformio-core #759: ValueError: Expecting Property Name on compile

cosmikwolf, 2016-08-22:
I can successfully install pio 3.0.0a6, but when I try to build, i get the following errors:

```
MacPro:flxs1 tenkai$ pio run -v
[Mon Aug 22 13:37:51 2016] Processing teensy31 (platform: teensy, board_f_cpu: 96000000, build_flags: , board: teensy31, framework: arduino)

ValueError: Expecting property name: line 11 column 1 (char 291):
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 125:

env.SConscript("$BUILD_SCRIPT")
File "/Users/tenkai/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return SConscript(self.fs, *files, **substkw)
File "/Users/tenkai/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec file in call_stack[-1].globals
File "/Users/tenkai/.platformio/platforms/teensy/builder/main.py", line 199:
target_elf = env.BuildProgram()
File "/Users/tenkai/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 61:
deplibs = env.BuildDependentLibraries("$PROJECTSRC_DIR")
File "/Users/tenkai/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 514:
lib_builders = env.GetLibBuilders()
File "/Users/tenkai/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 489:
lb = LibBuilderFactory.new(env, join(libs_dir, item))
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 50:
obj = getattr(sys.modules[name], clsname)(env, path)
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 85:
self.manifest = manifest if manifest else self.loadmanifest()
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 393:
manifest = util.load_json(join(self.path, "library.json"))
File "/Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages/platformio/util.py", line 132:
return json.load(f)
File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 290:
**kw)
File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 338:
return defaultdecoder.decode(s)
File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 366:
obj, end = self.raw_decode(s, idx=_w(s, 0).end())
File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382:
obj, end = self.scan_once(s, idx)
============================================================================================== [ERROR] Took 0.49 seconds ==============================================================================================
```

This is compiling for Teensy version 1.28 (latest teensy version on pio)

ivankravets, 2016-08-22:
Please reinstall (repeat `pip install`) PlatformIO 3.0 http://docs.platformio.org/en/stable/installation.html#development-version

You should see warning/yellow message about a broken library. This library contains `library.json`. Could you share it here? Thanks.

cosmikwolf, 2016-08-22:
I uninstalled and reinstalled the dev version and I did not get any errors about any libraries.

```
  Successfully uninstalled platformio-2.11.2
MacPro:flxs1 tenkai$ pip install -U https://github.com/platformio/platformio/archive/develop.zip
Collecting https://github.com/platformio/platformio/archive/develop.zip
  Downloading https://github.com/platformio/platformio/archive/develop.zip
     - 11.8MB 4.9MB/s
Requirement already up-to-date: bottle<0.13 in /Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages (from platformio==3.0.0a6)
Requirement already up-to-date: click<6,>=5 in /Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages (from platformio==3.0.0a6)
Requirement already up-to-date: lockfile<0.13,>=0.9.1 in /Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages (from platformio==3.0.0a6)
Requirement already up-to-date: requests<3,>=2.4.0 in /Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages (from platformio==3.0.0a6)
Requirement already up-to-date: semantic_version>=2.5.0 in /Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages (from platformio==3.0.0a6)
Requirement already up-to-date: colorama in /Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages (from platformio==3.0.0a6)
Requirement already up-to-date: pyserial<4 in /Users/tenkai/.atom/packages/platformio-ide/penv/lib/python2.7/site-packages (from platformio==3.0.0a6)
Installing collected packages: platformio
  Running setup.py install for platformio ... done
Successfully installed platformio-3.0.0a6

```

ivankravets, 2016-08-22:
I mean "embedded" library, not system. Please process/run your project.

cosmikwolf, 2016-08-22:
I ran the project, and i get a bunch of errors, but the first one is this one:
`
Skip library with broken manifest: /Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Artnet
`

here is the library.json file:
```
MacPro:Artnet tenkai$ pwd
/Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Artnet
MacPro:Artnet tenkai$ cat library.json 
{
"name": "ArtNet",
"frameworks": "Arduino",
"keywords": "ArtNet, LED, LEDs, DMX, E1.31", 
"description": "Control LEDs over Ethernet using the Artnet Protocol",
"url": "http://forum.pjrc.com/threads/24688-Artnet-to-OctoWS2811/page10",
"authors":
[
{
        "name": "Nathanaël Lécaudé",
},
{
        "name": "mortonkopf",
}
]
"repository":
{
    "name": "git",
    "url": "https://github.com/natcl/Artnet"
}
}

```

ivankravets, 2016-08-22:
I got, the problem is linked with unicode

ivankravets, 2016-08-22:
Where did you find this manifest? It's broken.

I've just fixed it:
```json
{  
  "name":"ArtNet",
  "frameworks":"Arduino",
  "keywords":"ArtNet, LED, LEDs, DMX, E1.31",
  "description":"Control LEDs over Ethernet using the Artnet Protocol",
  "url":"http://forum.pjrc.com/threads/24688-Artnet-to-OctoWS2811/page10",
  "authors":[  
    {  
      "name":"Nathanaël Lécaudé"
    },
    {  
      "name":"mortonkopf"
    }
  ],
  "repository":{  
    "name":"git",
    "url":"https://github.com/natcl/Artnet"
  }
}
```

Please use JSON validator https://jsonformatter.curiousconcept.com

closed 2016-08-22 by ivankravets

cosmikwolf, 2016-08-22:
It is what was installed with the framework-teensy that gets installed with platformio.

cosmikwolf, 2016-08-22:
ah! no, I figued that framework-arduinoteensy was re-installed with platformio. This is the framework-arduinoteensy that comes from the PJRC installer. I had updated it to 1.29 to try and troubleshoot another issue. So the broken unicode must come from PJRC.

cosmikwolf, 2016-08-22:
I deleted framework-arduinoteensy and re-ran my project and now it seems to work. Lots of other new errors, but I need to work through them.
Thanks!

ivankravets, 2016-08-22:
It's not broken unicode. The unicode is OK. The problem with JSON format. It's broken.

I've just checked `framework-arduinoteensy` and don't see this library:
```
tree -L 1 /Users/ikravets/.platformio/packages/framework-arduinoteensy/libraries
/Users/ikravets/.platformio/packages/framework-arduinoteensy/libraries
├── Audio
├── Bounce
├── Bridge
├── EEPROM
├── Esplora
├── Ethernet
├── Firmata
├── LiquidCrystal
├── RobotIRremote
├── Robot_Control
├── Robot_Motor
├── SD
├── SPI
├── Scheduler
├── Servo
├── SoftwareSerial
├── Stepper
├── TFT
├── Temboo
├── USBHost
├── WiFi
└── Wire
```

Is this new library located in 1.29?

ivankravets, 2016-08-22:
> Lots of other new errors, but I need to work through them.

This is PlatformIO related issues or yours?

cosmikwolf, 2016-08-22:
I think it might be platform related, but I am not sure:
````
In file included from /Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Audio/src/Audio.h:16:0,
from /Users/tenkai/Desktop/Projects/flxs1/src/flxs1.ino:1:
/Users/tenkai/.platformio/packages/framework-arduinoteensy/libraries/Audio/src/DAC.h:21:16: error: expected ')' before '*' token
DACClass(Dacc *_dac, uint32_t _dacId, IRQn_Type _isrId) :

````

cosmikwolf, 2016-08-22:
Lots of other errors, but that seems to be the root

cosmikwolf, 2016-08-22:
And yes, that new library is in 1.29

ivankravets, 2016-08-22:
😆 
```
name=Audio
version=1.0
author=Arduino
maintainer=Arduino <info@arduino.cc>
sentence=Allows playing audio files from an SD card. For Arduino DUE only.
paragraph=With this library you can use the Arduino Due DAC outputs to play audio files.<br />The audio files must be in the raw .wav format.
category=Signal Input/Output
url=http://arduino.cc/en/Reference/Audio
architectures=sam
```

What does this library do in Teensy framework? Could you compile it with Teeensyduino?

cosmikwolf, 2016-08-22:
This library is developed by PJRC, it utilizes a DSP instruction set within the Arm Cortex M4 for audio processing capabilities. It compiles fine with Teensyduino.

ivankravets, 2016-08-22:
I got you. Thanks a lot! 👍 

Please move this https://github.com/platformio/platformio/issues/759#issuecomment-241557365 to https://github.com/platformio/platform-teensy/issues 

This repository is intended for PlatformIO Core.

commit b3abee85 (2016-08-22): Ignore broken libraries; show link to the docs with lib_compat_mode // Resolve #759