platformio/platformio-core #762: Failure installing development version of Platformio CLI on Mac OS X El Capitan

blackketter, 2016-08-23:
------------------------------------------------------------------

### Configuration

**Operating system**:
Mac OS X El Capitan 10.11.6

**PlatformIO Version** (`platformio --version`):
3.0/Development

### Description of problem
Trying to install the development version of Platformio CLI on Mac but I get a failure trying to remove an older version of pyserial.

#### Steps to Reproduce

1. `brew install python`
2. `pip install -U https://github.com/platformio/platformio/archive/develop.zip`

### Actual Results

```
lucky13:Downloads dean$ pip install -U https://github.com/platformio/platformio/archive/develop.zip
Collecting https://github.com/platformio/platformio/archive/develop.zip
  Downloading https://github.com/platformio/platformio/archive/develop.zip
     - 11.8MB 295kB/s
Requirement already up-to-date: bottle<0.13 in /usr/local/lib/python2.7/site-packages (from platformio==3.0.0a7)
Requirement already up-to-date: click<6,>=5 in /usr/local/lib/python2.7/site-packages (from platformio==3.0.0a7)
Requirement already up-to-date: lockfile<0.13,>=0.9.1 in /usr/local/lib/python2.7/site-packages (from platformio==3.0.0a7)
Requirement already up-to-date: requests<3,>=2.4.0 in /usr/local/lib/python2.7/site-packages (from platformio==3.0.0a7)
Requirement already up-to-date: semantic_version>=2.5.0 in /usr/local/lib/python2.7/site-packages (from platformio==3.0.0a7)
Requirement already up-to-date: colorama in /usr/local/lib/python2.7/site-packages (from platformio==3.0.0a7)
Collecting pyserial<4 (from platformio==3.0.0a7)
  Using cached pyserial-3.1.1-py2.py3-none-any.whl
Installing collected packages: pyserial, platformio
  Found existing installation: pyserial 2.7
    Uninstalling pyserial-2.7:
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/local/lib/python2.7/site-packages/pip/commands/install.py", line 317, in run
    prefix=options.prefix_path,
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_set.py", line 736, in install
    requirement.uninstall(auto_confirm=True)
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_install.py", line 742, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_uninstall.py", line 115, in remove
    renames(path, new_path)
  File "/usr/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 267, in renames
    shutil.move(old, new)
  File "/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 303, in move
    os.unlink(src)
OSError: [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/pyserial-2.7-py2.7.egg'

```

ivankravets, 2016-08-23:
That is not a bug. You are going to install 2 copies of PlatformIO.

Please don't install development version of PlatformIO to global system. Use virtual environment instead.
```
sudo pip install -U virtualenv
virtualenv ~/.poienv
source ~/.poienv/bin/activate
pip install -U https://github.com/platformio/platformio/archive/develop.zip
pio --version
# deactivate - if you need to exit from env
```

Next time if you need to work with development version, you need just 1 command: `source ~/.poienv/bin/activate`

ivankravets, 2016-08-25:
Please reopen if you need a help

closed 2016-08-25 by ivankravets