platformio/platformio-core #764: Upload to Arduino Mega 2560 fail

sumarouno, 2016-08-29:
------------------------------------------------------------------

### Configuration

**Operating system**:
LSB Version:	:core-4.1-ia32:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 24 (Twenty Four)
Release:	24
Codename:	TwentyFour

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.11.2

### Description of problem
I try blink program, build is succesfull but when i try to upload to may atmega, the problem was occurs. It show me "/home/sumarsono/.platformio/packages/tool-avrdude/avrdude: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory"

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
#
# PlatformIO Project Configuration File
#
# Please make sure to read documentation with examples first
# http://docs.platformio.org/en/stable/projectconf.html
#
[env:megaatmega2560]
platform = atmelavr
framework = arduino
board = megaatmega2560

upload_port = /dev/ttyUSB0
```

**Source file to reproduce issue:**
```cpp
/*tes blink platformIO*/
/*Sumar*/

#include "Arduino.h"

void setup()
{
pinMode(LED_BUILTIN, OUTPUT);

}

void loop()
{
digitalWrite(LED_BUILTIN, HIGH);
delay(1000);
digitalWrite(LED_BUILTIN, LOW);
delay(1000);
}

```

### Additional info
here the log:
eforeUpload(["upload"], [".pioenvs/megaatmega2560/firmware.hex"])
Looking for upload port/disk...
Manually specified: /dev/ttyUSB0
"/home/sumarsono/.platformio/packages/tool-avrdude/avrdude" -v -p atmega2560 -C "/home/sumarsono/.platformio/packages/tool-avrdude/avrdude.conf" -c wiring -b 115200 -P "/dev/ttyUSB0" -D -U flash:w:.pioenvs/megaatmega2560/firmware.hex:i
/home/sumarsono/.platformio/packages/tool-avrdude/avrdude: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory
scons: *** [upload] Error 127
========================= [ ERROR ] Took 0.95 seconds =========================

ivankravets, 2016-08-29:
Please check 
* https://forums.opensuse.org/showthread.php/446927-missing-library-libtinfo-so-5
* http://forum.arduino.cc/index.php?topic=402333.0

ivankravets, 2016-09-04:
Please reopen if you need a help

closed 2016-09-04 by ivankravets

sumarouno, 2016-09-12:
Work like a charm, thank you.