platformio/platformio-core #765: error: missing terminating " character

jourjine, 2016-08-29:
------------------------------------------------------------------

### Configuration

**Operating system**:
MAC OSX
**PlatformIO Version** (`platformio --version`):
PlatformIO, version 2.11.2
### Description of problem
not build it
on original Arduino IDE no problem

#### Steps to Reproduce

1. just build
2.
3.

### Actual Results
/Users/KOCMOHABT/Documents/Arduino/Scetches/Wi-Fi-Sensor-master/esp_sensor/src/esp_sensor.ino:268:1: warning: missing t
erminating " character [enabled by default]
"<SCRIPT>\
^
/Users/KOCMOHABT/Documents/Arduino/Scetches/Wi-Fi-Sensor-master/esp_sensor/src/esp_sensor.ino:268:1: error: missing ter
minating " character

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
[env:esp8285]
platform = espressif
framework = arduino
board = esp8285

**Source file to reproduce issue:**
```
#include <ESP8266WiFi.h>
//#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>

extern "C" {
#include "user_interface.h"
}

#include <PubSubClient.h>
#include <Wire.h>
#include "SPI.h"
#include <BH1750.h>
#include <NTPClient.h>
#include "json_config.h"
#include <ArduinoJson.h>
#include <WiFiUdp.h>

#if defined(UART_ON)
#include "MY_ESP_UART.h"
Espuart Uart;
#endif

JsonConf JConf;

#if defined(DHT_ON)
#include "DHT.h"
#define DHTTYPE DHT22
DHT dht(atoi(JConf.dht_pin), DHTTYPE);
int errorDHTdata = 0;  // количество ошибок чтения датчика DHT
#endif

#if defined(BME280_ON)
[... 3255 lines cut ...]
      }

      if (millis() - publishTimer >= atoi(JConf.publish_delay) * 1000){
        publishTimer = millis();
        MqttPubData();
      }
    }
  }

  #ifdef UART_ON
  Uart.serialEvent();
  #endif

  FadeSwitchLoop();

  #ifdef DEBUG
    unsigned long load_time3 = millis() - start_time;
    if (load_time3 > 100){
      Serial.print(F("loop() Load Time:                               ")); Serial.println(load_time3);
    }
  #endif
}
```

### Additional info

ivankravets, 2016-08-29:
Please share here `esp_sensor.ino`

jourjine, 2016-08-29:
original version https://github.com/kmaximv/Wi-Fi-Sensor
little bit corrected my version, add few string because i have error in NTPClient
https://dl.dropboxusercontent.com/u/12954395/esp_sensor.ino

ivankravets, 2016-08-29:
Try to remove Cyrillic comments.

jourjine, 2016-08-29:
removed, still have problem

ivankravets, 2016-08-29:
Where is updated file?

jourjine, 2016-08-29:
[new version](https://dl.dropboxusercontent.com/u/12954395/esp_sensor_1.ino)

jourjine, 2016-08-29:
imho overly long string in quotes. if I it is reduced to a few lines, then it becomes normal

ivankravets, 2016-08-30:
The problem is linked with INO > CPP converter. If you convert your source code to CPP manually, you will not have problems.

In any case, this is a bug and I'm working on it.

ivankravets, 2016-09-01:
Please re-test with http://docs.platformio.org/en/stable/installation.html#development-version

commit bb62444f (2016-09-01): Handle C multiline strings when converting INO to CPP // Resolve #765
closed 2016-09-01 by ivankravets