platformio/platformio-core #766: if I comment on the braces, they are still counted

jourjine, 2016-08-29:
**Operating system**:
Mac OS X 10.11.6
**PlatformIO Version** (`platformio --version`):
2.11.2
### Description of problem
in CPP and INO files
if I comment on the braces, they are still counted in syntax helpers, but no build error

ivankravets, 2016-08-29:
Please provide source code to reproduce.

jourjine, 2016-08-29:
```
BLYNK_CONNECTED() {
  if (isFirstConnect) {                 
    Blynk.syncAll();                   
    isFirstConnect = false;             
  } 
}

```

```
BLYNK_CONNECTED() {
  //if (isFirstConnect) {      **<- to this**           
    Blynk.syncAll();                   
    isFirstConnect = false;             
  } **<- on mouse over this bracer still connected **  
}
```

jourjine, 2016-08-29:
```
BLYNK_CONNECTED() {
  //if (isFirstConnect) {   <- to this              
  if (isFirstConnect) {                 
    Blynk.syncAll();                   
    isFirstConnect = false;             
  }
} <- this
```

highlight mistake

ivankravets, 2016-08-29:
You comment 1 line... What is wrong?

jourjine, 2016-08-29:
[image]
[image]

jourjine, 2016-08-29:
[image]
[image]

ivankravets, 2016-08-30:
Pleas move your issue here https://github.com/atom/language-c/issues

closed 2016-08-30 by ivankravets