platformio/platformio-core #769: Add platform RTL8710 / RTL8711 / RTL8915

kissste, 2016-09-06:
Contains ARM Cortex-M3 processor

joscha, 2017-09-16:
Is there any news on this? Anyone working on it?

Alxandr, 2018-01-01:
$15 bounty added to the issue.

treii28, 2018-04-07:
I would be interested as well. How does one add a bounty?

Alxandr, 2018-04-08:
@treii28 https://www.bountysource.com/issues/37603319-add-platform-rtl8710-rtl8711-rtl8195

treii28, 2018-04-10:
Cool, matching $15 added.

ivankravets, 2018-07-19:
Could someone explain how do you program these chips now? Does Realtek have own SDK/IDE?

mtiutiu, 2018-08-04:
@ivankravets It seems to be supported by this arduino core [Ameba IOT](https://github.com/Ameba8195/Arduino). Works with rtl8710/8195 only and the project seems to be still alive. Regarding the flashing part it works via swd using openocd and a stlink device because it's stm32 based. I gave it a try some time ago and it worked. For openocd integration I used [this](https://github.com/sandeepmistry/rtl8710-openocd). I didn't played too much with it so I cannot offer more details. Hope it helps.

kgunnar, 2018-11-16:
While not immediately noticeable due to their re-branding, it's clear when digging into the specifications but PINE64's $1.99 [PADI IoT Stamp](https://www.pine64.org/?page_id=917) is a RTL8710AF module.

In addition to the Ameba link above providing Arduino support, there is also [RtlDuino](https://github.com/pvvx/RtlDuino).

chris-hatton, 2020-02-14:
@ivankravets Yes, Realtek do provide their '[Standard SDK](https://www.amebaiot.com/en/ameba-sdk-intro/)', based on RTOS, for these RTL8710 / RTL8195 chips.  I am also interested in the potential of interfacing PlatformIO directly with this SDK and not via the Arduino support.
I will happily add US$25 to the bounty for this feature.
Presumably with RTOS and gcc basis for the Realtek 'Ameba' SDK, it will not be an exotic thing to interface with PlatformIO?

ivankravets, 2020-02-14:
Could someone help us with adding new dev-platform? There are a lot of examples how to do it https://github.com/platformio/platform-espressif32

Thanks in advance!

chris-hatton, 2020-04-11:
There is an ongoing effort by [8Devices](https://www.8devices.com/) to bring the [RTL8710](https://en.wikipedia.org/wiki/RTL8710) (['Standard' GCC SDK](https://www.amebaiot.com/en/ameba-sdk-intro/)) to PlatformIO.

See: https://github.com/8devices/platformio-realtek-rtl8710b

benhadad, 2021-02-10:
I am coming back to this board myself as it seems a library based on this board will be useful going forward. For one, you can run rust embedded on it in a FreeRTOS setup.

chris-hatton, 2021-08-23:
I've added a further $50 USD to this issue taking it up to $95 USD
...just in case anyone is on the edge about spending the time...

mviher, 2022-09-02:
Any update on that?

kuba2k2, 2022-09-15:
RTL8195A, RTL8710B, RTL8720C and RTL8720D are totally different chips. Currently, there's an Arduino Core implementation for RTL8710B available in my project: https://github.com/kuba2k2/libretuya
I don't have support for other Realtek chips yet, but it's planned to be available in the future.