platformio/platformio-core #770: Cannot compile nodemcuv2 using version 3.0.0.b13

deladriere, 2016-09-06:
gives this error

collect2: error: ld returned 1 exit status
*** [.pioenvs/nodemcuv2/firmware.elf] Error 1

reverting to 2.11.2 works

ivankravets, 2016-09-06:
How to reproduce this issue? Could you provide test `platformio.ini` and source code?

deladriere, 2016-09-06:
Here is the complete folder

[Blynk.zip](https://github.com/platformio/platformio/files/456391/Blynk.zip)

deladriere, 2016-09-06:
[Blynk.zip](https://github.com/platformio/platformio/files/456395/Blynk.zip)
here is the folder

deladriere, 2016-09-06:
[Blynk.zip](https://github.com/platformio/platformio/files/456400/Blynk.zip)

ivankravets, 2016-09-06:
I've just tried your project and don't see any problems

# platformio.ini

```ini
[env:nodemcuv2]
platform = espressif8266
framework = arduino
board = nodemcuv2
lib_deps = Blynk, SimpleTimer, DallasTemperature, Encoder
```

See output http://pastebin.com/Qs064aru

closed 2016-09-06 by ivankravets

deladriere, 2016-09-06:
added lib_deps = Blynk, SimpleTimer, DallasTemperature, Encoder
to platformio.ini solved it !
very strange

ivankravets, 2016-09-06:
You are going to use a few external libraries in your project and you don't have them. The code could not be built.

deladriere, 2016-09-06:
I does compile fine under 2.11.2

deladriere, 2016-09-06:
now I have a new error
Exception OSError: OSError(2, 'No such file or directory') in <bound method dblite.__del__ of <SCons.dblite.dblite obje
ct at 0x10586f150>> ignored
if I keep ib_deps = Blynk, SimpleTimer, DallasTemperature, Encoder
removed it and now compiles fine  !!!

deladriere, 2016-09-06:
sorry cannot reproduce Exception OSError: OSError(2, 'No such file or directory') in ct at 0x10586f150>> ignored

ivankravets, 2016-09-06:
I don't know what do you do. Try to clean project via `pio run -t clean`. Now, run it again via `pio run`. Do you see any errors?