platformio/platformio-core #771: Can not install library in a path containing diacritics

zgoda, 2016-09-06:
Follow up to https://community.platformio.org/t/impossible-to-download-a-library-linux/701

Unicode path operations in Python2 are allowed as long as arguments to path functions are passed as unicode objects and not as byte strings. I believe the offending line is currently https://github.com/platformio/platformio/blob/develop/platformio/downloader.py#L50 where `join()` is performed and `dest_dir` there is byte string. In this function `self._fname` is unicode and Python tries to decode `dest_dir` into unicode using default ASCII codec.

zgoda, 2016-09-06:
To see this in interactive session ensure your terminal encoding is utf-8:

```
In [7]: os.path.join('one', u'more')
Out[7]: u'one/more'

In [8]: os.path.join('onę', u'more')
---------------------------------------------------------------------------
UnicodeDecodeError                        Traceback (most recent call last)
<ipython-input-8-e9c449c757d9> in <module>()
----> 1 os.path.join('onę', u'more')

/home/jazg/v/coup/lib/python2.7/posixpath.pyc in join(a, *p)
     71             path +=  b
     72         else:
---> 73             path += '/' + b
     74     return path
     75 

UnicodeDecodeError: 'ascii' codec can't decode byte 0xc4 in position 2: ordinal not in range(128)

In [9]: os.path.join('onę', 'more')
Out[9]: 'on\xc4\x99/more'

In [10]: os.path.join(u'onę', u'more')
Out[10]: u'on\u0119/more'
```

Input 7 and 8 is the situation we have currently, input 9 is when both arguments are byte strings and input 10 is when both arguments are unicode.

bricewge, 2016-10-24:
I stumbled upon this issue too when working on a project within this directory `/home/bricewge/projet/esp8266/thermomètre`. I'm using the version `3.1.0`.

Following are the modification I have made to be able to install the rebellious library. Note that I don't know what I'm doing; but now it's working!
``` diff
--- /usr/lib/python2.7/site-packages/platformio/downloader.py
+++ #<buffer downloader.py>
@@ -48,7 +48,7 @@
         self._progressbar = None
         self._destination = self._fname
         if dest_dir:
-            self.set_destination(join(dest_dir, self._fname))
+            self.set_destination(join(dest_dir.decode('utf-8'), self._fname))
 
     def set_destination(self, destination):
         self._destination = destination
```

``` diff
--- /usr/lib/python2.7/site-packages/platformio/managers/package.py
+++ #<buffer package.py>
@@ -216,6 +216,7 @@
 
     def _install_from_tmp_dir(self, tmp_dir, requirements=None):
         tmpmanifest = self.load_manifest(tmp_dir)
+        self.package_dir = self.package_dir.decode('utf-8')
         assert set(["name", "version"]) <= set(tmpmanifest.keys())
         name = tmpmanifest['name']
         pkg_dir = join(self.package_dir, name)
```

zgoda, 2016-10-25:
@bricewge this works because FS paths encoding is utf-8 on your system. In most cases this is safe assumption, at least on Windows and in most modern Linux systems where utf-8 based encodings are default. OSX has long history of broken locales and on Linux one can have multiple encodings in use, ex. for my own Polish I can have both iso-8859-2 and utf-8 encoded paths *at the same time* (of course this will break many things in system but hey, I can do it!). Generally this is rather tricky part.

ivankravets, 2017-11-02:
Thanks for the report!

Please open PIO IDE Terminal and type `pio upgrade`. It should be fixed in 3.5.0b1. Don't forget to restart IDE.

commit 35cbbeb4 (2016-09-12): Fix downloader with Unicode path // Resolve #771
closed 2017-11-02 by ivankravets