platformio/platformio-core #772: I can't update/upgrade Plaformio CLI to version 3.0

profra, 2016-09-07:
OS = Linux Mint
All directories (lib, packages, platforms) in directory .platformio are empty. 
Any advice?

```
fp@fp-VC62B ~/Downloads $ pio --version
PlatformIO, version 3.0.0
fp@fp-VC62B ~/Downloads $ pio update
Platform Manager
================
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 106, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/update.py", line 35, in cli
    pioplus_update()
  File "/usr/local/lib/python2.7/dist-packages/platformio/pioplus.py", line 46, in pioplus_update
    pm = PioPlusPackageManager()
  File "/usr/local/lib/python2.7/dist-packages/platformio/pioplus.py", line 32, in __init__
    "https://dl.platformio.org/packages/manifest.json"])
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/package.py", line 267, in __init__
    os.makedirs(self.package_dir)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/home/fp/.platformio/packages'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/stable/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================

fp@fp-VC62B ~/Downloads $ sudo pio update
Platform Manager
================

Library Manager
===============
fp@fp-VC62B ~/Downloads $ sudo pio upgrade
You're up-to-date!
PlatformIO 3.0.0 is currently the newest version available.
fp@fp-VC62B ~/Downloads $ 
fp@fp-VC62B ~/Downloads $ sudo pip install -U platformio
The directory '/home/fp/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/fp/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:318: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#snimissingwarning.
  SNIMissingWarning
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
Requirement already up-to-date: platformio in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: lockfile<0.13,>=0.9.1 in /usr/local/lib/python2.7/dist-packages (from platformio)
Requirement already up-to-date: requests<3,>=2.4.0 in /usr/local/lib/python2.7/dist-packages (from platformio)
Requirement already up-to-date: bottle<0.13 in /usr/local/lib/python2.7/dist-packages (from platformio)
Requirement already up-to-date: click<6,>=5 in /usr/local/lib/python2.7/dist-packages (from platformio)
Requirement already up-to-date: semantic-version>=2.5.0 in /usr/local/lib/python2.7/dist-packages (from platformio)
Requirement already up-to-date: colorama in /usr/local/lib/python2.7/dist-packages (from platformio)
Requirement already up-to-date: pyserial<4 in /usr/local/lib/python2.7/dist-packages (from platformio)
fp@fp-VC62B ~/Downloads $ 
```

ivankravets, 2016-09-07:
Please don't use `sudo` for PlatformIO commands. Looks like sometimes you ran `pio` with `sudo` and `pio` created root directories under sudo.

Solution

1. `sudo rm -rf ~/.platformio`
2. Use `pio` command WITHOUT `sudo`.

profra, 2016-09-07:
I did it ... I am trying to build something... without succes... see output
```
fp@fp-VC62B ~/opt/aa_myprojects/arduino_01 $ pio run
[Thu Sep  8 00:18:02 2016] Processing pro8MHzatmega328 (platform: atmelavr, upload_port: /dev/ttyUSB1, board: pro8MHzatmega328, framework: arduino)
--------------------------------------------------------------------------------
PlatformManager: Installing atmelavr
Error: Detected unknown package 'atmelavr'
fp@fp-VC62B ~/opt/aa_myprojects/arduino_01 $ platformio init --board pro8MHzatmega328
[API] ConnectionError: hostname 'api.platformio.org' doesn't match 'pioplus.com' (incremented retry: max=5, total=1)
[API] ConnectionError: hostname 'api.platformio.org' doesn't match 'pioplus.com' (incremented retry: max=5, total=2)
[API] ConnectionError: hostname 'api.platformio.org' doesn't match 'pioplus.com' (incremented retry: max=5, total=3)
[API] ConnectionError: hostname 'api.platformio.org' doesn't match 'pioplus.com' (incremented retry: max=5, total=4)
[API] ConnectionError: hostname 'api.platformio.org' doesn't match 'pioplus.com' (incremented retry: max=5, total=5)
Error: [API] Could not connect to PlatformIO Registry Service. Please try later.
fp@fp-VC62B ~/opt/aa_myprojects/arduino_01 $ 
```

ivankravets, 2016-09-07:
Wow, that is interesting! Turn off SSL (this is a new PIO3 feature) temporary.

```
pio settings set disable_ssl true
```

profra, 2016-09-07:
Yupee!!!!  pio run ...SUCCESS!!!! .... BIG THANKS....

ivankravets, 2016-09-07:
1 sec, I'll disable SSL temporary and make quick release

ivankravets, 2016-09-07:
Please upgrade to PlatformIO 3.0.1. I disabled SLL feature.

closed 2016-09-07 by ivankravets

ivankravets, 2016-09-07:
@profra What is your Python version? I found interesting information:

> Python3 and Python 2.7.9+ include native support for SNI in their SSL modules. For information on using SNI with Requests on Python < 2.7.9 refer to this Stack Overflow answer.

See http://docs.python-requests.org/en/master/community/faq/#what-are-hostname-doesn-t-match-errors

profra, 2016-09-07:
@ivankravets  I have Python 2.7.6 but also Python 3.4.3. ... sometimes I have problems with these TWO... I need both... :-)
I am trying to upgrade to 3.0.1 but I see output..
```
fp@fp-VC62B ~/opt/aa_myprojects/esp_01 $ pio upgrade
Please wait while upgrading PlatformIO ...

-----------------
Permission denied
-----------------
You need the `sudo` permission to install Python packages. Try

> sudo pip install -U platformio

WARNING! Don't use `sudo` for the rest PlatformIO commands.

fp@fp-VC62B ~/opt/aa_myprojects/esp_01 $ 

```
May I do that?

For your information... I did upgrade on my NB with W10 evening without any problems (Atom IDE)... and on my home machine with Linux Mint later in the night with above problems (CLI)... I don't see any key to this mystery but you could...

ivankravets, 2016-09-07:
Ok, let me explain in a few words. You installed PlatformIO to system Python sites. You need `sudo` in this case. When I said about "don't use `sudo` for PlatformIO`, I meant for other commands except  `pio upgrade` (if you want to install PIO Globally).

You have a lot better options:

1. Create Python virtual environment and install PlatformIO here. You will not have any problems
2. Install PlatformIO to user's Python sites and add user's bin directory to $PATH. For example,

```
# delete current PIO
sudo pip uninstall platformio

# install new
pip install --user platformio

```
Now, you can make upgrading without `sudo`. Please don't forget about adding `bin` directory to your PATH. I don't know where `pip` will install packages with `--user` option on your machine.

profra, 2016-09-07:
... and on W10 machine I have 2.7.11 and 3.5.1... it's a little brighter :-)

profra, 2016-09-07:
@ivankravets  Thank you for the explanation... now is time for me to go to bed...

ivankravets, 2016-09-07:
@profra the same :) Ukraine. 😊

profra, 2016-09-07:
Czech Republic :blush:

commit ff312f04 (2016-09-07): Disable temporary SSL for PlatformIO services // Resolve #772