platformio/platformio-core #773: 3.0.1 library configuration borked?

seclorum, 2016-09-08:
I'm not sure if this is an issue or just an oversight on my part, but it seems that out of the box, PlatformIO is not producing working project configurations for access to onboard built-in libraries.. or else I'm mis-interpreting the following session, which intention is to simply prove that current PlatformIO is suitable to use the Base64 library - however, a newly configured project seems unable to find the libraries:

```
 seclorum@github:~/ {}
$ pio --version
PlatformIO, version 3.0.1
 seclorum@github:~/ {}
$ mkdir ~/Documents/platformIO/hak/t_proj/ && cd ~/Documents/platformIO/hak/t_proj/
 seclorum@github:~/Documents/platformIO/hak/t_proj {}
$ pio init --board esp12e

The current working directory /Users/seclorum/Documents/platformIO/hak/t_proj will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in /Users/seclorum/Documents/platformIO/hak/t_proj
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Project has been successfully initialized!
Useful commands:
`platformio run` - process/build project from the current directory
`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information
 seclorum@github:~/Documents/platformIO/hak/t_proj {}
$ pio lib install Base64
 seclorum@github:~/Documents/platformIO/hak/t_proj {}
$ pio lib list
Library Storage: /Users/seclorum/Documents/platformIO/hak/t_proj/.piolibdeps
[ ID  ] Name             Compatibility         "Authors": Description
---------------------------------------------------------------------------------------------------------
[ 582 ] Base64           arduino, atmelavr, atmelsam, espressif8266, intel_arc32, microchippic32, nordicnrf51, teensy, timsp430 "Arturo Guadalupi": A Base64 library for Arduino | @0.0.1
 seclorum@github:~/Documents/platformIO/hak/t_proj {}
$ ls -l
total 8
drwxr-xr-x+ 3 j  staff  102 Sep  8 15:50 lib
-rw-r--r--+ 1 j  staff  495 Sep  8 16:04 platformio.ini
drwxr-xr-x+ 3 j  staff  102 Sep  8 16:11 src
 seclorum@github:~/Documents/platformIO/hak/t_proj {}
$ cp ~/.platformio/lib/Base64_ID582/examples/Base64Decode/Base64Decode.ino src/
 seclorum@github:~/Documents/platformIO/hak/t_proj {}
$ pio run -v
[Thu Sep  8 16:11:15 2016] Processing esp12e (platform: espressif8266, build_flags: -Wl,-Tesp8266.flash.4m.ld -lstdc++ -lsupc++  -Ilib/Base64 -lBase64, board: esp12e, framework: arduino)
---------------------------------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o "/Users/seclorum/Documents/platformIO/hak/t_proj/src/Base64Decode.ino.cpp" -x c++ -fpreprocessed -dD -E "/var/folders/48/yjv7rc1x3bd010gv32lkwyzm0000gq/T/tmp5gusvX"
Collected 32 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Base64> v0.0.1 (/Users/seclorum/Documents/platformIO/hak/t_proj/.piolibdeps/Base64_ID582)
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/src/Base64Decode.ino.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030001 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -U__STRICT_ANSI__ -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/lwip/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266 -I/Users/seclorum/.platformio/platforms/espressif8266/builder/lib/Base64 -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/variants/nodemcu -I.piolibdeps/Base64_ID582/src -Isrc src/Base64Decode.ino.cpp
/Users/seclorum/Documents/platformIO/hak/t_proj/src/Base64Decode.ino: In function 'void setup()':
/Users/seclorum/Documents/platformIO/hak/t_proj/src/Base64Decode.ino:33:23: error: 'Base64' was not declared in this scope
char decodedString[decodedLength];
^
xtensa-lx106-elf-gcc -o .pioenvs/esp12e/FrameworkArduino/core_esp8266_i2s.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030001 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -U__STRICT_ANSI__ -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/lwip/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266 -I/Users/seclorum/.platformio/platforms/espressif8266/builder/frameworks/lib/Base64 -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/variants/nodemcu /Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266/core_esp8266_i2s.c
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/FrameworkArduino/core_esp8266_main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030001 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -U__STRICT_ANSI__ -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/lwip/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266 -I/Users/seclorum/.platformio/platforms/espressif8266/builder/frameworks/lib/Base64 -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/variants/nodemcu /Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266/core_esp8266_main.cpp
xtensa-lx106-elf-gcc -o .pioenvs/esp12e/FrameworkArduino/core_esp8266_noniso.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030001 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -U__STRICT_ANSI__ -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/lwip/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266 -I/Users/seclorum/.platformio/platforms/espressif8266/builder/frameworks/lib/Base64 -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/variants/nodemcu /Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266/core_esp8266_noniso.c
xtensa-lx106-elf-gcc -o .pioenvs/esp12e/FrameworkArduino/core_esp8266_phy.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030001 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -U__STRICT_ANSI__ -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/tools/sdk/lwip/include -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266 -I/Users/seclorum/.platformio/platforms/espressif8266/builder/frameworks/lib/Base64 -I/Users/seclorum/.platformio/packages/framework-arduinoespressif/variants/nodemcu /Users/seclorum/.platformio/packages/framework-arduinoespressif/cores/esp8266/core_esp8266_phy.c
xtensa-lx106-elf-gcc -o .pioenvs/esp12e/FrameworkArduino/core_esp8266_postmortem.o -c -std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030001 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -U__STRICT_ANSI__ -I/Users/seclor [... cut ...]
    ; // wait for serial port to connect. Needed for native USB only
  }

  // encoding
  char inputString[] = "QmFzZTY0RGVjb2RlRXhhbXBsZQA=";
  int inputStringLength = sizeof(inputString);

  Serial.print("Input string is:\t");
  Serial.println(inputString);

  Serial.println();

  int decodedLength = Base64.decodedLength(inputString, inputStringLength);
  char decodedString[decodedLength];
  Base64.decode(decodedString, inputString, inputStringLength);
  Serial.print("Decoded string is:\t");
  Serial.println(decodedString);
}

void loop() {

}
```

ivankravets, 2016-09-08:
Please move this issue to https://github.com/agdl/Base64/issues
The problem is linked with library.

[image]

closed 2016-09-08 by ivankravets

seclorum, 2016-09-09:
Doesn't make sense .. been using this library without glitch for months, it has not changed, and I have not updated anything but PlatformIO ..

ivankravets, 2016-09-09:
Are you sure that you have used this library? It also don't compile with Arduino IDE.

seclorum, 2016-09-09:
I'm positive I've used it successfully on my project in the past - I created the above test-case to demonstrate, however. 

What I think is happening is that "Base64.h" is not being found/used in the right stages of compilation in pio, for some reason.

ivankravets, 2016-09-09:
Let's wait an answer from an author of the lib. Please ping me here if this issue relatedto PlatformIO.