platformio/platformio-core #774: SSLError: hostname 'dl.platformio.org' doesn't match 'pioplus.com'

zouden, 2016-09-08:
- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** 3.0.1

### Description of problem

I get the above error when trying to install a library (Homie).

#### Steps to Reproduce

1. pio lib install homie

### Actual Results

Download fails with that error

### Expected Results

Library should install

ivankravets, 2016-09-08:
Fixed in development version. Please upgrade http://docs.platformio.org/en/stable/installation.html#development-version

zouden, 2016-09-08:
Great it works now. Thanks for the extremely fast response!

Palatis, 2016-09-14:
It didn't work for https://platformio.org either...

I think a better way is to get a proper cert...

ivankravets, 2016-09-14:
@Palatis we use HTTPS for all our resources except home page (issue with CDN)

ivankravets, 2016-09-19:
Fixed in PlatformIO 3.1.0

closed 2016-09-19 by ivankravets