platformio/platformio-core #775: after the update, no program using Byink does not compile any new or old

jourjine, 2016-09-10:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform. All issues related to Development Platform should be reported to appropriate repository. Search it using link below
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
MAC OSX
**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.0.1
### Description of problem
after the update no program is compiled, no new and old

#### Steps to Reproduce

1.
2.
3.

### Actual Results
oClientGen<Client> >::processInput()]+0x1c): undefined reference to `BlynkOnConnected'
.pioenvs/espduino/src/main.o: In function `BlynkProtocol<BlynkArduinoClientGen<Client> >::processInput()':
main.cpp:(.text._ZN13BlynkProtocolI21BlynkArduinoClientGenI6ClientEE12processInputEv[BlynkProtocol<BlynkArduinoClientGen<Client> >
::processInput()]+0x122): undefined reference to `BlynkOnConnected'
collect2: error: ld returned 1 exit status
*** [.pioenvs/espduino/firmware.elf] Error 1

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:espduino]
platform = espressif8266
framework = arduino
board = espduino
```

**Source file to reproduce issue:**
```cpp
/**************************************************************
 * Blynk is a platform with iOS and Android apps to control
 * Arduino, Raspberry Pi and the likes over the Internet.
 * You can easily build graphic interfaces for all your
 * projects by simply dragging and dropping widgets.
 *
 *   Downloads, docs, tutorials: http://www.blynk.cc
 *   Blynk community:            http://community.blynk.cc
 *   Social networks:            http://www.fb.com/blynkapp
 *                               http://twitter.com/blynk_app
 *
 * Blynk library is licensed under MIT license
 * This example code is in public domain.
 *
 **************************************************************
 * This example runs directly on ESP8266 chip.
 *
 * You need to install this for ESP8266 development:
 *   https://github.com/esp8266/Arduino
 *
 * Please be sure to select the right ESP8266 module
 * in the Tools -> Board menu!
 *
 * Change WiFi ssid, pass, and Blynk auth token to run :)
 *
 **************************************************************/

#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "YourAuthToken";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "YourNetworkName";
char pass[] = "YourPassword";

void setup()
{
  Serial.begin(9600);
  Blynk.begin(auth, ssid, pass);
}

void loop()
{
  Blynk.run();
}

```

### Additional info

jourjine, 2016-09-10:
Verbose mode can be enabled via `-v, --verbose` option
Collected 38 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ESP8266WiFi> v1.0

|-- <Blynk> v0.3.8
|   |-- <ESP8266WiFi> v1.0
Compiling .pioenvs/espduino/src/main.o
In file included from /Users/KOCMOHABT/.platformio/lib/Blynk_ID415/Blynk/BlynkConfig.h:14:0,
from /Users/KOCMOHABT/.platformio/lib/Blynk_ID415/Blynk/BlynkApi.h:14,
from /Users/KOCMOHABT/.platformio/lib/Blynk_ID415/BlynkApiArduino.h:14,
from /Users/KOCMOHABT/.platformio/lib/Blynk_ID415/BlynkSimpleEsp8266.h:18,

ivankravets, 2016-09-10:
@vshymanskyy do you have CI test for this example?

ivankravets, 2016-09-10:
I've got it. You use a new Blynk example but an old library. Run `pio lib -g update`.
See my config
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[env:nodemcuv2]
platform = espressif8266
framework = arduino
board = nodemcuv2
lib_deps = Blynk
```

And output http://pastebin.com/vQTj0XY0

closed 2016-09-10 by ivankravets

jourjine, 2016-09-10:
[image]
after `pio lib -g update` same situation:
[image]

ivankravets, 2016-09-10:
Open PlatformIO IDE Terminal and run `pio run -v`. We will see where is library located. Looks like you have a few library storages.

jourjine, 2016-09-10:
```
Collected 38 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ESP8266WiFi> v1.0 (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi)
|-- <EEPROM> v1.0 (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/EEPROM)
|-- <PID> (/Users/KOCMOHABT/.platformio/lib/PID_ID2)
|-- <Blynk> v0.3.8 (/Users/KOCMOHABT/.platformio/lib/Blynk_ID415)
|   |-- <ESP8266WiFi> v1.0 (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi)
|-- <ArduinoOTA> v1.0 (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/ArduinoOTA)
|   |-- <ESP8266WiFi> v1.0 (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi)
|   |-- <ESP8266mDNS> (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/ESP8266mDNS)
|   |   |-- <ESP8266WiFi> v1.0 (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi)
|-- <ESP8266mDNS> (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/ESP8266mDNS)
|   |-- <ESP8266WiFi> v1.0 (/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi)
xtensa-lx106-elf-g++ -o .pioenvs/espduino/firmware.elf -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,-

-gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -Wl,-T"esp8266.flash.4m1m.ld" .pioenvs/espduino/src/mai
n.o -L/Users/KOCMOHABT/.platformio/packages/framework-arduinoespressif/tools/sdk/lib -L/Users/KOCMOHABT/.platformio/platforms/esp
ressif8266/ldscripts -L.pioenvs/espduino -Wl,--start-group .pioenvs/espduino/libFrameworkArduinoVariant.a .pioenvs/espduino/libFr
ameworkArduino.a -lmesh -lwpa2 -lsmartconfig -lpp -lmain -lwpa -llwip_gcc -lnet80211 -lwps -lcrypto -lphy -lhal -laxtls -lgcc -lm
 -lstdc++ .pioenvs/espduino/lib/libESP8266WiFi.a .pioenvs/espduino/lib/libEEPROM.a .pioenvs/espduino/lib/libPID_ID2.a .pioenvs/es
pduino/lib/libBlynk_ID415.a .pioenvs/espduino/lib/libESP8266mDNS.a .pioenvs/espduino/lib/libArduinoOTA.a -Wl,--end-group
```
```
**.pioenvs/espduino/src/main.o:(.text._Z12mainCookLoopv+0x94): undefined reference to `dtostrf_internal(double, signed char, unsign
ed char, char*)'
.pioenvs/espduino/src/main.o:(.text._Z12mainCookLoopv+0x151): undefined reference to `dtostrf_internal(double, signed char, unsig
ned char, char*)'
.pioenvs/espduino/src/main.o:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<Bly
nkProtocol<BlynkArduinoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x30): undefined reference to `BlynkWidgetRe
ad'
.pioenvs/espduino/src/main.o:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<Bly
nkProtocol<BlynkArduinoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x34): undefined reference to `BlynkWidgetWr
ite'
.pioenvs/espduino/src/main.o:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<Bly
nkProtocol<BlynkArduinoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x3c): undefined reference to `GetReadHandle
r'

.pioenvs/espduino/src/main.o:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<Bly
nkProtocol<BlynkArduinoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x40): undefined reference to `BlynkWidgetRe
adDefault'
.pioenvs/espduino/src/main.o:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<Bly
nkProtocol<BlynkArduinoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x44): undefined reference to `GetWriteHandl
er'
.pioenvs/espduino/src/main.o:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<Bly
nkProtocol<BlynkArduinoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x48): undefined reference to `BlynkWidgetWr
iteDefault'
.pioenvs/espduino/src/main.o: In function `BlynkApi<BlynkProtocol<BlynkArduinoClientGen<Client> > >::processCmd(void const*, unsi
gned int)':
main.cpp:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<BlynkProtocol<BlynkArdu
inoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x212): undefined reference to `GetReadHandler'
main.cpp:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<BlynkProtocol<BlynkArdu
inoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x22f): undefined reference to `BlynkWidgetReadDefault'
main.cpp:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<BlynkProtocol<BlynkArdu
inoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x253): undefined reference to `GetWriteHandler'
main.cpp:(.text._ZN8BlynkApiI13BlynkProtocolI21BlynkArduinoClientGenI6ClientEEE10processCmdEPKvj[BlynkApi<BlynkProtocol<BlynkArdu
inoClientGen<Client> > >::processCmd(void const*, unsigned int)]+0x276): undefined reference to `BlynkWidgetWriteDefault'**
```

**collect2: error: ld returned 1 exit status
*** [.pioenvs/espduino/firmware.elf] Error 1**

jourjine, 2016-09-10:
too many <ESP8266WiFi>...

ivankravets, 2016-09-10:
Could you try to clean project? Also, try
```
pio lib -g uninstall Blynk
pio lib -g install Blynk
```

jourjine, 2016-09-10:
i copy all lib to one global default folder
already updated but have new error
look about esp8266 lib
[image]
[image]

jourjine, 2016-09-10:
new project also don`t find libraries in default folder

ivankravets, 2016-09-10:
Could you share here `platformio.ini`? Do you set custom `lib_dir`?

jourjine, 2016-09-10:
```
[env:espduino]
platform = espressif8266
framework = arduino
board = espduino
//upload_port = /dev/cu.SLAB_USBtoUART
upload_port = 192.168.1.174

```

clean all libs from all places
install local only one lib esp8266wifi (w/o -g)
same problem
clean
install Blynk lib
install esp8266wifi
blynk is visible, esp8266wifi not
[image]

jourjine, 2016-09-10:
[image]

jourjine, 2016-09-10:
deleted row 5 and 6
same same

jourjine, 2016-09-10:
if i add manualy - visible
[image]
[image]

if i add via `pio lib` no

jourjine, 2016-09-10:
build it
[image]
manually add 4 libs

[image]
only Blynk and PID via Platformio

jourjine, 2016-09-10:
[image]

jourjine, 2016-09-10:
wtf? :)

jourjine, 2016-09-11:
lib ESP8266wifi-1101 not same ESP8266WiFi but have similar name

lib 1101
```
//
//  ESP8266wifi.h
//
//
//  Created by Jonas Ekstrand on 2015-02-20.
//  ESP8266 AT cmd ref from https://github.com/espressif/esp8266_at/wiki/CIPSERVER
//
```

 ESP8266WiFi
```
/*
 ESP8266WiFi.h - esp8266 Wifi support.
 Based on WiFi.h from Arduino WiFi shield library.
 Copyright (c) 2011-2014 Arduino.  All right reserved.
 Modified by Ivan Grokhotkov, December 2014

 This library is free software; you can redistribute it and/or
 modify it under the terms of the GNU Lesser General Public
 License as published by the Free Software Foundation; either
 version 2.1 of the License, or (at your option) any later version.

 This library is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 Lesser General Public License for more details.

 You should have received a copy of the GNU Lesser General Public
 License along with this library; if not, write to the Free Software
 Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
 */
```

jourjine, 2016-09-11:
библиотека WiFi-870 аналог ESP8266WiFi, но старая, не модифицированная Ivan Grokhotkov и с ней библиотека Блинка не работает....

ivankravets, 2016-09-11:
People will receive a lot of errors while they will develop "for Arduino IDE". I don't know why you installed all libraries in GLOBAL storages. I understand that you do the same with Arduino IDE. You do it with Arduino IDE because they don't provide normal package management system. Please forget about Arduino IDE workflow.

Each project should have OWNN dependencies. Don't use global. PlatformIO provides multiple instruments for professional development that will save your and our time.

Please remove totally `~/.platformio/lib` folder and use [lib_deps](http://docs.platformio.org/en/stable/projectconf.html#lib-deps) .

Se valid workflow https://github.com/platformio/platformio/issues/775#issuecomment-246139564