platformio/platformio-core #776: section '.text' will not fit in region 'iram1_0_seg'

profra, 2016-09-10:
What kind of issue is this?

- [x ] PlatformIO Core. If you’ve found a bug, please provide an information below.

I am not sure if it is bug... I have found this link https://github.com/esp8266/Arduino/issues/1582 but I am not able to solve this problem... Please help me if possible

------------------------------------------------------------------

### Configuration

**Operating system**: W10

**PlatformIO Version** (`platformio --version`): CLI 3.0.1, IDE 1.5.0 (Atom)

### Description of problem
I cannot build simple programm with library....

#### Steps to Reproduce

1. pio run

### Actual Results
... see the attached picture 

### Expected Results
... simple build

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
[env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini

```
**Source file to reproduce issue:**

used library https://github.com/sumotoy/TFT_ILI9163C/tree/Pre-Release-1.0p7
main programm ... basicSetup.ino
```
#include <SPI.h>
#include <TFT_ILI9163C.h>

#define __CS  0   //(D3) ... fp
#define __DC  5   //(D1)
#define __RST 4   //(D2)
/*
 SCLK:D5
 MOSI:D7
*/

TFT_ILI9163C tft = TFT_ILI9163C(__CS, __DC, __RST);

void setup() {
  tft.begin();
  tft.print("Hello World!");
}

void loop(void) {
}...
```

### Additional info
[image]

profra, 2016-09-10:
I had to install Arduino IDE :-( .... 1.6.11... and it builds without any problem... hence is it BUG?

ivankravets, 2016-09-10:
Please move to appropriate repo https://github.com/platformio/platform-espressif8266/issues

closed 2016-09-10 by ivankravets

profra, 2016-09-10:
OK...thanks.