platformio/platformio-core #779: Development only library.json dependencies handling

marvinroger, 2016-09-11:
It's me again from [homie-esp8266](https://github.com/marvinroger/homie-esp8266), we're quite not here yet.

The v1.5.0 has `PubSubClient` as a dependency, and this is published on pio. Fine. From [library.json#version](http://docs.platformio.org/en/stable/librarymanager/config.html#version):

> If you push new commits to repository and do not update version field, the library will not be updated until you change the version.

The v2.0.0 needs `AsyncMqttClient` instead of `PubSubClient`. So I [simply changed the dependency](https://github.com/marvinroger/homie-esp8266/commit/9224e7b64b9e4ee21f2445e400ec074440be7604). I did not update the `version` field right, so the library should not have been updated in PlatformIO.  But [it did](https://github.com/marvinroger/homie-esp8266/issues/129), so I had to [revert the change](https://github.com/marvinroger/homie-esp8266/commit/2f28f1dca183b0c7f5186264c4a5a0f60f6bce69).

So I am stuck as [the CI fails](https://travis-ci.org/marvinroger/homie-esp8266/builds/159052921) because  the [current library.json](https://github.com/marvinroger/homie-esp8266/blob/8150360d612f1b8d2f9c384fe1b5fc6485f205ee/library.json) does not reference `AsyncMqttClient`, for the above reason. It was fine with PlatformIO v2 because libraries were installed manually in `.travis.yml`, as we discussed.

Any idea?

ivankravets, 2016-09-11:
1. There is library manifest file. It has UNIQUE MANIFEST url which PlatformIO Library Crawler monitors occasionally. If you change version field or modify MANIFEST CONTENT, it will force to update a  library.
2. The problem here that your UNIQUE MANIFEST url points to 2 different libraries at the same time. That is not possible. 1 MANIFEST per ONE version.
3. Yes, I understand what do you need. You want to have RELEASE manifest for USERS, and DEVELOPMENT manifest for developers of library.

We 2 solutions:
1. You will create somewhere RELEASE manifest and fill it as you need. The only one requirement is that it should be visible from HTTP. AS for me, this is a dirty solution.
2. Normal and recommended solution. Need to have 2 different branches: `master` for releases and `develop` for development purpose. It's normal practice. If you want to save your time on it, I recommend to use [git flow](http://nvie.com/posts/a-successful-git-branching-model/). We use it in our repository and are very happy.

As result, we will switch PlatformIO Library Registry to MASTER branch, where in DEVELOP you will work on the new features and can modify manifest.

P.S: If you don't want to play with it, we can host your config here https://github.com/platformio/platformio-libmirror/tree/master/configs

marvinroger, 2016-09-22:
Thanks @ivankravets! I did not start the project with git flow as I did not think it would become such a big project. I've just created the develop branch and added a revert commit on master to the last stable release. The registry is already pointing to master, so everything should be fine as dev now happens in develop.

marvinroger, 2016-09-22:
@ivankravets Almost there... https://travis-ci.org/marvinroger/homie-esp8266/jobs/161860234

> Error: Could not find `ESPAsyncTCP` dependency for `AsyncMqttClient` library

`Homie` requires `AsyncMqttClient`, that requires `ESPAsyncTCP`. PlatformIO did not install `ESPAsyncTCP` by itself. So I have to manually add dependencies of dependencies in the manifest? Sounds like a bug.

ivankravets, 2016-09-22:
I got it. See https://travis-ci.org/marvinroger/homie-esp8266/jobs/161860234#L205

The `espressif` dev platform was renamed to `espressif8266` (we wait for `espressif32` :) ). We have multiple hooks with "accepting" `espressif` as `espressif8266`, but it doesn't work for dependency search. It is 100% strict (because we allow to have more than 1 library with the same name, etc) :(

Could you fix it in your manifest [here](https://github.com/marvinroger/homie-esp8266/blob/develop/library.json#L21)? 

Thanks!

marvinroger, 2016-09-22:
Still failing [with same error](https://travis-ci.org/marvinroger/homie-esp8266/jobs/161906887#L205).

ivankravets, 2016-09-22:
OK, I'll fix "renamed" platforms on-the-fly. Just updated search API. Could you re-start a Travis.CI job?

marvinroger, 2016-09-22:
Awesome, it works now, thanks!

closed 2016-09-22 by marvinroger