platformio/platformio-core #780: SoftwareSerial library creates DCF77 error

thearduinoguy, 2016-09-12:
When trying to run the SoftwareSerial examples I get the following error:

Error: Could not find `Timezone` dependency for `DCF77` library

ivankravets, 2016-09-12:
> platformio lib install DCF77

Library Storage: ****
Looking for DCF77 library in registry
Found: http://platformio.org/lib/show/169/DCF77
LibraryManager: Installing id=169
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
DCF77 @ 1688eaab55 has been successfully installed!
Installing dependencies
Looking for Time library in registry
Found: http://platformio.org/lib/show/44/Time
LibraryManager: Installing id=44
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Time @ 18ce15b202 has been successfully installed!
Looking for Timezone library in registry
Found: http://platformio.org/lib/show/76/Timezone
LibraryManager: Installing id=76
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Timezone @ 510ae2f6b6 has been successfully installed!

ivankravets, 2016-09-12:
Please try again and reopen this issue if you still have problems

closed 2016-09-12 by ivankravets