platformio/platformio-core #783: Cannot compile Adafruit Adafruit_SSD1306

deladriere, 2016-09-14:
using IDE 1.5.0 | CLI 3.0.1
Was compiling fine under version 2
Compiles  fine with Arduino 1.6.11

[env:adafruit_feather_m0_usb]
platform = atmelsam
framework = arduino
board = adafruit_feather_m0_usb

code : 
`
#include <Arduino.h>
#include <SPI.h>
#include <SD.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

/*
 █████  ██████  ██████  ██████  ███████ ███████ ███████ ███████ ███████
██   ██ ██   ██ ██   ██ ██   ██ ██      ██      ██      ██      ██
███████ ██   ██ ██   ██ ██████  █████   ███████ ███████ █████   ███████
██   ██ ██   ██ ██   ██ ██   ██ ██           ██      ██ ██           ██
██   ██ ██████  ██████  ██   ██ ███████ ███████ ███████ ███████ ███████

OLED 0x3C
AQUESCLICK 0X2E

*/

 /*
    ██   ██  █████  ██████  ██████  ██     ██  █████  ██████  ███████
    ██   ██ ██   ██ ██   ██ ██   ██ ██     ██ ██   ██ ██   ██ ██
    ███████ ███████ ██████  ██   ██ ██  █  ██ ███████ ██████  █████
    ██   ██ ██   ██ ██   ██ ██   ██ ██ ███ ██ ██   ██ ██   ██ ██
    ██   ██ ██   ██ ██   ██ ██████   ███ ███  ██   ██ ██   ██ ███████

  */

  #define SW0 0
  #define SW1 1
  #define GATE 2
  #define PUSH 3
  #define RST_Microbus 4
  #define CS_Microbus 5
  #define ROTA 6
  #define ROTB 7
// A6 is also D8 on Adafruit Feather
  #define AN_Microbus 9 // A7 can be D9 on Adafruit Feather
  #define CS_SD 10
  #define INT_Microbus 11
  #define BLUE_LED  12
  #define RED_LED 13

  /*
     ██████  ██████       ██ ███████  ██████ ████████ ███████
     ██    ██ ██   ██      ██ ██      ██         ██    ██
     ██    ██ ██████       ██ █████   ██         ██    ███████
     ██    ██ ██   ██ ██   ██ ██      ██         ██         ██
     ██████  ██████   █████  ███████  ██████    ██    ███████
   */

  File root;
  Adafruit_SSD1306 display(0); // modif library for 64 pxls

  /*
  ███████ ██    ██ ███    ██  ██████ ████████ ██  ██████  ███    ██ ███████
  ██      ██    ██ ████   ██ ██         ██    ██ ██    ██ ████   ██ ██
  █████   ██    ██ ██ ██  ██ ██         ██    ██ ██    ██ ██ ██  ██ ███████
  ██      ██    ██ ██  ██ ██ ██         ██    ██ ██    ██ ██  ██ ██      ██
  ██       ██████  ██   ████  ██████    ██    ██  ██████  ██   ████ ███████
  */

void toggleOUTPUTS()
{
static boolean toggle;
toggle =! toggle;

digitalWrite(BLUE_LED,toggle);
digitalWrite(RED_LED,toggle);
digitalWrite(RST_Microbus,toggle);
digitalWrite(CS_Microbus,toggle);
digitalWrite(CS_SD,toggle);

}

  /*
     ███████ ███████ ████████ ██    ██ ██████
     ██      ██         ██    ██    ██ ██   ██
     ███████ █████      ██    ██    ██ ██████
          ██ ██         ██    ██    ██ ██
     ███████ ███████    ██     ██████  ██
   */

void setup() {

  Serial.begin(9600);

  pinMode(SW0, INPUT_PULLUP);
  pinMode(SW1, INPUT_PULLUP);
  pinMode(GATE, INPUT_PULLUP);
  pinMode(PUSH, INPUT_PULLUP);
  pinMode(RST_Microbus, OUTPUT);
  pinMode(CS_Microbus, OUTPUT);
  pinMode(ROTA, INPUT_PULLUP);
  pinMode(ROTB, INPUT_PULLUP);
  pinMode(CS_SD, OUTPUT);
  pinMode(INT_Microbus,INPUT_PULLUP);
  pinMode(BLUE_LED,OUTPUT);
  pinMode(RED_LED,OUTPUT);

  // turn transistorized outputs low ;
  digitalWrite(BLUE_LED,HIGH);
  digitalWrite(RED_LED,HIGH);

}

/*
██       ██████   ██████  ██████
██      ██    ██ ██    ██ ██   ██
██      ██    ██ ██    ██ ██████
██      ██    ██ ██    ██ ██
███████  ██████   ██████  ██
*/

void loop() {

toggleOUTPUTS();
delay(1000);

}
`

deladriere, 2016-09-14:
seems that the include are gone 

#include <Arduino.h>
#include <SPI.h>
#include <SD.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

ivankravets, 2016-09-14:
Please use triple of ``` (in new line, begin and end) for code highlighting

deladriere, 2016-09-14:
using paste bin : http://pastebin.com/gZ8rr5c2

deladriere, 2016-09-15:
can you reproduce this issue ?

ivankravets, 2016-09-15:
Please read about LDF and Dependency Finder Mode http://docs.platformio.org/en/stable/librarymanager/ldf.html#dependency-finder-mode

Adafruit SSD1306 depends on Wire library but doesn't include it directly on header file. You should manually resolve this issue or switch `lib_ldf_mode = 2`. However, I don't recommend to do it.

See other normal solutions:

1. You can manually `#include <Wire.h>` in your project main file.
2. You can ask PlatformIO Library Builder include `Wire` library automatically. See:

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[env:adafruit_feather_m0_usb]
platform = atmelsam
board = adafruit_feather_m0_usb
framework = arduino
lib_force = Wire
lib_deps = Adafruit SSD1306
```

closed 2016-09-15 by ivankravets