platformio/platformio-core #784: Problem with OTA PlatformIO 3.0

sembsa, 2016-09-16:
### Configuration

**Operating system**: macOS 10.12 GM

**PlatformIO Version** (`platformio --version`): 3.0

### Description of problem

My problem with ArduinoOTA, when i try upload new sketch ESP8266 show me Connect Failed.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp01_1m]
platform = espressif8266
framework = arduino
board = esp01_1m
lib_ldf_mode = 1
upload_port = Brama.local
upload_flags = --port=737 --auth=pass
```

**Source file to reproduce issue:**
```cpp
setup() {
  // Port defaults to 8266
  ArduinoOTA.setPort(737);

  // Hostname defaults to esp8266-[ChipID]
  ArduinoOTA.setHostname("Brama");
  Serial.print("Hostname: ");
  Serial.println(ArduinoOTA.getHostname());

  // No authentication by default
  ArduinoOTA.setPassword("mQa83zseba");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("4f46dbec237ea39ec49021fcea6095cc");

  ArduinoOTA.onStart([]() {
    String type;
    Serial.println(ArduinoOTA.getCommand());
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

loop() {
   ArduinoOTA.handle();
}
```

### Additional info

This work on PlatformIO 2.x

ivankravets, 2016-09-16:
Are you sure that it works with PIO2 but doesn't work with PIO3?

sembsa, 2016-09-17:
Yes, i'm try example: https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino

On PIO2 it's work before, now is problem.
It connects to esp8266, authorizes, when he wants to write it shows in the series "Processing: 0%" and connect Error

sembsa, 2016-09-17:
How turn debugging mode, maybe show you error.

ivankravets, 2016-09-17:
Do you use PlatformIO IDE or CLI?

sembsa, 2016-09-17:
I'm use PlatformIO IDE

ivankravets, 2016-09-17:
1. Please open PlatformIO IDE Terminal
2. Downgrade to PlatformIO 2.0:
```
pip install -U "platformio<3"
```

Does it really work with PlatformIO 2.0?

sembsa, 2016-09-17:
Ok I try downgrade if go to home.

Yes it's really work. This is work code on PIO 2.x
’’’#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char *ssid =	"SembsaMi_2DAB";		// cannot be longer than 32 characters!
const char *pass =	"***";		//

// Update these with values suitable for your network.
IPAddress server(192, 168, 54, 132);

const int relay = 2;

#define BUFFER_SIZE 100

void callback(const MQTT::Publish& pub) {
  // handle message arrived
  Serial.print(pub.topic());
  Serial.print(" => ");

    Serial.println(pub.payload_string());

    if(pub.payload_string() == "on")
    {
      digitalWrite(relay, HIGH);
    }
    else
    {
      digitalWrite(relay, LOW);
    }

}

WiFiClient espClient;
PubSubClient client(espClient);

void setup() {

  pinMode(relay, OUTPUT);

  // Setup console
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  delay(10);
  Serial.println();
  Serial.println();
  delay(100);

  client.set_server(server, 1883);
  client.set_callback(callback);

}

void setupOTA() {
  // Port defaults to 8266
  ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  ArduinoOTA.setHostname("esp8266-WeMo");

  // No authentication by default
  ArduinoOTA.setPassword((const char *)"***");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
}

void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    Serial.print("Connecting to ");
    Serial.print(ssid);
    Serial.println("...");
    WiFi.begin(ssid, pass);

    if (WiFi.waitForConnectResult() != WL_CONNECTED) return;
    Serial.println("WiFi connected");
    Serial.println(WiFi.localIP());
    setupOTA();
  }

  if (WiFi.status() == WL_CONNECTED) {
    if (!client.connected()) {
      ArduinoOTA.handle();
      Serial.println("Client ESP8266 not connect");
      if (client.connect("ESP8266: AdyLight")) {
	         client.publish("outTopic","hello world");
	         client.subscribe("AdyLight");
           Serial.println("Client ESP8266 connect");
      }
    }

    if (client.connected())
      client.loop();
  }
  ArduinoOTA.handle();
}
’’’

sembsa, 2016-09-17:
Now don't work this code :( this same problem

ivankravets, 2016-09-17:
Can I close this issue?

ivankravets, 2016-09-17:
Check it with Arduino IDE. Does it work?

sembsa, 2016-09-17:
No, now every my code with ArduinoOTA not working

ivankravets, 2016-09-17:
Please fix your code.

closed 2016-09-17 by ivankravets

sembsa, 2016-09-17:
Ok thanks for help