platformio/platformio-core #785: PIO 3 Invalid character escape '\U'

MeGaPk, 2016-09-16:
### Configuration

**Operating system**: Windows 10 Pro

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.0.1
### Description of problem
Using last version clion.
After ```pio init --ide clion``` i have error:
Error:Syntax error in cmake code at
C:/Users/Ivan/Documents/esp8266/CMakeListsPrivate.txt:4
when parsing string
C:\Users\Ivan.platformio\packages\toolchain-xtensa\bin\xtensa-lx106-elf-gcc.exe
Invalid character escape '\U'.

How i can resolve it?

#### Steps to Reproduce

1. create new dir
2. in new dir just exec ```pio init --ide clion -b nodemcuv2```
3. Open project in clion

Link from forum: https://community.platformio.org/t/invalid-character-escape-u/763/10

ivankravets, 2016-09-17:
Please update to 3.1.0a3 http://docs.platformio.org/en/stable/installation.html#development-version

Also, need to re-init project.

Thanks a lot for the report! 👍

closed 2016-09-17 by ivankravets

MeGaPk, 2016-09-17:
@ivankravets Yeap, but not completely fixed.

I get new version by: ```pip install -U https://github.com/platformio/platformio/archive/develop.zip```
create new project for clion.
But..., new error:
```
Error:Configuration Debug
Syntax error in cmake code at
  C:/Users/Ivan/Documents/test/CMakeListsPrivate.txt:51
when parsing string
  C:\Users\Ivan\Documents\test\src
Invalid character escape '\U'.
```

My file:
```
set(ENV{PATH} "C:\\Python27\\;C:\\Python27\\Scripts;C:\\ProgramData\\Oracle\\Java\\javapath;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files (x86)\\Skype\\Phone\\;C:\\Program Files\\Git\\cmd;C:\\Go\\bin;C:\\Users\\Ivan\\work\\bin;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\MinGW\\bin;C:\\MinGW\\msys\\1.0\\bin;C:\\Users\\Ivan\\AppData\\Local\\Microsoft\\WindowsApps;")
set(PLATFORMIO_CMD "C:\\Python27\\Scripts\\platformio.exe")

SET(CMAKE_C_COMPILER "C:/Users/Ivan/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc.exe")
SET(CMAKE_CXX_COMPILER "C:/Users/Ivan/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++.exe")
SET(CMAKE_CXX_FLAGS_DISTRIBUTION "-fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030100 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -U__STRICT_ANSI__")
SET(CMAKE_C_FLAGS_DISTRIBUTION "-std=gnu99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=030100 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -U__STRICT_ANSI__")

add_definitions(-DARDUINO=20300)
add_definitions(-DLWIP_OPEN_SRC)
add_definitions(-DF_CPU=80000000L)
add_definitions(-D__ets__)
add_definitions(-DICACHE_FLASH)
add_definitions(-DPLATFORMIO=030100)
add_definitions(-DESP8266)
add_definitions(-DARDUINO_ARCH_ESP8266)
add_definitions(-DARDUINO_ESP8266_NODEMCU)

include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/cores/esp8266")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/variants/nodemcu")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/Documents/test/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ArduinoOTA")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/DNSServer/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/EEPROM")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266AVRISP/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266HTTPClient/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266HTTPUpdateServer/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266SSDP")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFiMesh/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266httpUpdate/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/Ethernet/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/GDBStub/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/Hash/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/SD/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/SPI")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/Servo/src")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/framework-arduinoespressif8266/libraries/TFT_Touch_Shield_V2")
[... cut ...]
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include")
include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include-fixed")

aux_source_directory(C:\Users\Ivan\Documents\test\src SRC_LIST)
```

Problem in line: ```aux_source_directory(C:\Users\Ivan\Documents\test\src SRC_LIST)```
Again  slashes.

ivankravets, 2016-09-17:
Could you repeat again? 
"""
I get new version by: pip install -U https://github.com/platformio/platformio/archive/develop.zip
create new project for clion.
But..., new error:
"""

MeGaPk, 2016-09-17:
@ivankravets 
Yeap, fixed :+1:

ivankravets, 2016-09-17:
Happy Coding with PlatformIO! :)

ivankravets, 2016-09-17:
Don't forget that you need perform "Rebuild C/C++ Index" each time when add new library or change project settings.

MeGaPk, 2016-09-17:
@ivankravets In Clion you mean "Reload Cmake project"? 

And i have question, why after "pio lib instal ..." need use "pio init" for add this library to CMake file?

ivankravets, 2016-09-17:
If you add new library, you need to run [PLATFORMIO_REBUILD_PROJECT_INDEX](http://docs.platformio.org/en/stable/ide/clion.html#integration) target.

`pio init` commands work in both directions: as project generator, also as a way to update existing project. This complicated system. We don't use make or Cmake. We have own Python-based build system based on SCons tool.

So, when you run `pio init` we launch 100% real build environment and gather all required data for the specific IDE.

MeGaPk, 2016-09-17:
@ivankravets just in my case:
I install new lib, but my Clion don't see new library... its very strange for newbie, like me.
By google, i found (not easy), need just use pio init...  But for newbie, this is hell :).

ivankravets, 2016-09-17:
Hm.. We write about it in documentation :) Firstly, need to check docs, then google.

We had a discussion with JetBrains as for the this issue. Need to write some plugin that will monitor PlatformIO's Library storages and `platformio.ini` file. If there are new folders or project file is changed, need to regenerate project index.

We do these in PlatformIO IDE where all work from a box without "google" :)

commit 37552a81 (2016-09-17): Fix Project Generator for CLion IDE using Windows OS // Resolve #785
commit c12df19f (2016-09-17): Fix Project Generator for CLion IDE using Windows OS // Resolve #785